bigtreetech/BIGTREETECH-TouchScreenFirmware #2761: [Q] Random pauses or hesitation when printing.

V1EngineeringInc, 2023-04-06:
While printing once or twice a print, sometimes more, the machine will stop right in the middle of printing for a second, maybe even two. No errors or anything, it leaves an ugly blob, and it just continues on. It does not happen the same place on the file or on the machine. If I had to say anything, it might happen more often during circle moves, or curves. This does not happen while using the same screens for my CNC machines.

I have 6 printers  and 2 CNC's running TFT35 V3 E3 screens. I have noticed this issue for probably as long as I have used the screens, more than two years I would guess. Some screens are brand new and some are very old. I update marlin and the TFT several times a year. Some of my gcode is old and ran perfectly fine on older 1286 screens, and some of my files are new.

So I think this is either some sort of configuration issue, slicer setting, or a screen firmware bug. Any ideas?

Config, 
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11165582/config.zip)

V1EngineeringInc, 2023-04-06:
Is there any sort of debugging I can run, check the buffer of the screen, or anything else that might help?

radek8, 2023-04-07:
Can you try recompiling marli with these values?
#define BLOCK_BUFFER_SIZE 16
#define MAX_CMD_SIZE 96
#define BUFSIZE 16
#define TX_BUFFER_SIZE 32
#define RX_BUFFER_SIZE 32

V1EngineeringInc, 2023-04-07:
Thanks I will try that.

> #define BLOCK_BUFFER_SIZE 16
> #define MAX_CMD_SIZE 96

Those are default values.

I will change the others. Is this in the documentation anywhere, seems like it kind of should be?

kisslorand, 2023-04-07:
I would also check the slicer for something like "Maximum Resolution" and/or "Maximum Deviation" and raise them a little. I do not know what slicer you're using, the parameters mentioned are from Cura. I remember the days when I was using Octoprint with an 8 bit mainboard printer and I had a lot of blobs on curved surfaces. Changing the values of these settings mitigated the issue.
By now I use ARC Welder so there's no issue whatsoever having tiny minuscule segments for curved lines. In fact the problem also went away on my 8 bit board using ARC Welder plugin in Octoprint.

V1EngineeringInc, 2023-04-07:
I am using a SKR Pro, and Simplify3D I can reduce the segments. Thanks for the tips.

Seems a bit off to me that it works fine on a rambo (8bit) with a 1286 screen, but the same Gcode only hesitates on the 32 bit board and TFT.

V1EngineeringInc, 2023-04-07:
Waiting for this print to finish before I change the firmware for the memory tweaks. I just watched it happen again. It happened on an outline (90% speed) that was a long 4" straight line and just as it hit the corner it paused before the curves.... So not printing fast, buffer should have easily been full, hmm.

This is really puzzling to me. Neither of you have seen your printers hesitate?

kisslorand, 2023-04-07:
I totally forgot about this issue since this TFT FW and 32 bits mainboard (and using ARC Welder).

V1EngineeringInc, 2023-04-07:
I hear you. I have sold these screens for a while now but I only sold them to use the Marlin side. You made it so the TFT was even reliable. I really hope they give this some more attention, if not I think I will be switching out to ESP3D control direct from the board and skip the screen all together. It is a bummer though I think these screens have huge potential. They need to be more receptive to some changes, stability and customizations.

I appreciate the help I get here though. I have always received sincere help.

(testing the mem changes now, just need to hang out near the printer to see if I can catch some hesitations)

ETE-Design, 2023-04-08:
> I totally forgot about this issue since this TFT FW and 32 bits mainboard (and using ARC Welder). Just for science... try Cura with ARC... :)
> 
> On the other hand this TFT FW has a major setback, it sends every command only after receiving "OK" from the mainboard from the previous command. I am contemplating for a while to integrate "Advanced OK" (it allows filling the mainboard's buffer with commands, not send them one by one), but giving the lack of interest from BTT towards this repository I do not feel comfortable wasting my time for nothing. It seems even the free work of the contributors is too expensive for them...
> 
> I do not want to spoil the game but I do not think the suggestions of @radek8 will help because of the aforementioned reasons. I can only hope I'm wrong... :)

I often tell BTT to merge new PR's but they are really slow at it, I even asked them to hand over the merge process to a person who develop on this but with no luck :-(
Would really love if you could find time to integrate Advanced OK to this Firmware... I know that BTT lack interest of firmware cause they don't make money on it, but I think a lot of us users really like all the work you guys put into this.

/ETE-Design

V1EngineeringInc, 2023-04-08:
I could try and ask my BTT rep to get something looked at or pushed to the front of the list if we had something specific. I have no idea if it would do anything but I could try. Maybe, just maybe, having the request come from two sides would help.

ETE-Design, 2023-04-08:
> I could try and ask my BTT rep to get something looked at or pushed to the front of the list if we had something specific. I have no idea if it would do anything but I could try. Maybe, just maybe, having the request come from two sides would help.

There is already a lot of PR who need to be merged, most of then seem to be things who need to be merged

V1EngineeringInc, 2023-04-09:
So I think those memory changes worked. I have been kinda busy but floating around one of the printers all day and I have not caught a single hesitation!!!

Should I (or one of us) add this to the firmware changes section of the INI file?

kisslorand, 2023-04-09:
Just another thought...
Do you have PLR (Power Loss Recovery) enabled on your TFTs? Writing speed is much slower than reading speed, printing high poly surfaces can exhaust the buffer pretty quick and it might be happening right when PLR checkpoints are written to SD or USB so the printer makes a slight pause thus creating blobs.
Maybe it worth checking this too.

V1EngineeringInc, 2023-04-09:
No, No power loss recovery. Good idea though.

ETE-Design, 2023-04-11:
So seems like it helped I written to BTT again over messenger... A lot of PR got merged now 😀
@kisslorand Hope you still will consider making Advanced OK for this TFT 😀

kisslorand, 2023-04-11:
@ETE-Design 

> @kisslorand Hope you still will consider making Advanced OK for this TFT 😀

Sure.

rondlh, 2023-04-14:
I also noticed some stuttering. My buffers are setup even larger than suggested. I use a serial baud rate of 1M bit. Arc welder could possibly help, I will give it a try.
Support for ADVANCED_OK would be amazing, but seems like a big project.

slwise, 2023-04-15:
A friend and I were both running a SKR Mini E3 V3 with the TFT35 V3 E3. We were both experiencing the same issue you described with pausing and blobs in the print. I changed the buffer size in Marlin to 64 but no change. I also increased the baud rate from 115200 to 250000, still no change. I finally fixed the issue on both printers by lowering the baud rate to 57600. We've been running this way for months with no issues. My best guess is apparently 115200 was too fast for the SKR to handle and causing the TFT to have to retransmit the data. Hope this helps.

V1EngineeringInc, 2023-04-15:
Interesting, I need to try that. As well, I think I caught another hesitation with the memory tweaks, but I am not sure. It is definitely happening much much less. I am set to 250000....

digant73, 2023-04-17:
> A friend and I were both running a SKR Mini E3 V3 with the TFT35 V3 E3. We were both experiencing the same issue you described with pausing and blobs in the print. I changed the buffer size in [...]

TFT simply sends a gcode wen the printer's FW (Marlin etc...) ackwoledges the gcode previously sent by TFT. There is no retransmission from TFT. The issue seems to be in Marlin (probably related to buffers and/or connection speed). Ironically, some speed improvements made on TFT fw triggered this issue on some printer's fw with an unbalanced setup

V1EngineeringInc, 2023-04-18:
I still have not gotten to testing the Baud rate change. I can now confirm the memory changes did not fix it. It might be slightly better but my printers still hesitate randomly.

I will test the Baud rate after RMRRF.

africsnail, 2023-04-30:
I also have a similar issue after updating my board's fw to Marlin 2.1 (previously I had 2.0 and I had no problems). I also made sure to update the TFT35. Now it seems to have some kind of mess ups with the gcode being corrupted (like a missing or additional character in a command) in some way or it just pauses randomly as described here. I tried many different buffer sizes and neither of them worked. 

I am going to try the lower baudrate as mentioned here. I was running at the default 115200.

kisslorand, 2023-04-30:
> @kisslorand Hope you still will consider making Advanced OK for this TFT 😀

I just remembered why I haven't got working on "Advanced OK". There's a bug on SKR and SKR Mini motherboards (there might be others affected too), a bug that can be avoided only if commands are sent one by one, so "Advanced OK" would be a disaster on these motherboards. I saw the issue came up a while ago on the Marlin repo but there is no sign of anyone working on a solution.

ETE-Design, 2023-05-04:
> I just remembered why I haven't got working on "Advanced OK". There's a bug on SKR and SKR Mini motherboards (there might be others affected too), a bug that can be avoided only if commands are sent [...]

And how about all other boards that isen't SKR-Series? 😀
Is it not just to make a Note in the first part of Config.ini to tell people not to activate it if they have SKR-Series Boards?

kisslorand, 2023-05-04:
Each configuration that has to be taken into account during printing slows down the sending of the gcodes which itself defeats the advantages of the Advanced OK.

ETE-Design, 2023-05-05:
> Each configuration that has to be taken into account during printing slows down the sending of the gcodes which itself defeats the advantages of the Advanced OK.

@kisslorand  think you misunderstood me, make a note in the top of Config.ini like:

"#  Advanced_OK (in Configuration_adv.h) - Issue with SKR-Series"

That shouldn't slow down sending of gcodes?

rondlh, 2023-05-05:
> A friend and I were both running a SKR Mini E3 V3 with the TFT35 V3 E3. We were both experiencing the same issue you described with pausing and blobs in the print. I changed the buffer size in [...]

Interesting, but serial communication is handled in hardware, not in software. The software will get a signal when a command is received completely ('\n' at the end. Then it's up to the processor to handle the command.). I personally run all serial communication at 1M bit. If a command is 40 characters long then it will take at least 10 * 40 / 57600 seconds (=7ms) to transmit. As long as you don't have data corruption I do not see any disadvantage on running the serial lines faster.

kisslorand, 2023-05-05:
> @kisslorand Think you misunderstood me, make a note in the top of Config.ini like:
> 
> "# Advanced_OK (in Configuration_adv.h) - Issue with SKR-Series"

I might have misunderstood you but the issue remains. If Advanced OK is implemented, it should be configurable (enabled/disabled) because of the problematic motherboards. If it's configurable than it adds latency in preparing the gcodes to be sent during printing which defeats the purpose of Advanced OK. The added latency will surely hurt at least the problematic motherboards, they will not benefit the advantages of Advanced OK but will be burdened by this extra latency.
Also for other motherboards I have no way to estimate if the added latency will be less of a problem than the benefits of the Advanced OK. I keep wondering why Gina didn't add Advanced OK to Octoprint, knowing that the Raspberry Pi's MCU is much more powerful than the ones we have in our TFTs.

Only real tests can answer the dilemma, but I do not feel comfortable working huge number of hours for possible negative results, not to mention that a particular user will rip my head off accusing me I made another "useless, stupid, not necessary, full of bugs" modification to this firmware when "everything was working perfectly".

V1EngineeringInc, 2023-05-05:
Any interest in your own fork?

Forgive me if I am not really understanding how this all works but it seems the firmware is fairly stable. If you made a custom fork, your way, would it be easier to just get these things fast and stable. Even if you did not bother to keep them in sync. For most of us, in terms of 3D printing the screen has all the features we need right, heck maybe even some we don't? Then we have a solid stable version to run.

As my printer designs increase in speed, the pitfalls of this screen are starting to really hinder print quality. I have been actively researching other board and screen combos. I don't want my users to think it is a printer issue when it is a screen issue.

Beyond that, I would like to make a few buttons CNC specific, but I might be able to get help with that.

kisslorand, 2023-05-06:
> As my printer designs increase in speed, the pitfalls of this screen are starting to really hinder print quality. I have been actively researching other board and screen combos. I don't want my users to think it is a printer issue when it is a screen issue.

Maybe it would be a good idea to use TFTs with the most powerful MCU and with custom FW that have the not needed, unused features eliminated to free the MCU during printing (ex. checking input from all serial ports used by ESP3D, Octoprint, etc;; comment parsing, eliminating RRF support and so on)

Also you might consider switching to Klipper...

V1EngineeringInc, 2023-05-06:
I am not really sure how to take that. I hope my message came off as supportive and encouraging. I think the board and screen combo has so much potential to grow into, but if you think it doesn't than I will make the switch sooner than later for the printers I work with.

I really admire your work, and was hoping if you did your own fork we could have a stand-out setup, and you would have the ability to develop without getting hindered at each pull request.

digant73, 2023-05-12:
@V1EngineeringInc Hopefully #2788 recently merged could fix the stuttering on printing you reported

rondlh, 2023-05-15:
@kisslorand 
> Only real tests can answer the dilemma, but I do not feel comfortable working huge number of hours for possible negative results, not to mention that a particular user will rip my head off accusing [...]

You have my support if you need a beta tester. I have several test prints, like a vase mode cylinder that shows the effects of stuttering very clearly. I did some test with Marlin using D576 (Buffer Monitoring) which shows that buffer underruns happen frequently, but the wait time is quite low (< 50ms). I believe advanced_ok should completely solve the stuttering.
I agree, this the outcome of implementing advanced_ok is not certain, and bugs will happen, but either way I hope the community can be constructive and supportive. Currently advanced_ok is the most important missing feature of the TFT.

kisslorand, 2023-05-16:
@V1EngineeringInc
@rondlh
@ETE-Design
@radek8 
@AllTheOthers

I created a [Discord channel](https://discord.gg/dM5wqktj) where we can discuss about these issues.

kisslorand, 2023-05-22:
I found a possible solution for the hesitations during prints. Whoever wants to test it let me know.

BTW, Advanced_OK is already in Alpha phase (virtual prints have positive results).

makersteve3d, 2023-05-22:
Count me in for testing too :)

kisslorand, 2023-05-22:
Let me know your TFT variant, I will compile it for you

V1EngineeringInc, 2023-05-22:
tft35 e3 v3 gd version and regular non GD version

V1EngineeringInc, 2023-05-22:
Testing it now.

makersteve3d, 2023-05-22:
I won’t be able to test until tomorrow evening, but watching with interest!

rondlh, 2023-05-23:
Great job!
Testing the BTT TFT35 V3.0 version. I will run some test prints and report back.

kisslorand, 2023-05-23:
> NOTE: Don't forget to enable ADVANCED_OK in Marlin's Configuration_adv.h

Advanced_OK is not implemented in the files posted on Mega. For a test run of Advanced_OK we can discuss on the [Discord channel](https://discord.gg/dM5wqktj).

V1EngineeringInc, 2023-05-23:
So far so good, I have not sat and watched them for very long but I am not seeing any blobs on the outer surface yet. More prints today. I have 3 printers running the test BIN's.

V1EngineeringInc, 2023-05-23:
Yes, But I am looking for specific prints that have more than others. Since I have a small print farm running and an inventory of parts sitting here I am looking for some with more issues than others. I am just a little behind right now so I have to print specific parts to keep up with orders.

rondlh, 2023-05-24:
> Gaps appear at 150mm/s like before.
This is a limit of my hotend, with a 0.5mm nozzle/line width, and 0.25mm layer height, 145mm/s works fine (about 450mm of 1.75mm filament per minute), but 150mm/s fails (gaps), the extrusion skips for a short time and gaps become visible in the printed (test) part, a spiralized cylinder. 

> The version on Mega seems to solve the hiccups for me.
I call a hiccup when I visibly see that the movement stops for a fraction of a second, normally the movement should be continues.

> > The Discord doesn't like my VPN... I cannot get in.
> I can see you already joined the Discord channel a few days ago,
Right, I tried to join, perhaps successfully, but it doesn't allow me to login because of suspicious network activity. That's probably just the VPN I use, I'm stationed in China.

V1EngineeringInc, 2023-05-24:
Okay it seems this doesn't change anything. I am running one of my printers at 80% speed to use up some filament I don't care for. The prints are still clearly getting blobs. The blabs are in completely different places, but they are all on curves, I have seen them on large very subtle curves and very small 5-6mm curves. No blobs on straight lines. (oddly it the one I ran slower has more blobs??))

So that seems to rule out speed, and possibly even buffer issues right?

[SKR Box.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11557810/SKR.Box.zip)
This file gets them on the outer two circular edges. Any info to learn from this?

V1EngineeringInc, 2023-05-24:
Oh and I almost forgot, your code messes up homing and leveling. The printers will start before temp is reached somehow. normally you start a print and when the bed and extruder are at temp homing and leveling start, yours starts as soon as I hit go. That is not good.

rondlh, 2023-05-25:
> Oh and I almost forgot, your code messes up homing and leveling. The printers will start before temp is reached somehow. normally you start a print and when the bed and extruder are at temp homing and leveling start, yours starts as soon as I hit go. That is not good.

Right, I also noticed that printing started before the bed was at the right temperature. Luckily my bed warms up fast so it was not a big issue.

kisslorand, 2023-05-28:
v1 was "proof of concept". In the meantime there was a v2 and now there's [v3](https://mega.nz/folder/K4IllSJb#7rAmrVNuYALpUdOeBtU5Vg) version. [v3](https://mega.nz/folder/K4IllSJb#7rAmrVNuYALpUdOeBtU5Vg) has the fully implemented "hiccup avoidance" and it comes with a hefty RAM reduction (+1600 bytes) and without FW size change (actually there's a 6 bytes increase :) ). [It](https://mega.nz/folder/K4IllSJb#7rAmrVNuYALpUdOeBtU5Vg) has many changes under the hood, if you want to take [v3](https://mega.nz/folder/K4IllSJb#7rAmrVNuYALpUdOeBtU5Vg) for a ride it is available on [Mega](https://mega.nz/folder/K4IllSJb#7rAmrVNuYALpUdOeBtU5Vg).

Please let me know how it goes for you.

rondlh, 2023-05-31:
Thank, testing V3...

kisslorand, 2023-05-31:
There's a nasty bug at the end of a print. I am working on it.

kisslorand, 2023-06-05:
Made some further work, synced to master, did some bugfixes and even more stability/speed enhancements.
As a result [v3.1](https://mega.nz/folder/2pZ3SLCB#TKxTWyAGr6EVsxmITHOwrA) is up on [Mega](https://mega.nz/folder/2pZ3SLCB#TKxTWyAGr6EVsxmITHOwrA).

Note: The builds also include PR #2701, PR #2793, PR #2798, PR #2799, PR #2805 and a rework of #2788 (which is a partial fix only and introduces a new bug).

rondlh, 2023-06-11:
I also tested v3.1(1 test print, 2 small 1 hours functional parts), works fine, no issues founds during start or at end of the print, no pauses or hesitations during printing. The added PRs are of course also very welcome. 
I hope this build can be merged soon with the main build.
Great job again!

Nerdcaught, 2023-06-21:
Hey everyone, so I was wondering if I can get some help I've been at this problem for a long time, and this thread on GitHub and anywhere has been the closest to solving my problem. My issue stems from compiling my own Marlin. I'm 98% sure its two settings contradicting themselves. But I've tried a lot and still cant get it to work. So far I've narrowed it down to software as I've tried a new Motherboard ( SKR mini E3 V3 ) and a New TFT35 E3 V3.0.1 GD. replacements and issues still persist. I cant really see it when printing slowly, only when adjusting the speed itself on the TFT35. Printing directly from the Motherboard and Octoprint works just fine. From what I can see now Leaving  M154 AUTO_REPORT_POSITION/AUTO_REPORT_REAL_POSITION   enabled and disabling anything M114 seems to work fine when Adjusting speeds, but I'm not sure if that is 100% viable down the [road.](url) also not sure why nothing will work on baud rate 115200 only 57600. Here is a video to describe what happens. its hard to explain in words.  Let me know if there's anymore details you need first time  posting. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/assets/137300961/0ebd4117-5a01-4201-b0ce-cec1bca6fc5b

Finally and Im not sure if this is known or not if I have S-Curve on it will always fail, I think maybe its incompatible with Input shaping?

rondlh, 2023-06-22:
@Nerdcaught This is not the correct place for your question, please delete your post. This is not a support forum. You can create a bug report if you believe the issue is caused by the TFT.

Nerdcaught, 2023-06-23:
> @Nerdcaught This is not the correct place for your question, please delete your post. This is not a support forum. You can create a bug report if you believe the issue is caused by the TFT.

I thought this would be similar since the video shows the printer pauses at the end while printing?

rondlh, 2023-06-24:
> I thought this would be similar since the video shows the printer pauses at the end while printing?

There is something seriously wrong with your print process, far beyond the scope of the issue discussed here.
Check your print end codes, both in your slicer and TFT.

rondlh, 2023-06-26:
I've been testing V3.1 (BTT TFT35 V3.0) for 3 weeks now, with various parts, and did not encounter any issues, and even better, no hesitations or pauses. I think this is a great improvement. Well done!

V1EngineeringInc, 2023-06-27:
Sorry for the delay, I am back from MRRF and flashing all the printers now!

V1EngineeringInc, 2023-06-27:
While I was there, I got to talk with the CEO and head of the partner company. Some notes were taken we will see if they look into some of the hardware and firmware issues.

[68 more replies]

PR #2827 (2023-08-24): Fixed random pause/hesitation when printing
**BUGFIXES:**
- Fixed #2761: Based on deep analysis and testing shared by @rondlh on #2761
- Improved partial message retrieval handling: `Serial_Get` finction in `SerialConnection` API improved to prevent the possibility to read multiple times the same partial message on next function invokation

fixes #2761 

**PR STATE:** Ready for merge
closed 2023-08-24 by bigtreetech