bigtreetech/BIGTREETECH-TouchScreenFirmware #2769: [Q] No details from the SKR 1.4 turbo on the TFT35 V2 display. RRF.

sssvolk, 2023-04-13:
Board SKR 1.4 turbo vs RRF 3.2
Display TFT35 V2, fw 2.0.25.2

The display is stitched on any firmware, but firmware 2.0.25.2 (mod) - the only one on which the inscription "not printer attached" disappears. When you do this, the data from the printer is not displayed!

On the the board, RRF:
config.g - M575 P1 B57600 S2, parameter S try changed from 0 to 4
board.txt - serial.aux.rxTxPins = { 0.3, 0.2 }

rondlh, 2023-04-14:
I do not understand the question, nor the situation around the above stated "question".

sssvolk, 2023-04-14:
There is a board SKR 1.4 turbo with firmware RRF 3.2. 
The board has a TFT35 V2 display with firmware 2.0.25.2. 
All this is installed on a 3D printer. 
I tried different settings, but the display does not transmit data from the printer. 

p.s.   I translate in google translate.

radek8, 2023-04-15:
Use current FW. no one will deal with you with the old FW.
That the inscription is not displayed. does not mean that the display is compatible with KSR.

sssvolk, 2023-04-16:
On firmware version 2.0.27, the inscription about the absence of contact with the printer is lit. 
On all firmware, I tried to change the data transfer rate - this does not help.

radek8, 2023-04-18:
You must define serial port 0 on the printer
unfortunately I don't use RRF, so I can't help you with RRF configuration.

closed 2023-07-19 by stale[bot]