bigtreetech/BIGTREETECH-TouchScreenFirmware #2771: [FR] G-code override feature for temps and fan speed

rondlh, 2023-04-15:
The TFT allows to dynamically control temperatures and fan speed. This is very useful, but there is a problem. The set values might be changed again in the G-code, like the fan speed ramping up or the print temp being lowered after the first layers. For me it would be useful to fix the set values (as an option). When this option is selected the set value will be enforced by the TFT.

kisslorand, 2023-05-01:
A quick fix would be to delete those g-codes from the file itself (M104, M106, M109, M140, M190, M220, M221).

kisslorand, 2023-05-06:
If TFT overrides temperature related gcodes from the files what would happen with the cooldown commands at the end of a print? You could say to use end gcodes from TFT but what if users want/need use different end gcodes based on their preferences, materials, machines, etc?

If fan speed is locked by the TFT what will happen at bridges, overhangs? If temp is locked what will happen if temp needs to change during print (small areas, different temp for large infills, etc)? You could say to disable the lock of the TFT but what if the user forgot to do so just as you forgot to set the right temperatures/fan speed in your slicer?
So this proposal might fix one single particular situation whilst creating much more other type of possible problems.

It's always better to fix a problem at its source than mitigating the outcome.

Of course these is only my opinion, not an universal truth.

rondlh, 2023-05-07:
> If TFT overrides temperature related gcodes from the files what would happen with the cooldown commands at the end of a print? You could say to use end gcodes from TFT but what if users want/need [...]

All useful input, of course the broader use needs to be considered.
For me to fix the problem means to abort the print, correct the settings in the slicer and print again, that's fine sometimes, but not always.
If you manually change settings then that should have consequences, currently the changes you made might be cancelled within seconds so it's often useless. When you take over control then I hope you know what you are doing. It's true of course that solving your immediate issue might have possible unforeseen consequences. The question is, what is better, having controls that might not do anything or taking control and possibly having unforeseen consequences?

> If TFT overrides temperature related gcodes from the files what would happen with the cooldown commands at the end of a print?

Override only M104, M140 and M141, not "wait commands" M109, M190 and M191. You could use a relative factor which increases/decreases the parameter by a fixed number (temp +10 degrees, fanspeed +25%).
And of course never ignore a command to switch off the heater completely.

>You could say to use end gcodes from TFT but what if users want/need use different end gcodes based on their preferences, materials, machines, etc?

Agree, the end gcodes are not flexible enough.

> If fan speed is locked by the TFT what will happen at bridges, overhangs? If temp is locked what will happen if temp needs to change during print (small areas, different temp for large infills, [...]

Correct, a warning message could be shown to remind the user. Relative values could mitigate most unforeseen consequences. But at any time, I hope the user knows what they are doing.

I can hack a quick implementation and see what works and what doesn't.

kisslorand, 2023-05-07:
> But at any time, I hope the user knows what they are doing.

That should apply first at slicing stage. :)

> If you manually change settings then that should have consequences, currently the changes you made might be cancelled within seconds so it's often useless.

What if the changes are coming after several hours and they would be correct changes but they are overridden?

How would you handle with your method a 30 hours print where fan speed, temperature must change several times?

rondlh, 2023-05-07:
OK

closed 2023-05-07 by rondlh