bigtreetech/BIGTREETECH-TouchScreenFirmware #2773: TFT Error: filament data not found

jivanmp, 2023-04-17:
Hello good
I have updated to the latest version that allows me to put the BTT-TFT on an artillery SW-X2 and it works great, but the other day when I went to print from USB/SD it didn't let me, the print ends as soon as 1 second passes and in the summary of the impression gives me this message:

     filament data not found

The configuration I have is:
System: Marlin 2.1.2 Maniac V1 (Dec 18 2022 19:48:39)
Machine: Artillery Sidewinder x2
Board: MKS_GD_TFT_V4.0
Firmware: V27 .x Mar 28 2023 in 120MHz
SPIFlash: Used 96.78% (7.74MB/8MB)

Thanks

jivanmp, 2023-04-17:
I only drive Cura and when doing the test with prusa if it lets me.

In cure I have tried to enter the codes indicated on the web, but still nothing.

Does anyone know what codes are necessary for it to work with Cura?

radek8, 2023-04-18:
Check in the TFT Menu if you have switched off the filament sensor

jivanmp, 2023-04-19:
I have the filament sensor as activated.

The strange thing is that as I indicate rondlh if I do it from prusaslicer it works.

It is as if in cure we had to add the information that is missing from the filament.

I tried netting the lines that are recommended in the documentation but nothing.

The inclusion of the filament data into the G-code can be automated. In Cura all you have to do is to insert the following into the Start G-Code:

M118 P0 filament_data L:{filament_amount}m
M118 P0 filament_data W:{filament_weight}g
M118 P0 filament_data C:{filament_cost}

rondlh, 2023-04-27:
Are you printing from the TFT or the motherboard SD/USB?

> M118 P0 filament_data L:{filament_amount}m
> M118 P0 filament_data W:{filament_weight}g
> M118 P0 filament_data C:{filament_cost}

This code seems to be working, I tested it in Cura 5.3.0
Make sure to define the weight and cost in Cura under Preferences/Materials
Which generated the following code in the G-code file:
```
M118 P0 filament_data L:[110.17]m
M118 P0 filament_data W:[336.53118812499997]g
M118 P0 filament_data C:[2.355718316875]
```
parceACK.cpp triggers on "filament_data", the check for "L:" (length) or "W:" (weight) or "C:" (costs) and picks up the first valid float behind on the same line.
Either way, this should not affect your printing, unless you printer requires some specific input before it allows you to print.

closed 2023-09-17 by stale[bot]