bigtreetech/BIGTREETECH-TouchScreenFirmware #2774: Display TFT 70 v3.0 not work

salvioni13, 2023-04-17:
I buy new touch panel, but, is not working....

salvioni13, 2023-04-17:
anyone can help me?

radek8, 2023-04-18:
What doesn't work?
Describe your problem exactly.

salvioni13, 2023-04-18:
The panel won't turn on

kisslorand, 2023-04-18:
Can you feed power to it from an external source? I would try to power it on without any connection to the motherboard just to rule out the possibility of  wrong ribbon cable config/connection.

salvioni13, 2023-04-18:
[image]

[image]

I try with charger of my smartphone (5v), not again

oldman4U, 2023-04-18:
Have we removed all the related information from the troubleshooting section!!??

radek8, 2023-04-18:
Save to SD card the FW for your display along with the icon and font folder.
Put the card in the slot on the display and connect the power.
The FW should be uploaded to the display.

If this does not fix the display, you should return the display to the dealer.

There can be several reasons for the error.
- corrupted bootloader,
- damaged LCD panel,
- damaged flexi tape between the LCD and the board,
- defective 3.3V power supply on the board,
- or anything else,

salvioni13, 2023-04-18:
thanks  guys... but not work, i think the screen is damaged :(

closed 2023-04-18 by salvioni13