bigtreetech/BIGTREETECH-TouchScreenFirmware #2779: [BUG] Issuing a reset to Marlin causes issues for the TFT

rondlh, 2023-04-21:
If you issue a Marlin reset command ("KILL" externally or via terminal) then the TFT seems to get disconnected, also the set value like Flow, speed, fan speed etc. stay at the previously set values.

I added the following in parseACK.c so the TFT can survive the motherboard reset:

```
else if (ack_seen("start\n")) { // IRON, ADDED TO SURVIVE MOTHERBOARD RESTART
		infoHost.connected = false;
	    infoHost.wait = false;

```
Still the set values are not reset to their original values.

kisslorand, 2023-04-23:
Can you share please what is the "Marlin reset command" command you're using?
Can you please describe what behaviour do you expect?

rondlh, 2023-04-23:
> Can you share please what is the "Marlin reset command" command you're using? Can you please describe what behaviour do you expect?
The command is "KILL", I hope the TFT can recover from this and reconnect to the display.
See SOFT_RESET_VIA_SERIAL in Marin's Configuration_adv.h

kisslorand, 2023-04-23:
Thanks for the clarification, got it.

It's the same situation I have when using M997. It didn't bothered me enough to spend time on it but since I am not alone in this I took some time to look into it. A found an easy fix for it just need time to make it properly. I guess I'll have a PR ready tomorrow.

If you want, once I have some work done, I can send you the source codes to test it. Four eyes can see more than two. :)

rondlh, 2023-04-24:
> Thanks for the clarification, got it.
> It's the same situation I have when using M997. It didn't bothered me enough to spend time on it but since I am not alone in this I took some time to look into [...]

Great, I will give it a try when available. Please note that my code above has an issue because the PID tuning also uses "start". The next code works for me (parseACK.c):

```
else if (ack_seen(" start\n")) { }
else if (ack_seen("start\n")) { // IRON, ADDED TO SURVIVE MOTHERBOARD RESTART
    infoHost.connected = false;
    infoHost.wait = false;
}

```

kisslorand, 2023-04-24:
Check PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2783

rondlh, 2023-04-25:
> Check PR #2783

**I just tested this code.**
"KILL" via TFT terminal: Works
Reset via motherboard reset button: Works
"KILL" and "^X" via motherboard serial connection: Works
"KILL" and "^X" via Wifi interface: Works

When I say "Works", I mean the TFT detects the restart and connects again.
This is a very useful improvement in my view.
Great job again!

kisslorand, 2023-04-25:
Added "No printer attached!" reminder message whilst the TFT is reconnecting to the motherboard.

rondlh, 2023-04-26:
> Added "No printer attached!" reminder message whilst the TFT is reconnecting to the motherboard.

OK.
I never understood this message, as there obviously is a printer attached, on my own build I replaced the message with "Connecting...".

kisslorand, 2023-04-26:
I never thought about it like that but you're right. :)

Probably "No printer attached!" dates way back when 3D printers didn't have their own display and an external TFT was an extra addon so you had to "attach" the printer to it. I don't know...
Anyway, I just thought a familiar visual sign announcing that something's going on would be welcomed.

rondlh, 2023-04-27:
Great solution created by @kisslorand, waiting for merging...

PR #2783 (2023-05-17): MB Reset Fix
### Requirements

BTT or MKS TFT

### Description

This PR fixes the TFT not communicating with the motherboard after it has ben reset (software reset, external reset, watchdog reset, brown out reset)

### Related Issues

Fixes #2779

### Credit

All credit goes to @rondlh, he found the problem and the solution for it, I just expanded it for more possible situations.

### Notes

~Marked as draft as it is waiting for user feedback~
closed 2023-04-27 by rondlh