bigtreetech/BIGTREETECH-TouchScreenFirmware #2782: GD TFT35 V3 E3 firmware build, UBL mesh edit error

FrozenIceman01, 2023-04-22:
The latest GD TFT 35 V3 E3 build on the repository currently does not allow the editing of any other row than the first row of any of the prior created mesh.

I used the prebuilt firmware as well as built my own version in visual studios with the same result.

Step 1:  Flash firmware via SD Card (config, language. En, firmware, and tft35 icon folder)
Step 2:  Preheat extruder to 240 c and bed to 70 c
Step 3:  UBL mesh start, 9x9
Step 4:  Save mesh in slot 0 and 1
Step 5:  Open Mesh edit
Step 6:  attempt to select mesh point on any row below row 1 (fail)

Prior version circa 3 months ago allowed allowed for mesh edit.  However it would periodically throw an m error (14 or 48 I believe) and wouldn't allow to adjust the jesh point any farther.  Solution would be to save the mesh, power cycle printer, and continue mesh edit until it fails again.

digant73, 2023-04-23:
Fixed by #2781

closed 2023-04-27 by bigtreetech

FrozenIceman01, 2023-04-30:
Latest build tested, issues not resolved.

First issue which is the row column issue is resolved.

Second issue that throws m11 or m14 errors and not allowing mesh adjustment any further is not resolved.

Please re-open.