bigtreetech/BIGTREETECH-TouchScreenFirmware #2785: [FR] Soft endstop enable/disable button Request

sarvenn, 2023-04-29:
In Marlin mode, I can enable and disable soft endstops. It would be nice to have it under movement menu also, nearby disable steppers.

kisslorand, 2023-04-29:
You can always have them with custom commands.

sarvenn, 2023-05-01:
Hi @kisslorand. Yeah, I know. I can also use terminal to send M211 S0/1 commands manually. This is what I do currently.
But I would like to recommend this as a new functional button in TFT mode just like Marlin mode (under move menu).
I'm not a developer but I think hardest part to implement this suggestion is to create a new icon for this button. Doesn't it worth to keep this in the backlog? Usage frequency might not be high, I'm also aware of that. Just wanted to suggest a feature to fill the gap between marlin and tft modes.

kisslorand, 2023-05-01:
> Hi @kisslorand. Yeah, I know. I can also use terminal to send M211 S0/1 commands manually. This is what I do currently. But I would like to recommend this as a new functional button in TFT mode just [...]

The SPI flash of the TFT is almost full, it's at 96.78% of its capacity. I know that not used memory is wasted memory, but I still suggest to keep a little headroom. I also use custom commands for not so frequent tasks.
Of course this is only my opinion, I am not entitled to impose anything.

> Just wanted to suggest a feature to fill the gap between marlin and tft modes.

Was this ever a goal?

sarvenn, 2023-05-01:
Flash size probably would be needed for another important feature. So it is better to go with custom menu.

closed 2023-05-01 by sarvenn