bigtreetech/BIGTREETECH-TouchScreenFirmware #2786: [BUG] (TFT35 Mesh Edit throws M14 and unable to adjust Mesh

FrozenIceman01, 2023-04-30:
### Description

The latest GD TFT 35 V3 E3 build on the repository throws an m11 or M14 error after mesh adjustment and results in inability to adjust mesh further requiring a power cycle.  In addition after the UBL is complete it locks up the display and requires a reset before it will accept commands again.  This is the second half of the previous closed issue posted here"

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2782

Step 1: Flash firmware via SD Card (config, language. En, firmware, and tft35 icon folder)
Step 2: Preheat extruder to 240 c and bed to 70 c
Step 3: UBL mesh start, 9x9
Step 4: Save mesh in slot 0 and 1
Step 5: Open Mesh edit
Step 6: Select Mesh Point to travel to (Fail)
Step 7: Power Cycle Machine
Step 8: Preheat extruder to 240 c and bed to 70 c
Step 9: Home 
Step 10: Disable UBL
Step 11: Load Slot 0 or 1
Step 12: Select Mesh points to verify and Adjust
Step 13: Repeat Step 12 Mesh point adjustment until TFT 35 throws an M11 or M14 error
Step 14: Attempt to adjust mesh point (Fail)
Step 15: Power Cycle Printer
Step 16: Repeat Steps 8 through 15 until all mesh points verified. 

**Expected behavior**
1.  Able to adjust Mesh Points after UBL Mesh Start
2. Able to adjust all Mesh Points without throwing an m11/M14 error resulting in a power cycle.

**Actual behavior**
TFT35 is unable to adjust the Mesh Point, Menu is NOT frozen.  It just won't allow you to change the mesh point numeric value.

### Hardware Variant
TFT35 V3 E3 connected to a SKR Mini E3
SKR Mini E3 V3
### TFT Firmware Version & Main Board Firmware details
BIGTREE_GD_TFT35_V3.0_E3.27.x.bin
Marlin-2.1.2 

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11360405/Marlin.zip)
[20230429 TFT Install.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11360406/20230429.TFT.Install.zip)

kisslorand, 2023-04-30:
Please check [this](https://mega.nz/file/rhICiDLD#TeNVVHJlfJCgjHgv1EWSLMdkKEubitoE5rihlAo-_HI) file if it fixes your issue with M11 and M14. Also please check the navigation between mesh points (with the controls on the screen itself and the rotary encoder too) in MeshEditor menu as the FW in the link is based on PR #2776 which restores (broken by  PR #2781) the circular column navigation in "MeshEditor" menu when using the screen touch controls (left, right), restoring it without affecting the rotary encoder's linear navigation style.

Also please describe what do you mean by "after the UBL is complete". Please describe the steps how the TFT locks up so I can reproduce it and eventually fix that.
Thx

FrozenIceman01, 2023-05-04:
> Did this occur with the file provided?
> the adjust point interface is incredibly slow
> Can you please explain this? What is slow exactly? How is the interface
> "slow"? It became "slow" only with the file provided?

I did not rerun the ubl mesh builder on your firmware.  I loaded the prior
mesh I adjusted

The slow part was where the t35 wheel was being used to adjust each
individual point up or down.  In the prior build each . 010 adjustment took
around 100 milliseconds to complete before I could increment to the next
.010 position.

The new build takes about a second or two for every adjustment step before
I can select the next adjustment step with the adjuster wheel.

kisslorand, 2023-05-05:
Please test [this](https://mega.nz/file/n1x3kKjY#odPsRuJZdsrGgTD_QaEh9fWRdhC8aVJeF88-0De8i0I) version (v2). It's even more snappier than the original (master). Check if exhibits the M11, M14 issue.
Also would be nice if you got some answers for earlier questions too.
Thx

FrozenIceman01, 2023-05-06:
Hello Kisslorand,

That V2 is fantastic!  The responsiveness to the step adjustment is great!

For the other questions:
"after the UBL is complete" - This was after completion of Mesh Generation Step 5 (When you populate the final mesh points
"Does this still occur with the file provided?" - I just reran UBL and adjusted all of the bed points, it did not lock up at all.  Nice work!
"Can you please explain this? What is slow exactly? How is the interface "slow" - This was the mesh adjustment steps, only adjust the steps, Your V2 solved the issue
" It became "slow" only with the file provided" - It became slow with V1, your V2 fixed it

New Bug:
1.  Mesh Generation  Issue: On Generating Mesh Step 4, as it was going through the mesh generation, it would periodically echo bu-1.13, see image 1.  It did not halt the mesh generation 
[image]
2.  When Setting Z Probe distance it would throw a "????" error after I hit back.  I troubleshooted this several times and it appears to consistently throw it if I hit "Mesh Edit" before the home move completes after I hit back in the Z Probe menu.  Order of Operations: Select Z Probe off (to get into Z Probe Mode), Then adjust Z Probe (First in .1 then in .01 incriments), with the menu save mesh, then hit back, after hitting back select mesh edit.  Once it throws this issue I need to power cycle printer, preheat, align, and restart Z Probe  See image 2.
[image]

Feature Request:
Add Auto Z Align somewhere on one of the movement screens.  This is a Marlin Feature that auto indexes the Z axis like the Prussa winters do with dual stepper single drivers.  I needed to go to Marlin to use the Auto Z Align feature (or send a G34 command in the text menu).

Overall, very nice work tracking this issue down!

FrozenIceman01, 2023-05-06:
Follow up on the speed issue.

After resetting the speed to 100% on the marlin side, the next print on the
touch screen side was set back to 10% speed.

kisslorand, 2023-05-07:
> 4\. Travel Speed, the new TFT firmware slowed down the travel speed to what appears to be 10%.  It travels at normal speed when when homing but it does appear to be an issue when moving the X,Y, or Z positions through the movement display.  I also verified this while printing too.  It would not let me increase the speed from 10% to 100%.  I had to change the setting on the Marlin interface side of the TFT.

Please try to be more precise in your descriptions. I have no idea what is "movement display".  What settings did you change on the Marlin interface? When you print, the XY travels are slower? What did you verified exactly while printing?

I checked the movement of the axes in Menu -> Movement -> Move menu and I cannot replicate any speed inconsistency. The movement speed there is according to what is set in the "config.ini" for Slow, Normal and Fast speeds and how it is set in Menu->Settings->Feature->MoveSpeed(X Y Z). If I set it to "Slow" the movements are slow, if I set it to "Normal" the movements are at normal speed and if I set it to "Fast" the movements are fast all three options respecting the values set in "config.ini".

FrozenIceman01, 2023-05-07:
Yes, the X, Y, and Z traveled slower while printing as well as manually
moving in the move menu

TFT move menu
Movement -> movement -> set travel to 10mm -> + or - x, y, or z button

While printing in Touch Mode:

Options ->percent->speed-> increase to 100%

While printing in Marlin mode:

Tune -> Speed -> 100%

kisslorand, 2023-05-07:
Please test [this](https://mega.nz/file/n5JlnRzZ#fxrdNhQqljbfzCq_kvjNYIdKKAJ2h4sg8uJlrDdba_0) version (v3). Most probably the "?????" issue is solved.

Can you please check with M220 to see if it's the source of the slow speed issue?

FrozenIceman01, 2023-05-15:
I am still testing firmware, I was running into an issue with the last firmware where I was starting to see a Z Axis shift on long prints greater than 3 hours.  I originally thought the issue was hardware or z offset side/fade and recrimped all my connectors, adjusted offset, and set fade to 0 without much change..  

It ends up that it is TFT35 firmware side as when I switched to Marlin it didn't have any issues.  I swapped back to TFT side and I verified that it wasn't a Z axis collision by pulling the filament out of the hot end half way through the print and saw the Z axis shift about 30 minutes later.  I did reload the g code and verified that the error in the Z shift wasn't in the code, see attached.

I also uploaded your latest firmware tonight, I am still checking it out (including Z offset), however it did provide a list of unknown command error M2 and M error messages during a shorter print.  See attached zip file.  The speed change issue appears to be gone.

[AA8_Toolhead Box Lid.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11474207/AA8_Toolhead.Box.Lid.zip)

Now that I know it isn't my machine that is having issues, I'll continue to test your firmware.

FrozenIceman01, 2023-05-15:
I just tested the z probe ??? It is resolved.  Nice work!

FrozenIceman01, 2023-05-15:
> <bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2786/1537499315@
> github.com>

Follow up, I have confirmed the speed setting reverts to 10% after saving z
offset.

kisslorand, 2023-05-16:
@FrozenIceman01

Do you have Y axis shift with "master" WF also?
I cannot reproduce Y axis shift, speed decrease, and involuntary "M" and "M2" messages. In fact I cannot reproduce any involuntary messages. I can only rely on your testing. What is the connection speed between the TFT and motherboard? Do you have Y axis shift only when printing in TFT mode and from TFT SD Card or also when printing in TFT mode and onboard SD Card too?

In the meantime please check [this](https://mega.nz/file/aggXgCQC#eO6O-TJRJBF3F0St3XhJcv7Z3BgT9HtGFZX2dsBHaBk) file (v4). It has improvements to v3 and also includes a rework based in the idea of PR #2788, a needed rework because that PR is only a partial fix whilst introducing a new bug.

FrozenIceman01, 2023-05-16:
Master WF?  I do not understand.

I am continuing work on the axis shift, it may be tied to additional
vibration from a higher torque motor I installed causing the tensioner to
loosen.  If I can adjust it out I'll let you know.

I am using 250000 for my baud rate

I am printing off of an SD card mounted on my skr mini e3 v3 board.  I am
not using the tft 35 SD slot.

Sure, I'll give v4 a shot.

Thank you,
Andrew

FrozenIceman01, 2023-05-16:
> @FrozenIceman01 <https://github.com/FrozenIceman01>
>
> Master WF? I do not understand.
>
> It was a typo, I meant "Master FW (firmware)".

Yes, I was seeing several error codes thrown before we started to test your
firmware.

Your firmware has reduced quite a few of the error messages sent.

kisslorand, 2023-05-16:
@FrozenIceman01 

When did the speed reduction started? With what FW (firmware)?

BTW, for a quicker/easier conversation you can join a [Discord channel](https://discord.gg/dM5wqktj) I created for BTT TFT FW discussions.

FrozenIceman01, 2023-07-19:
> Dear stale bot, let's wait a little bit longer, maybe @FrozenIceman01
> <https://github.com/FrozenIceman01> will eventually help us to help him.

I am still here, sorry about that.  Ran into issue while upgrading to E3D
0.9 degree steppers.  This may have contributed to the layer shift problem.

It ends up that stealth shop's pulse width is apparently not tuned for 0.9
degree steppers.  I confirmed this deficiency with Bigtree.  It ends up
bigtree and their main boards does not support 0.9 steppers nor have they
experimented with it or provided any guidance for tuning the pulse
parameter.

My solution was to switch to Spread Cycle and the layer shift problem went
away.  My printer is back up and running as of about two weeks ago.

Do you still need me to test this flavor of the tft35 interface?  It looked
like there were a few firmware updates while I worked through the 0.9
degree stepper problem.

kisslorand, 2023-07-19:
You previously mentioned that you have a speed reduction issue (from 100% to 10%). It was never elucidated if it is inherited from the master or it is a new bug introduced by my builds. Because of this I never released a PR with the fixes provided to you, because I cannot replicate any of your issues so I am only can rely on your feedbacks.
Yes, there were some updates to the master since I provided the last version of the fixes (v4) but nothing major that would concern any of the issues discussed in this topic. If you think it is necessary I can build a new version of the fix to include the latest state of the master.

All I would like to know if the slowing down issue is only present with the builds I provided here or is it also present in the master too.

FrozenIceman01, 2023-09-03:
Hello @kisslorand, I managed to take some time this weekend and test the settings.  The first is that I believe the speed issue is related to your V4 branch.  

V4

I first adjusted the H offset using the Tuning screen below.  I did adjust the increment a few steps 
from .01 to .1.  It is possible that selecting these changes the speed settings.
[image]

I then began to print a Plunger test print for PETG.  Note this includes Z axis gantry align, 4 point align, and Unified Bed Level
[AA8_Plunger.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/12505015/AA8_Plunger.zip)

As the print was warming up the speed indicator showed 203% in your firmware
[image]

I then started to see some odd commands being thrown.

[image]
[image]

And the Print failed.

I loaded the most recent Master BIGTREE_GD_TFT35_V3.0_E3.27.x.bin from 8/23 and tried again.  Same H offset from the steps above and same print part.

This time it set the Speed to 100%

[image]

It also threw this two error codes and failed to print.
[image]

[image]

Worked fine on the Marlin Side.

PR #2873 (2024-01-15): cumulative patches + cleanup
**IMPROVEMENTS:**
- cumulative patches/improvements: it includes some fixes/improvements provided up to PR #2844. They have been refined/fixed here and there to be fully working.  All credits to kisslorand
- cumulative cleanup: cleanup on different files just to remove from _.h_ files useless references to functions invoked only internally on related _.c_ files. Order of functions have been changed in _.c_ files just to make cleanup on related _.h_ files. _inline_ function qualifier used when needed to reduce code size
- other minor improvements/cleanup

**BUGFIXES [... cut]
closed 2024-01-15 by bigtreetech