bigtreetech/BIGTREETECH-TouchScreenFirmware #2787: Problems with TFT35 Z-Offset Wizard with endstops

SuperSpooner, 2023-05-02:
I have swapped my old mainboard for an SKR3 with TMC2209 drivers and a genuine BLTouch. I paired this with a TFT35 V3.0. I compiled Marlin 2.1.2 myself from original source.

My printer uses an endstop switch at the top so homes upwards. The switch is wired into Z endstop plug and the probe is wired into the probe plug. I have configured Marlin to not use probe as z endstop

This Marlin config works as it did before I installed the TFT35. Probing, levelling and offset work as expected in Marlin mode, but in touch mode I am getting some odd behavior when trying to use the Probe offset wizard.

From the touchscreen, ABL levlelling works - homes up, probes down in the correct places and saves mesh to EEPROM. When I start/turn on the probe offset, it homes, then instead of coming down, it tried to go further up, crashing into top swicth.

Like I say, this config worked before the SKR3/TFT35 and it all works in Marlin mode, but cannot get probe offset to work in touch mode. As the problem appears to be with the TFT35, I presume it is an issue in the config.ini or a communication issue with Marlin.

I am using the very latest TFT35 firmware V3.0.27.x - as of 27th April 2023
I have tried changing the "inverted_axis" and "probing_z_offset" settings, but keeps trying to go up after homing.
I asked the same question on Reddit and got no response, but there are other questions about the use of the Z-Offset feature when using endstops.

Is this a bug in TFT35 firmware or is there anything someone can think of I can change to get P-Offset to move down in touch mode?

Thanks

radek8, 2023-05-02:
Looks like a bad Marlin configuration.
TFT35 V3.0 also supports simulated Marlin mode.
If you run Marlin mode with the current configuration, are the movements in Marlin OK?

SuperSpooner, 2023-05-02:
Yes, as mentioned homing and probing in Marlin  mode work as expected - all homing and BLTouch moves are in the correct direction.
In touch mode, the BLTouch probes the bed correctly also.
It is only when the offset wizard is started from touch mode that it moves the wrong way after homing.
The "inverted_axis" option is working (the direction/icons are changing) but not having an effect on the direction.

radek8, 2023-05-02:
Does your printer raise the bed or lower the nozzle?

SuperSpooner, 2023-05-03:
My printer lowers the nozzle
Z endstop at top, so homes up and probes down
Im clutching at straws trying the "inverted_axis" option to be honest, I didnt expect it to work.

Its just odd that offset wizard doesnt work in touch mode when the ABL probing does; it homes up, then moves down to probe bed 9 places and saves to EEPROM.

To be clear, the ONLY error I am seeing is when starting offset wizard from touch mode; after homing up correctly, the gantry tries to move up instead of down - crashing into endstop switch.

radek8, 2023-05-03:
Send a video

kisslorand, 2023-05-03:
@SuperSpooner

Set `probing_z_raise` in "config.ini" to 0 (zero) or to a negative value.

oldman4U, 2023-05-04:
@SuperSpooner

This is open source. So you can add what you learned to the ReadMe document which is available for this repo.

Thank you

closed 2023-05-04 by SuperSpooner