bigtreetech/BIGTREETECH-TouchScreenFirmware #2789: [BUG] IDEX Incorrect extrude low temp warning and target temp change

rondlh, 2023-05-09:
### Description
For an IDEX system, when extruding from E1 (T1) while E1 is at the required temp, a warning is generated that seems to be referring to the E0 temp. 
When selecting OK in the warning popup, the E1 target temperature is changed to a value that is lower than the value I previously set.

### Steps to reproduce

1. IDEX system, heat T1/E1 to the required temp (200 degrees C)
2. Go to Extruder menu, select E1 and wait until E1 has reached the target temperature (check persistent temp on top)
3. Press Load or Unload
4. A warning message pops up to warn that the temperature is too low, but this message is clearly referring to the E0 temp
5. If I click "OK" then the target temperature of E1 is changed
6. It seems that if I go out of the Extruder menu and go back in that now things seem to be working correctly (no warning when selected extruder temp is ok) 

**Expected behavior**
No low temp warning when selected extruder is at the correct temp.

**Actual behavior**
Incorrect warning, and incorrect target temperature change.

### Hardware Variant
BTT TFT35 V3.0

### TFT Firmware Version & Main Board Firmware details
Latest (2023-05-03)

### Additional Information
I use a low required temp for the extruder of 40 degrees, so I get a reminder, but no need to heat up the nozzle when, for example, loading new filament.

digant73, 2023-05-11:
Fix reported in #2790. Thanks for your work

rondlh, 2023-05-12:
Tested, it solves the issue, great job again!

PR #2790 (2023-05-17): bug fixes on Mesh Editor, Move and on IDEX Incorrect extrude low temp warning (#2789)
**BUG FIXES:**
* broken "undo/restore" button on Mesh Editor menu: another bug introduced by #2737. In case a mesh value is changed, the undo/restore button doesn't restore the original value on Marlin side
* broken release of memory exiting from Mesh Editor: another bug introduced by #2737. It causes invalid access to mesh data already released (thus possible crashes/freezes/wrong TFT behavior etc...) and that visible delay when closing the menu (e.g. 1-2 seconds on less powerful TFTs)
* wrong move distance  [... cut]
closed 2023-05-12 by rondlh