bigtreetech/BIGTREETECH-TouchScreenFirmware #2796: [BUG] BTT GD TFT35 and SKR mini e3 v3 on Ender3 v2 freezes while printing

jesseab, 2023-05-23:
I saw this thread :  
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2530

I have the same problem. When printing from usb (and maybe sd too) from the TFT, it randomly stops printing, and the TFT freezes. The motherboard still respond to serial commands if I plug the usb.

Was it ever solved ? I updated the firmware of the TFT to the latest version as well.
I am using a TFT35 V3.0 E3

jesseab, 2023-05-24:
Thanks for the answer, my printer doesn't hesitate tho, the screen completely freezes and the print stops, however the motherboard still communicate so it's really a TFT problem, however I will try to increase the buffer sizes as suggested !

rondlh, 2023-06-14:
Do a disk check on your USB drive, make sure it doesn't have errors. You can also try another USB drive.

PR #2901 (2024-02-28): Bugfixes on top of last merged PR
**BUGFIXES:**
- Fixed #2895: Bug introduced by #2889 and affecting TFT variants with a LED on knob (e.g. TFT35 E3 V3.0 etc.) in case `knob_led_idle` setting on `config.ini` file was enabled. The bug caused a freeze exiting from PID/MPC menus or during the `Event LED` process, if enabled, at the beginning of a print. The bug was due to the missing `volatile` qualifier on `ms` attribute of OS_COUNTER data struct.
  **NOTE:** The bug severity can be considered major for the affected TFT variants
- Fixed #2898: Bug causing a freeze entering on `Preheat` menu. The  [... cut]
closed 2024-02-28 by bigtreetech