bigtreetech/BIGTREETECH-TouchScreenFirmware #2802: [BUG] Not flashing?

IoakeimSogiakas, 2023-06-01:
I am helping some configure there printer up. They are using an skr v3. I have set up the board correctly (at least to my knowledge if needed I can sent a configuration) but the screen is not working. The screen does function like normal in marlin mode but in tft mode it doesn't connect. At first I thought it might be an issue with the serial interface or the baudrate but I have tried everything and set it up to speck (the serial stuff 😂) also I tried both 115200 & 250000. Nothing worked. So I did a test because I had a suspicion on the flashing. I changed the marlin interface to full-screen (it was not in full screen before) and also the color of the letters to red. After power on the screen does the usual flashing sequence (I have used this screen before many times with many people) but the changes that I made (I made a specific bin file not consumed ini file) where not present. Like nothing changed. Am I missing something. Also don't know if this is relevant but. When the person first came to my for help the screen was blank (white) so I requested him to try the reset.txt file which resulted in a illegal user app error (fist time seeing it)

IoakeimSogiakas, 2023-06-02:
That's a good point let me check and contact you back

oldman4U, 2023-06-12:
What is the status of this ticket. Please help the community and close the ticket once you do not need it anymore. Thank you

IoakeimSogiakas, 2023-06-12:
Sorry the guy I was helping hasn't responded yet I am going to close the ticket and reopen it as soon as I get a response

closed 2023-06-12 by IoakeimSogiakas