bigtreetech/BIGTREETECH-TouchScreenFirmware #2803: Icons not updating

ryguyy46, 2023-06-05:
I followed the instructions multiple times to update. I made sure the SD card is formatted in FAT32, its not bigger than 8GB. all the icons are in the bmp file. the TFT file has the .CUR at the end after I put it in the machine and turn it on, everything updates except the icons. not sure what else to try or what's going on.

ryguyy46, 2023-06-05:
[image]

ryguyy46, 2023-06-05:
[image]

kisslorand, 2023-06-05:
Rename "TFT35.CUR" folder to "TFT35" and make sure you have the latest set of icons in there.

closed 2023-06-05 by ryguyy46