bigtreetech/BIGTREETECH-TouchScreenFirmware #2808: Marlin mode glitching out

IoakeimSogiakas, 2023-06-20:
I am using a normal tft 35(gd version) the issue is this when doing filament load /unload it does this https://youtube.com/shorts/qAIealqtr9k?feature=share
Also in marlin mode it does this for some reason https://youtube.com/shorts/IROpqpYHfAc?feature=share
Everything is connected correctly (the board used is an skr ez. And flashing also is functional (the files are getting renamed)

rondlh, 2023-06-20:
You will need to provide much more information to be able to answer this question, have a look at the template...

IoakeimSogiakas, 2023-06-20:
Sorry my bad the issue fixed it self. I was just trying to find a way to delete this issue

rondlh, 2023-06-21:
Great, then please close the issue

oldman4U, 2023-06-22:
Please help the community and close the ticket once you do not need it anymore. Thank you

IoakeimSogiakas, 2023-06-22:
Sorry I thought I closed it

closed 2023-06-22 by IoakeimSogiakas