bigtreetech/BIGTREETECH-TouchScreenFirmware #2810: [Q] (TFT70 V3.0 & SKR 3 EZ issue where only marlin mode works)

Bradford1040, 2023-06-25:
I bought them separate from each other, after I couldn't get my Ender 3 V2 screen to work with the SKR 3 EZ

using marlin bugfix 2.1 for STM32H723VG_btt & the newest firmware from github for the TFT70 (GD) variant
I have all 3 ribbon cables plugged in correctly and also checked the pinout for the 232 cable, now the touch screen mode shows its connected and displays temperatures from thermistors but I can't move axis or make any changes while in TFT mode, only if I switch to marlin mode will it work (make temperature changes & move axis)

I've asked quite a few people to help including the Marlin Discord, shared my configuration.h & h.adv as well as the config.ini for my screen. Others have said it all looks good, so what am I doing wrong?

radek8, 2023-06-25:
Similar problem as here
https://github.com/bigtreetech/BIGTREETECH-TFT24-V1.1/issues/41

Bradford1040, 2023-06-25:
I think I just solved my issue, I did a `reset.txt` & it seems to work, at least enough for me to continue working forward to completing this install, 

My only other issue is getting the led knob to change color with the heated bed

oldman4U, 2023-06-29:
Please help the community and close the ticket once you do not need it anymore. Thank you

radek8, 2023-07-07:
This is not how the ticket closes :-(

closed 2023-09-26 by bigtreetech