bigtreetech/BIGTREETECH-TouchScreenFirmware #2811: BTT SKR MINI E3 V3 AND TFT35 E3 V3.01 TOUCH MODE CONSTANTLY RESETS AND 'NO PRINTER ATTACHED' MESSAGE IS CONTINUOUS.

Jhendrix0875, 2023-06-30:
I HAVE AN ENDER 3 PRO AND I JUST REPLACED THE MOTHERBOARD AND THE SCREEN. I UPDATED THE FIRMWARE TO THE LATEST ON THE TFT DEVICE. I HAVE CHECKED 50 TIMES TO MAKE SURE THAT THE SERIAL CABLE IS PLUGGED IN CORRECTLY, AND I HAVE VERIFIED THE BAUD RATE WITH THE PRINTER AND THE TFT ARE THE SAME. IT WORKS IN MARLIN MODE AND I AM ABLE TO CONTROL THE PRINTER THAT WAY. HOW DO I GET THE TOUCH MODE TO WORK?

radek8, 2023-07-01:
Did you use the last stable marlin, or the last bugfix marlin?
If the latter, restarting Marlin may be related to this issue.
https://github.com/MarlinFirmware/Marlin/issues/26030

PS: Do you also have a problem with the keyboard, CapsLock stuck?

Jhendrix0875, 2023-07-02:
> Did you use the last stable Marlin, or the last bugfix Marlin? If the latter, restarting Marlin may be related to this issue. [MarlinFirmware/Marlin#26030](https://github.com/MarlinFirmware/Marlin/issues/26030)
> 
> PS: Do you also have a problem with the keyboard, CapsLock stuck?
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/11931264/Configuration.zip)

Here are my config files. 
No, CAPSLOCK is not stuck. I work in all caps at my work, and it bled over here.....

radek8, 2023-07-02:
There is nothing in your configuration that prevents the touch mode of the display from working.

You did not answer which version of Marlin you are compiling.

Jhendrix0875, 2023-07-03:
Latest bugfix version. I believe it is Marlin-bugfix-2.1.x

radek8, 2023-07-04:
A patch was merged in Marlin 2 days ago. 
Can you compile a new Marlin with this fix?

Jhendrix0875, 2023-07-04:
I deleted all of the files from my computer and started from scratch. TFT is no working.

Jhendrix0875, 2023-07-04:
I deleted all of the files from my computer and started from scratch. TFT is now working.

closed 2023-07-04 by Jhendrix0875

devinsheppard, 2023-09-23:
What did you end up doing to get the screen to work. I'm having the same issue but I'm running Klipper and not Marlin?