bigtreetech/BIGTREETECH-TouchScreenFirmware #2813: TFT43 installing firmware stm32f4 board not working z axis stepper motors

AliImrangang, 2023-07-15:
Description
After installing the display firmware on my TFT43, the Z-axis is not functioning properly.

Steps to reproduce
Installed the new display firmware on the TFT43.
After installation, testing the printer's movement.
Noticed that the Z-axis was not responding.
Expected behavior

The Z-axis should have responded to commands and moved as per instructions.

Actual behavior

The Z-axis is not moving at all. All other axes (X and Y) are working as expected.

Hardware Variant
TFT43

rondlh, 2023-07-16:
Are you using the move menu? Does it work when you use the TFT terminal? (G0 Z1)

AliImrangang, 2023-07-16:
Yes I am using move menu , ok now I am using g code then I will let you
know if any problem
Thanks

AliImrangang, 2023-07-17:
> <bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2813/1637133871@
> github.com>

yes now I run above command in the terminal my z movement is working
through movement manu can I configure by default when tft43 is on and i go
to movement menu z is automatically working.

closed 2024-03-15 by github-actions[bot]