bigtreetech/BIGTREETECH-TouchScreenFirmware #2814: esp-01 integration failure

NiKoLo2002, 2023-07-19:
**Your question or details you need. Please describe.**
A clear and concise question or detail you would like to ask. Like Hardware related or firmware related.

**specify the device variant related to your question.**

**Additional context**
Add any other context or screenshots about the question here.

JLiraEJ8, 2023-09-16:
I have the same problem. flash with esp3d.bin, everything ok, I put it on the tft35 e3 v3.0.1 but it doesn't appear on the wifi network or the IP.

digant73, 2024-01-10:
the esp-01 communicates with the TFT with serial port. The esp-01's IP you configured is just to reach the esp-01 from the WiFi network.
On TFT side you only need to set the serial port where you plugged in the esp-01 with the baudrate you configured on esp-01. E.g . from `Menu->Settings->Connection->S. Port` menu set the WiFi port to 250000 if your esp-01 was configured to 250000

PR #2873 (2024-01-15): cumulative patches + cleanup
**IMPROVEMENTS:**
- cumulative patches/improvements: it includes some fixes/improvements provided up to PR #2844. They have been refined/fixed here and there to be fully working.  All credits to kisslorand
- cumulative cleanup: cleanup on different files just to remove from _.h_ files useless references to functions invoked only internally on related _.c_ files. Order of functions have been changed in _.c_ files just to make cleanup on related _.h_ files. _inline_ function qualifier used when needed to reduce code size
- other minor improvements/cleanup

**BUGFIXES [... cut]
closed 2024-01-15 by bigtreetech