bigtreetech/BIGTREETECH-TouchScreenFirmware #2815: [BUG] (short description)

blazejkarwowski, 2023-07-21:
### Description

Probing corners using UBL "L Corner" 

### Steps to reproduce

1. Update TFT35 Firmware with bin.
2. Go to L corners via Movement -> Bed Level -> UBL -> L Corner.
3. Select any corner
4. Bed gets probed
5. No result showed on screen

**Expected behavior**
Bed height should be displayed

**Actual behavior**
Nothing is changing on the screen

### Hardware Variant
BTT TFT35 E3 V3.0
### TFT Firmware Version & Main Board Firmware details
[BIGTREE_TFT35_V3.0_E3.27.x.bin](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_TFT35_V3.0_E3.27.x.bin) Marlin bugfix-2.1.x

### Additional Information
Config of printer https://pastebin.com/YdqK0FMG
Working video https://youtu.be/lQHiDFrEKaw

kisslorand, 2023-07-22:
I tested but cannot reproduce the issue you have. Please make sure you have the Marlin dependencies set as instructed here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies

Note: the "SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)" is not needed, it's obsolete.

thephantom1492, 2023-08-12:
TFT70 here.

This is a "regression" in marlin code.
They killed the SERIAL_FLOAT_PRECISION, and fixed it to 2 digits.
This break the functionality.

Marlin nightly of 2023-05-14 is working fine. Not sure when they actually broke the code, but possibly commit ed4e22b of 2023-06-04.

kisslorand, 2023-08-12:
They just reverted that change, SERIAL_FLOAT_PRECISION is back.

thephantom1492, 2023-08-12:
I do not see any sign of reversal. The last commit is 5 days ago. All I see is a bug report.

kisslorand, 2023-08-13:
Yes, you're right, I remembered wrong.

kisslorand, 2023-08-23:
I made a fix for it but for 2 weeks I am unable to make any more PR.

PR #2873 (2024-01-15): cumulative patches + cleanup
**IMPROVEMENTS:**
- cumulative patches/improvements: it includes some fixes/improvements provided up to PR #2844. They have been refined/fixed here and there to be fully working.  All credits to kisslorand
- cumulative cleanup: cleanup on different files just to remove from _.h_ files useless references to functions invoked only internally on related _.c_ files. Order of functions have been changed in _.c_ files just to make cleanup on related _.h_ files. _inline_ function qualifier used when needed to reduce code size
- other minor improvements/cleanup

**BUGFIXES [... cut]
closed 2024-01-15 by bigtreetech