bigtreetech/BIGTREETECH-TouchScreenFirmware #2816: [Q] (Manual moving speed to Z axis ?)

SubJORJ, 2023-07-22:
Hi !
What speed command is sent to the 3D printer's controller when the Z-axis is manually controlled? 
I have an MKS TFT35 v1.0 and it sends an F3000 speed command to my 3D printer, when I manually control the z-axis movement. My printer almost jumps at that speed.
If I burn "BIGTREETECH-TouchScreenFirmware" into my MKS TFT what is the movement speed of the Z-axis with manual movement

**specify the device variant related to your question.**
MKS TFT35 v1.0

**Additional context**
Add any other context or screenshots about the question here.

kisslorand, 2023-07-22:
In the "config.ini" you have the following section:
```
#### Z Speeds/Feedrates
# Move speeds used in Move menu to move Z axis.
#   Format: [move_speed: S<feedrate> N<feedrate> F<feedrate>]
#   Unit: [feedrate in mm/min]
#   Value range: [min: 10, max: 12000]
z_speed:S500 N1000 F2000
```

Where 'S' stands for "slow", 'N' stands for "normal" and 'F' stands for "fast". In the menu (Menu -> Setting -> Feature -> Move speed (X Y Z)) you can select whether you want the moves to be executed at slow, normal or fast speed.

I hope this helps.

SubJORJ, 2023-07-22:
[kisslorand](https://github.com/kisslorand) Thank you very much for your response.
I asked this question about speed in MKS TFT (https://github.com/makerbase-mks/MKS-TFT/issues), but apparently there is a dead thread and there are no answers.

I will switch to "BIGTREETECH-TouchScreenFirmware"

closed 2023-09-26 by bigtreetech