bigtreetech/BIGTREETECH-TouchScreenFirmware #2819: [FR] ADVANCED_OK support, base code implemented already

rondlh, 2023-07-22:
Marlin supports ADVANCED_OK, which returns info about the available buffers when acknowledging a command with "ok".
This is currently not used by the TFT, which sends 1 command and waits for the "ok" response before sending the next command.
I suspect that this is the reason for some random pauses and hesitations when printing #2761

Marlin support BUFFER_MONITORING, which can give you information about the buffer underruns, a buffer underrun occurs when Marlin is ready to process the next command, but no command is available yet. I found **thousands** of buffer underruns per hour. These underruns are likely to be of very short duration, but clearly that is not what you want. Some buffer underruns are normal and unavoidable because the TFT is holding back data during the heating phases, this accounts for about 100 underruns in my case.

I made a rudimentary support of ADVANCED_OK which works as follows:

1. When you have ADVANCED_OK enabled in Marlin, then Marlin responds to commands like this "ok P31 B15". The B-parameter indicates the amount of available buffers.
In parseACK.c the "ok" acknowledgement is processed. I added a check for the "B" parameter. The parameter value is stored in the HOST struct (defined in main.h). 

```
    // "ok" response handling
    if (ack_starts_with("ok"))  {
      infoHost.wait = false;
      // if regular "ok\n" response or ADVANCED_OK response (Marlin) (e.g. "ok N10 P15 B3\n")
      if (ack_cache[ack_index] == '\n')
        goto parse_end;  // there's nothing else to check for  
	  if (ack_continue_seen("B")) 
        infoHost.AvailableBuffers = ack_value();
	  infoHost.AvailableBuffers = MAX(infoHost.AvailableBuffers, 1); // At least 1 buffer is available after an "ok"
      goto parse_end;  // there's nothing else to check for
  }
```

AvailableBuffers is initialized to 4 (Marlin's default) when the printer connection is established. (parseACK.c)

```
  infoHost.connected = true;
  infoHost.AvailableBuffers = 4;
  requestCommandInfo.inJson = false;

```

2. In interfaceCmd.c I changed the 2 sections with the sendCmd. For each send I deduct the buffer counter, and pause if no more buffers are available.

```
      if (infoHost.AvailableBuffers || !infoHost.connected)
        if (sendCmd(false, avoid_terminal) == true)  // if the command was sent
        {        
		  infoHost.AvailableBuffers--;
		  if (infoHost.AvailableBuffers)
		    infoHost.wait = false;
		  else
			infoHost.wait = infoHost.connected;
	    }
```

Now there are no more additional buffer underruns during the printing process and at least basic operations seem to work as before. (Move menu works, terminal works, adjusting values during printing works).

carlosjln, 2023-07-27:
Wow I finally stumble upon the right track. I have recently rebuilt my delta and I started to notice these blobs and pauses. Even when I dial the feedrate/speed to 75%.

I'll keep an eye here :)

carlosjln, 2023-07-27:
As I shared here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2761#issuecomment-1652926699 -- BAUDRATE 57600 seems to have solved or greatly improved the situation 🤔

rondlh, 2023-07-27:
> As I shared here [#2761 (comment)](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2761#issuecomment-1652926699) -- BAUDRATE 57600 seems to have solved or greatly improved the situation 🤔

Certainly too high baudrates can cause issues, I run 3 printers at 500K, but 1 printer doesn't work well if I don't go down to 250K. Note that at 57K, you can only send 5.7K bytes per second, so if the gcode command is 57 bytes long, then you can only send 100 commands per seconds and basically at least have a 10ms delay between each command (displays waits for an "ok" after each command before sending the next command).
You can give the code above a try, enable ADVANCED_OK and make sure that RX_BUFFER_SIZE is at least 256 bytes, or more. I use 1024 bytes. The code is still experimental, so don't play around with the display too much during printing, which is good advice in any case! I have great results with it and can see that the buffers are almost full the whole time.

carlosjln, 2023-07-27:
@rondlh thanks for that info, I'll recompile with those settings tomorrow and try it with more customer prints as they come in.

rondlh, 2023-07-27:
> @rondlh thanks for that info, I'll recompile with those settings tomorrow and try it with more customer prints as they come in.

I attached src/User/API/interfaceCmd.c and src/User/API/parseACK.c and src/User/main.h, the code changes are minimal, just pickup the ADVANCED_OK feedback (parseACK.c) and do some buffer administration in interfaceCmd.c.
[interfaceCmd parseACK main.h.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/12180717/interfaceCmd.parseACK.main.h.zip)

The code is a bit different than above because it also handles the case that ADVANCED_OK is not enabled. I recommend to enable ADVANCED_OK, that's your safest bet. Note, there is no Reprap support added, only Marlin.

rondlh, 2023-08-06:
@carlosjln Any results? Questions?
I did some more tests, and found all is working very well, both with ADVANCED_OK enabled and disabled.

Nerdcaught, 2023-08-11:
Hi, I would like to implement this Test, but whenever Im replacing the files and compiling, I get this error. 
[image]

rondlh, 2023-08-11:
@Nerdcaught Sorry, there was a missing opening bracket in interfaceCMD.c
I updated the download above, please try again.

Nerdcaught, 2023-08-12:
Between you and me I wont officially know the true difference but if I have any hiccups Ill let you know. It compiled with your new updated files. Thank you!

rondlh, 2023-08-12:
> Between you and me I wont officially know the true difference but if I have any hiccups Ill let you know. It compiled with your new updated files. Thank you!

Great, make sure to enlarge RX_BUFFER_SIZE in Configuration_adv.h, I recommend 512 or 1024.
Enabling ADVANCED_OK is useful too, it will work fine without it, then it just assumes there are 4 buffers and keep track of the buffer use manually.

digant73, 2023-08-17:
I can provide this solution on a PR. That feature will be configurable at runtime (Advanced OK feature ON or OFF). For backward compatibility (also with mainboard fw different than Marlin), in case Advanced OK feature is disabled in TFT The current logic (one gcode at a time) will be used.
Eventually., this PR will include (if confirmed by the ongoing tests) also the fix discussed for #2761.

carlosjln, 2023-08-17:
I've been using the pre-compiled version I was shared some weeks ago and so far had two freezes, one at normal speeds on a big print, but I was able to restart the printer and resume it. The other one I increased the speed to 150% just to mess around as the print was not coming well and I was going to stop it anyways.

I think I'll try the 512 or 1024 buffer config later.

rondlh, 2023-08-23:
Implemented in #2824
Thanks go to @digant73, great job!

digant73, 2023-08-23:
it is better you don't close a FR o BR once a PR is available. PR marked this FR as resolved so it will automatically closed once the PR is merged. Just to let github makes its work

rondlh, 2023-08-24:
I thought it was merged already... but it is not..

rondlh, 2023-08-29:
> I've been using the pre-compiled version I was shared some weeks ago and so far had two freezes, one at normal speeds on a big print, but I was able to restart the printer and resume it. The other [...]

@carlosjln I'm not sure what pre-compiled version you are talking about. I did not provide one, and as far as I know there is **none** for ADVANCED_OK, but there is one for the hesitations issues. The hesitations issue should be solved in the current release (2023-08-24). 
#2824 Adds ADVANCED_OK support to the firmware, but it is not yet merged. (You can still get the files, do you know how to do that?)
For RX_BUFFER_SIZE, if you have 4 command buffers (BUFSIZE in Configuration_adv.h) then 512 bytes is already beyond the maximum theoretical requirement, and practically you only need less than half ((BUFSIZE - 1) * 50 bytes).
You can enable RX_BUFFER_MONITOR (Configuration_adv.h) if you worry about buffer under runs, then Marlin will inform you if it is running out of buffers.

carlosjln, 2023-08-29:
Sorry I mixed up issues because I'm tracking this other conversation https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2761

And I was referring to these pre-compiled versions https://github.com/kisslorand/BTT-TFT-FW

rondlh, 2023-08-30:
> And I was referring to these pre-compiled versions https://github.com/kisslorand/BTT-TFT-FW

OK, that is what I guessed, unfortunately I don't know what is changed in this FW, but it seems to solve the hesitation issue too. I recommend to just go for the latest FW or even add the ADVANCED_OK update.

rondlh, 2023-08-31:
@digant73 I just tested #2824, it doesn't seem to work for me. Marlin has ADVANCED_OK enabled, TFT too, tx_slots is set to 8 (also on Marlin), but I can see that Marlin never needs to queue more than 1 command.

digant73, 2023-08-31:
> @digant73 I just tested #2824, it doesn't seem to work for me. Marlin has ADVANCED_OK enabled, TFT too, tx_slots is set to 8 (also on Marlin), but I can see that Marlin never needs to queue more than 1 command.

yes I also saw that. Furthermore, printing from onboard (Marlin) I see 0 tx_slots and not possible to stop/control the print.
PR converted to draft just to solve the issue

rondlh, 2023-08-31:
> yes I also saw that. Furthermore, printing from onboard (Marlin) I see 0 tx_slots and not possible to stop/control the print. PR converted to draft just to solve the issue

The problem is in interfaceCmd.c, **infoHost.tx_count == 0**, so if only 1 command is send it's true already. The condition should be something like:
  infoHost.tx_count >= (tx_slots - 1) 
tx_slots as defined in the config.ini

```
bool isNotEmptyCmdQueue(void)
{
  return (cmdQueue.count != 0 || **infoHost.tx_count == 0**);  // if queue not empty or pending command
}

```

digant73, 2023-08-31:
just edited my previous post with the same issue (I remembered that I changed that condition in the past).

rondlh, 2023-09-01:
**I found one more issue:**

The TFT can recover from a motherboard reset, see code in parseACK.h

```
    // check for motherboard reset (external, software, etc)
    else if (ack_seen("Reset"))
    {
      if (ack_seen("External") || ack_seen("Software") || ack_seen("Watchdog") || ack_seen("Brown out"))
      {
        // proceed to reset the command queue, host status, fan speeds and load default machine settings.
        // These functions will also trigger the query of temperatures which together with the resets
        // done will also trigger the query of the motherboard capabilities and settings. It is necessary
        // to do so because after the motherboard reset things might have changed (ex. FW update by M997)

        clearCmdQueue();
        memset(&infoHost, 0, sizeof(infoHost));
	initMachineSettings();
        fanResetSpeed();
        coordinateSetKnown(false);
        setReminderMsg(LABEL_UNCONNECTED, SYS_STATUS_DISCONNECTED);  // set the no printer attached reminder
      }
    }

```

But the code there doesn't support the new definition of infoHost.

```
void InfoHost_Init(bool isConnected)
{
  infoHost.tx_slots = 1;  // set to 1 just to allow a soft start
  infoHost.tx_count = 0;
  infoHost.connected = isConnected;
  infoHost.status = HOST_STATUS_IDLE;
}
```

I believe
`     memset(&infoHost, 0, sizeof(infoHost));`
should be replaced by:
`     InfoHost_Init(false);`

digant73, 2023-09-01:
yes correct, Thanks for the report. I will apply the change. I spent all the time asking people to use/provide API's functions (code more readable and safe) to avoid this kind of bugs when a bit is changed in the TFT logic...
As you can see on my last commit I added a Monitor menu (accessible on Tuning menu (so even when printing). To see the menu you need to enable DEBUG_MONITORING in Configuration.h. Obviously other info could be added if helpful.

rondlh, 2023-09-01:
Here some feedback...

About HOST_SLOTS_GENERIC_OK and HOST_SLOTS_REGULAR_OK
I feel these are not very descriptive, "generic" and "regular" is about the same

They seem to represent the old way of handing commands (one at a time) and static ADVANCED_OK.

Perhaps something like this?
HOST_SLOTS_ONE_COMMAND_ONLY or HOST_NO_ADVANCED_OK
HOST_SLOTS_STATIC_ADVANCED_OK

Let me do some testing of new debug capabilities...

digant73, 2023-09-01:
I would simply convert HOST_SLOTS_GENERIC_OK to HOST_SLOTS_INCREMENTAL_OK.
HOST_SLOTS_REGULAR_OK is referred to "ok\n" while advanced ok feature is then controlled by TFT's configuration setting `advanced_ok`. If set to false, no advanced ok is applied

rondlh, 2023-09-03:
OK, I understand.
I did several prints and some stress tests with ADVANCED_OK enabled both on Marlin and TFT, seems to work fine, no issues at all. I believe this is the way to go.
The only minor disadvantage is that the commands from the TFT, like temp change etc. will be slightly delayed (usually just 1 or 2 seconds) because they are also in the queue and need to wait their turn. Worst case the delay could be longer.

It would be great if more people could test this code so we can make a push to get it merged.

digant73, 2023-09-03:
Just pushed an update fixing issue of the mainboard reply out of synch with the current pending gcodes (it happens on my printer). E.g. TFT was notified by mainboard to have 15 free slots (the maximum allowed in my printer) while the TFT had already some pending gcodes so the pending gcodes could exceed the Marlin queue maximum size (e.g. 30 pending gcodes in my case)

rondlh, 2023-09-03:
OK, I see, quite a lot of changes and additional code.

Seems to be getting too complicated compared to the original base implementation because you try to balance the send and received commands. Can't you just drop "tx_count" and only used "tx_slots"?
Controlling the max allowed "tx_count" gives you full control on how the TFT should behave.

digant73, 2023-09-03:
you cannot avoid the use of tx_count. tx_slots is not enough to avoid out of synch information. I had a lot of skipped gcodes at the beginning of print. I put a lot of comments in the code.
The code should be almost ok. I see just a strange behavior after the first print. When the next print starts some gcodes seem to be skipped at the beginning although tx_count is correctly stable to 15 (the buffer size in my Marlin configuration).

rondlh, 2023-09-04:
What is out of sync information?
With ADVANCED_OK enabled Marlin will report how many buffers are available, but Marlin cannot see how many commands are still in it's serial input buffers, do you mean that?

digant73, 2023-09-04:
> What is out of sync information? With ADVANCED_OK enabled Marlin will report how many buffers are available, but Marlin cannot see how many commands are still in it's serial input buffers, do you mean that? Do you want me to test the current status?

yes. I meant that (I had even 33 pending gcodes while the buffers were only 15). No there is no need to test thanks. I will reduce the logic in main.c

digant73, 2023-09-04:
last cleanup in main.c. PR ready for merge.

rondlh, 2023-09-05:
1. Is this still valid? (#2761)

> * `ADVANCED_OK` **OFF** -> `tx_slots` is always used. e.g. if 1 -> current situation, if 2 -> static A.davanced OK (improved but conservative)
> 
> * `ADVANCED_OK` **ON**:
>       
>    * `tx_slots` is used if ADVANCED_OK not enabled in Marlin
>    * Marlin indication is used if ADVANCED_OK enabled in Marlin

If so, it seems not to be working for me, ADVANCED_OK enabled in Marlin(BUFSIZE=8) and TFT(Reset default settings, but ADVANCED_OK enabled). I can see that only 1 command is queued anyway, I would expect 8.

2. This section in Configuration.h is a bit odd:

```
 *   Value range: [min: 2, max: 16]
 */
#define TX_SLOTS 1  // Default: 1

```

3. The HOST_SLOTS enumeration seems a bit odd to me:
```
typedef enum
{
  HOST_SLOTS_GENERIC_OK = -1,
} HOST_SLOTS;
```

4. The config.ini in "\TFT\src\User\" is not in sync anymore with the one in "\Copy to SD Card root directory to update\\"
I'm not sure why there is one in "\TFT\src\User\" anyway, as it seems not to be used anywhere (I know it has been like this for a while, not a new thing).

digant73, 2023-09-05:
no, I changed the logic to be easy to understand:
- `advanced_ok off` -> 1 gcode per time (backward compatibility with current solution)
- `advanced_of on` -> Marlin ADVANCED_OK if detected by TFT or static ADVANCED_OK (using `tx_slots` configured in `config.ini`)

rondlh, 2023-09-05:
> * `advanced_ok off` -> 1 gcode per time (backward compatibility with current solution)
> 
> * `advanced_of on` -> Marlin ADVANCED_OK if detected by TFT or static ADVANCED_OK (using `tx_slots` configured in `config.ini`)
>   Description on PR and config.ini have been already updated in the past

So does this mean that ADVANCED_OK is not correctly detected? It's enabled in Marlin and the TFT, but still not used by the TFT.

digant73, 2023-09-05:
Marlin ADVANCED_OK (its buffer size (e.g. 15 in my case)) is properly detected by TFT if ADAVNCED_OK is enabled in Marlin. If not detected then static ADVANCED_OK is used.
But in both cases `advanced_ok` must be also enabled. Otherwise standard 1 gcode per time is used).

rondlh, 2023-09-05:
About 1. Fixed now, good job.
About 2. Issue still remains
About 3. My point is that an enumeration with only 1 item is a bit odd, a simple #DEFINE would do the job, but no big deal.
About 4. 
> The master config.ini must be provided only in "\TFT\src\User" since some time now. They will be copied in "\Copy to 
SD Card root directory to update" when merged by BTT.

In that case be aware that the config.ini in "\TFT\src\User" is incorrect, the one in  "\Copy to 
SD Card root directory to update" is the correct one.

digant73, 2023-09-05:
> About 1. Fixed now, good job. About 2. Issue still remains About 3. My point is that an enumeration with only 1 item is a bit odd, a simple #DEFINE would do the job, but no big deal. About 4.
>
>
> In [...]

oh yes right. Fixed 2 and 4 (as I said master folder for `config.ini` is `\TFT\src\User`). For 3, yes I know but it is better than a #define or passing a value < 0 on handleOkAck() function

digant73, 2023-09-06:
yes I know that. I previously put a delay of 10 ms on that loop just to avoid that errors. I will see how to improve it

rondlh, 2023-09-06:
A delay would probably do the trick just fine, or something like this:
```
if (Serial_Available(SERIAL_PORT) != 0)
  for (int i = 0; (i < 4) && condCallback(); i++)
    sendEmergencyCmd("M108\n");
```

The amount of commands that needs to be cancelled is quite limited.

rondlh, 2023-09-07:
While your are changing interfaceCmd.c, perhaps you could correct the error below ("for" and "wait" should be swapped in the last line)

```
// Store gcode cmd to cmdQueue queue.
// This command will be sent to the printer by sendQueueCmd().
// If the cmdQueue queue is full, a reminder message is displayed
// and it will for wait the queue to be able to store the command.
```

digant73, 2023-09-07:
> While your are changing interfaceCmd.c, perhaps you could correct the error below ("for" and "wait" should be swapped in the last line)
>
> ```
> // Store gcode cmd to cmdQueue queue.
> // This command [...]

ok, swapped. Fixed also the possible issue on print abort

rondlh, 2023-09-07:
@digant73 YOU ARE UNSTOPPABLE! Great job.
BTW: I just posted another BR, not a big one, but a nice one :D ... not related to ADVANCED_OK

kisslorand, 2023-09-09:
> I noticed another issue. When I start a print and immediately cancel it via the TFT, then the motherboard is flooded with M108 commands, sometimes causing a buffer overruns (my buffer is even at 1024 bytes). I'm not sure what is causing this

It is caused by the "improvements" done to the serial port read. I warned about it being buggy several times already, but I saw the new "enhancements" made it even more chaotic. Just check earlier builds (before PR #2788) or [my builds](https://github.com/kisslorand/BTT-TFT-FW) where `M108` do not exhibit this behaviour.

> I guess it's better to limit the max number of M108 commands that can be send.

It is the very unprofessional to mitigate the outcome of a bug instead of fixing its core root. It should be absolutely the last resort. I saw in a current open PR that it does exactly that, a patch was made to somehow overcome the issue...

rondlh, 2023-09-11:
@kisslorand More code, less words please. If you want to help, then be concrete, your vague statements are useless to me, and so is closed source software.
What changes do you think cause this issue? (Which lines of code? Which strategy aspects? What assumptions are wrong?)
Can you show me your solution (code please...)? Or at least describe a strategy?

digant73, 2023-09-11:
> What changes do you think cause this issue? (Which lines of code? Which strategy aspects? What assumptions are wrong?) Can you show me your solution (code please...)? Or at least describe a strategy?

Transmission of M108 was previously based on polling rx_ok although there was a scenario where it failed not sending any M108, so not aborting the print.
The logic in #2824 is basically the same plus it always works in any scenario (just read the comment in the line where rIndex_old is defined). Variable rIndex could be also avoided. It has been used simply because the binary is even smaller than using two invokations of Serial_GetReadingIndex(SERIAL_PORT).

digant73, 2023-09-11:
> Thanks, that's helpful, this is one of the areas in the code that I don't fully understand, it must be boring to have to explain things to newbies sometimes, thanks for that. It seems to work fine in my experience.

no problem. It works fine and even better than the old solution. I would not take into account the continuous proclaims / biased / catastrophic words from some contributors. The TFT is pretty much stable since some time. It only requires some minor bug fixes / minor code optimization / minor new features. Basically it needs only minor maintenance. I asked BTT to possible merge many of the old PRs but consider that possibly they have no more interest on this project as they are moving towards Klipper.

rondlh, 2023-09-11:
@digant73 Could you tell me what `detectAdvancedOk` does? It seems to be trying to query the amount of Marlin command buffer from a M105 response. But my M105 response looks like this:
ok T:36.88/0 B:34.83/0 @:0

My advanced OK response looks like this:
ok P15 B7

I added Target TX Slots to the monitoring screen, which reports a value of 5, my Marlin command buffer size is set to 8.
Free TX slots also reports 5. Pending/buffered gcodes are both 0. Does this make sense?

digant73, 2023-09-11:
> @digant73 Could you tell me what `detectAdvancedOk` does? It seems to be trying to query the amount of Marlin command buffer from a M105 response. But my M105 response looks like this: ok T:36.88/0 B:34.83/0 @:0
> 
> My advanced OK response looks like this: ok P15 B7
> 
> I added Target TX Slots to the monitoring screen, which reports a value of 5, my Marlin command buffer size is set to 8. Free TX slots also reports 5. Pending/buffered gcodes are both 0. Does this make sense? I debugged the code a bit and can see it's looking at a correct advanced ok response: "ok P15 B5" So the reported 5 is correct, but that is not the amount of my command buffers.

my mistake! I forgot to push the correct `gcode.c` file. M107 (or any other gcode replied with `ok Px By` must be used. I used also M105 just for testing detectAdvancedOk. Just pushed the proper file gcode.c now.
detectAdvancedOk() will simply parse the ok reply and retrieve the Marlin queue size (e.g. 7 in your and also my case).

rondlh, 2023-09-11:
OK, got it. I changed "M105" to "M107", now I get the correct value of 7. Thanks.

I see instability in the Marlin RX data...
I monitor this from Marlin's side, with every received command I report how much RX data is still available and how many command buffers are used. I can see all (8) buffers are full, but still 600 bytes in the input buffer.  (1 command is about 40 characters) so something is going wrong. I have DMA writing enabled, but with my simple advance ok implementation I never saw this issue. I will try the #2824 code without any changes

rondlh, 2023-09-12:
@digant73 I found a little issue in #2824... 
NotificationMenu.c

```
void menuNotification(void)
{
  LISTITEMS notificationItems = {
    LABEL_NOTIFICATIONS,
    // icon            item type   item title     item value text(only for custom value)
    {
      {CHARICON_NULL,  LIST_LABEL, LABEL_DYNAMIC, LABEL_NULL},
      {CHARICON_NULL,  LIST_LABEL, LABEL_DYNAMIC, LABEL_NULL},
      {CHARICON_NULL,  LIST_LABEL, LABEL_DYNAMIC, LABEL_NULL},
      {CHARICON_NULL,  LIST_LABEL, LABEL_DYNAMIC, LABEL_NULL},
      {CHARICON_NULL,  LIST_LABEL, LABEL_DYNAMIC, LABEL_NULL},
      {CHARICON_BLANK, LIST_LABEL, LABEL_CLEAR,   LABEL_NULL},
      #ifdef DEBUG_MONITORING
        {CHARICON_BLANK, LIST_LABEL, LABEL_INFO,    LABEL_NULL},
      #else
        {CHARICON_BACK,  LIST_LABEL, LABEL_NULL,    LABEL_NULL},
      #endif
      {CHARICON_BACK,  LIST_LABEL, LABEL_NULL,    LABEL_NULL},
    }
  };
```
I think the "else" case should be:
     ` {CHARICON_NULL,  LIST_LABEL, LABEL_DYNAMIC, LABEL_NULL},`

Otherwise you still get a back icon in the notification menu, while `DEBUG_MONITORING` is not defined.

Advanced_OK seems to work fine, Marlin RX buffer use is very moderate, but when I add the DMA writing code, the buffer usage can become unstable, very odd, because your Advanced_ok implementation is smarter than mine and should not be affected by the serial send speed. I will look into it...

digant73, 2023-09-12:
> @digant73 I found a little issue in #2824... NotificationMenu.c
>
> ```
> void menuNotification(void)
> {
>   LISTITEMS notificationItems = {
>     LABEL_NOTIFICATIONS,
>     // icon            item type   item [...]

yes of course. Wrong cut & paste. I will push the minor fix

> Otherwise you still get a back icon in the notification menu, while `DEBUG_MONITORING` is not defined.
>
> Advanced_OK seems to work fine, Marlin RX buffer use is very moderate, but when I add the DMA [...]

Good it works now. Yes I think DMA writing should be investigated separately

rondlh, 2023-09-12:
> Good it works now. Yes I think DMA writing should be investigated separately

Yes, of course.
I don't like to post the DMA code when it causes issues for #2824, because it's build on it.
I found a bug in my "improvements" causing some issues, now Advanced ok and DMA writing work together fine. 

I guess you are aware of this... but anyway.
The number of command buffers Marlin reports in the Advanced_ok response is actually 1 too low, because the command causing the ok response is occupying a command buffer when the message is send. Sending the ok response is the last thing done by the gcode command, so the command buffer will become available about 1us after the message is sent. As a result the TFT targets 7 command buffers when there are actually 8 available. Perhaps we can consider it a safety margin to not stress the Marlin RX buffers too much.

Some more ideas for the monitoring feature.... commands per second, kbytes sent, kbytes received, polling rate (Serial_Get).

digant73, 2023-09-12:
> Yes, of course. I don't like to post the DMA code when it causes issues for #2824, because it's build on it. I found a bug in my "improvements" causing some issues, now Advanced ok and DMA writing [...]

Yes I know the reported available buffer is 1 less than the configured value in Marlin. That's ok for me (safety value).
Yes I also thought to KPIs such as msg/sec etc... I always provide those kind of KPIs in applications developed for my work.

rondlh, 2023-09-13:
@V1EngineeringInc/Everyone This would be a good time to test ADVANCED_OK #2824. I tested it fully, and it is working very well.
I hope we can get this merged asap, because not having this is holding me back currently. This PR involves a language update to add the word "Advanced OK", so witching back is painful. I have the DMA serial writing ready for test, and some good performance improvements are in progress, but they depend on this update. Perhaps you can help?

rondlh, 2023-09-21:
#2824 now contains a performance benchmark that you can enable in Configuration.h (DEBUG_MONITORING). When enabled, you will find an info button in the in the notification screen. The button brings you to the monitoring screen. There you can monitor some system parameters, like the scan rate which shows how many times the core process (loopProcess) is executed every second. This way the impact of changes can be tested easily.

closed 2023-09-26 by bigtreetech