bigtreetech/BIGTREETECH-TouchScreenFirmware #2823: [BUG] TFT35 Not connecting

IoakeimSogiakas, 2023-07-31:
I have a weird issue. I have a tft35 screen and a skr mini e3 v3 board. My configuration for both is as follows (see config files). The screen doesn't connect in tft mode. Nor does it saw life in marlin mode what am doing wrong?

IoakeimSogiakas, 2023-07-31:
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/12213634/Marlin.zip)here is the marlin side of things and the ini file for the screen
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/12213657/config.zip)

rondlh, 2023-08-01:
There is not enough information to evaluate this issue, but you can start with checking the wiring.

IoakeimSogiakas, 2023-08-01:
OK thanks I will close the issue and if I find anything worthwhile I will inform you

closed 2023-08-01 by IoakeimSogiakas