bigtreetech/BIGTREETECH-TouchScreenFirmware #2841: IDEX Printer, setting Hotend Count and Extruder Count to 2 is not working

badonj001, 2023-09-23:
Hi, I have the BIGTREETECH TFT35 E3 V3.0.1 Touch Screen connected to the BIGTREETECH SKR PRO V1.1. I have an IDEX printer, or independent dual extruders.  When I am in Marlin Mode I have complete control over both extruders as marlin is configured correctly for that. I have a couple of these printers I would like to upgrade using this display and your help is appreciated.

The issue I am having is in the BTT TFT35 firmware config.ini file I configure both hotend_count:2, and ext_count:2. It is my understanding that this is the right configuration to specify I have two separate hotends, and those hotends have their own separate stepper motors.

I put the firmware, bmps, edited config.ini, and reset.txt on the SDCARD and reboot to do the firmware update. Doing this does not give me the ability to control the second extruder Marlin mode allows the control of both.

So the config.ini file says the following:

#### Extruder Count
# Set extruder count.
#
# NOTE: This value is overridden by the TFT if provided by Marlin firmware.
#
#   Value range: [min: 0, max: 6]
ext_count:2

So I am assuming that Marlin is sending conflicting information? Do I need to #define DISABLE_M503 to resolve this issue? Or dop I need to commit out this #define EEPROM_BOOT_SILENT? 

Marlin-Mode definitely knows there are 2 extruders, and I can print a *.gocode file using both and it works. It is only in TFT mode that I can not control T1 or Extruder 2 however you look at it. 

Your help is appreciated and video is attached along with my config.ini with the .txt extension as it is the only way to upload it for review without zipping. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/assets/18022435/0c06cc7c-f5a4-4511-881d-a04c1a11aa2d

[config.ini.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/12705521/config.ini.txt)

rondlh, 2023-09-23:
> So I am assuming that Marlin is sending conflicting information?

Perhaps, probably not.

> Do I need to #define DISABLE_M503 to resolve this issue?

No

Or do I need to commit out this #define EEPROM_BOOT_SILENT?

No

> Marlin-Mode definitely knows there are 2 extruders, and I can print a *.gcode file using both and it works.

Marlin-Mode knows nothing, it just receives graphics from Marlin and puts them on the screen.

I have the same IDEX machine, and it's working fine for me. A MKS TFT35 V1.0 will fit in perfectly as a replacement of the original TFT, but you will need extenders for the SDCard and USB.
Do you have the latest screen firmware?
What version of Marlin do you use?

Make sure to follow the Marlin requirements as stated here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies

badonj001, 2023-09-24:
Ok,
So I am using Marlin 2.1.2.1, that is the absolute most current version. My
printer works flawlessly with it.

I do understand that marlin mode makes the TFT work just like the Full
Graphic Smart Controller and is why it is necessary to connect the
ribbon cables as well. TFT mode uses the RS232 cable, and marlin-mode uses
the EXP1 and EXP2 cables as  Full Graphic Smart Controller displays.

I also do understand that the TFT-Mode works just like printing from
Octo-Print, or from your computer. At that point marlin is just executing
commands and sending acknowledgements back.

I already own the BTT TFT35 and the BTT SKR PRO. I like having the knob on
the display. Also, I removed the MKS electronics and replaced them due to
random interniting bugs including unwanted bltouch deploying during
printing. There is something unstable about the MKS Motherboard (MKS Robin
Pro). I have a couple of these IDEX printers because JG Maker is basically
giving them away now. It is a printer with great bones, but not the
greatest overall implementation.

I am trying to figure out why a display that is supposed to allow for
multiple tools in the configuration is not working. The only information it
says if Marlin sends it information in conflict it will use what marlin
tells it.

Attached is the screenshot from platform.io in Visual Studio Code and both
my configuration.h, configuration_adv.h, and the display config.ini.

All of my research says this should work? That is why I posted the
question.

On another note, I tried out klipperscreen last night and got it working
with the exception of the BlTouch, and the extruders are not extruding for
some reason. One of the things that turned me off to klipper was even with
a RPI 3B+, there is an undesirable latency between selecting a command, and
having it executed. So Home all axis, takes 3-5 second wait after pressing
the button. I have not been able to print with it yet as it has some kinks,
but if the latency is also there when pringtn, then the print will reflect
that in defects.

I think that I can have a dedicated screen with the BTT TFT35, and
octoprint if I need a remote is the way I want to go. Klipper requires
thier own firmware, and any other marlin type display is not supported.

With the TFT and the SKR Pro, this is the smoothest interfacing with my
printer I have ever had. I would just like to figure out why the setting
the TFT is designed for is not working, and not to just replace it.

So this part in the config.ini makes me think the TFT is getting some
information from Marlin that is overriding the Extruder count. Because that
is what the config file says in that section.I am trying to figure out if
it is a bug, or me not knowing I am missing a configuration in marlin. In
the machine settings in TFT-Mode I can set the 2nd tools offset relative to
the 1st Primary Tool.

#### Extruder Count
# Set extruder count.
#
# NOTE: This value is overridden by the TFT if provided by Marlin firmware.
#
#   Value range: [min: 0, max: 6]
ext_count:2

All help is appreciated.

[image: image.png]
[image: image.png]

/**
[... 1369 lines cut ...]
// LCD Backlight Timeout
//
//#define LCD_BACKLIGHT_TIMEOUT_MINS 1  // (minutes) Timeout before turning off the backlight

#if HAS_BED_PROBE && EITHER(HAS_MARLINUI_MENU, HAS_TFT_LVGL_UI)
  //#define PROBE_OFFSET_WIZARD       // Add a Probe Z Offset calibration option to the LCD menu
  #if ENABLED(PROBE_OFFSET_WIZARD)
    /**
     * Enable to init the Probe Z-Offset when starting the Wizard.
     * Use a height slightly above the estimated nozzle-to-probe Z offset.
     * For example, with an offset of -5, consider a starting height of -4.
     */
    //#define PROBE_OFFSET_WIZARD_START_Z -4.0

    // Set a convenient position to do the calibration (probing point and nozzle/bed-distance)
    //#define PROBE_OFFSET_WIZARD_XY_POS { X_CENTER, Y_CENTER }
  #endif
#endif

#if HAS_MARLINUI_MENU

  #if HAS_BED_PROBE
    // Add calibration in the Probe Offsets menu to compensate for X-axis twist.
    //#define X_AXIS_TWIST_COMPENSATION
    #if ENABLED(X_AXIS_

badonj001, 2023-09-24:
Ok, as of now it would appear for the BTT TFT35 I downloaded the firmware again and uploaded it and NOW I have T0 and T1.

Thank you for your time and help.

badonj001, 2023-09-24:
Well, I changed some setting in the ini file and now it does not work again? THIS IS A BUG!

rondlh, 2023-09-25:
> Well, I changed some setting in the ini file and now it does not work again? THIS IS A BUG!

I recommend to undo the changes you did to the ini-file, so you can go back to:

>Ok, as of now it would appear for the BTT TFT35 I downloaded the firmware again and uploaded it and NOW I have T0 and T1.
>Thank you for your time and help.

All is working fine for my IDEX. So I'm afraid there is a BUG, but not in the TFT firmware.

badonj001, 2023-09-26:
> I recommend to undo the changes you did to the ini-file, so you can go back to:
>
>
> All is working fine for my IDEX. So I'm afraid there is a BUG, but not in the TFT firmware.

rondlh,

You are correct, it does work. The problem was as is sometimes the case operator headspace,lol. My problem is relying on my memory from about 7-months ago when I first configured one of these display for 3d printer donation. I do not remember why, but I had to put an empty text file on the sdcard to reset the firmware. So apparently after reading the instructions again I noticed that the reset.txt is the absolute last thing the firmwares looks for after updating everything else. So It was loading the config settings I set, then running the reset undoing all of the settings it just changed. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#resetting-to-default-settings

closed 2023-09-26 by badonj001