bigtreetech/BIGTREETECH-TouchScreenFirmware #2843: [Q] (How do i fix "all low" error message")

Chao5uns, 2023-09-24:
i replaced a board on an ender 3 clone and when i flashed firmware it said "all low" and it refuses to home the z
this is the result when i put in M122

M122 code:
 X Y Z E
Address   0  2  1  3
Enabled   false  false  false  false
Set current   725  800  800  800
RMS current   703  795  1436  1436
MAX current   991  1121  2025  2025
Run current   22/31  25/31  25/31  25/31
Hold current   11/31  12/31  12/31 12/31
Cs actual   11/31  12/31  0/31  0/31
PWM scale
vsense   1=.18  1=.18  0=.325 0=.325
stealthchop   true  true  false  false
msteps   16  16  256  256
interp   true  true  false  false
tstep   max  max  0  0
PWM thresh.   98  98  2108  4534
[mm/s]   100  100  15  30
OT prewarn   false  false  false  false
triggered
OTP   false  false  false  false
pwm scale sum   13  14  0  0
pwm scale auto   0  0  0  0
pwm offset auto   36  36  0  0
pwm grad auto   14  14  0  0
off time   4  4  0  0
blank time   24  24  16  16
hysteresis
-end   2  2  -3  -3
-start   1  1  1  1
Stallguard thrs   0  0  0  0
ustep count   8  8  0  0
DRVSTATUS   X  Y  Z  E
sg_result   0  0  0  0
stst              *   *
olb
ola
s2gb
s2ga
otpw
ot
157C
150C
143C
120C
s2vsa
s2vsb
Driver registers:
    X  0xC0:0B:00:00
    Y  0xC0:0C:00:00
    Z  0x00:00:00:00   Bad response!
    E  0x00:00:00:00   Bad response!
Testing X connection... OK
Testing Y connection... OK
Testing Z connection...  Error: ALL LOW
Testing E connection...  Error: ALL LOW

This is my hardware:
BTT E3 SKR MINI E3
BTT E3 touchscreen

i dont know what to do i read and tried every solution online

rondlh, 2023-09-25:
This is not the correct place for this question, but luckily your description is very detailed: "a board on an ender 3 clone", which makes answering your question very easy...

Check the stepstick configuration jumpers (under the stepsticks), make sure they are setup for the communication type you use (Serial or SPI). Copy the jumper settings of your old motherboard.
If you don't use the exact same motherboard then you will need to look into your pin configuration.

Chao5uns, 2023-09-26:
Where should I ask this question plus it was a btt mini skr e3 and btt touchscreen

rondlh, 2023-09-26:
> Where should I ask this question plus it was a btt mini skr e3 and btt touchscreen

Check the BTT Mini SKR forums on github and/or FB.
Here you can report/discuss specific TFT Touchscreen software related bugs and features.
Please close this topic by using "Close with comment" at the bottom.

Chao5uns, 2023-09-29:
okay closing question

closed 2023-09-29 by Chao5uns