bigtreetech/BIGTREETECH-TouchScreenFirmware #2847: [Q] @bigtreetech Faulty/buggy PR @2844 by Kisslorand, PLEASE DO NOT MERGE

rondlh, 2023-09-28:
你好 @bigtreetech,

I want to bring to your attention that **@Kisslorand PR #2844 is faulty/buggy/non-functional and should NOT be merged**.
I have been working with @Kisslorand several times to try to help him to raise the quality of his PRs.
But Kisslorand doesn't like constructive criticism, he takes things personal, resulting in him blocking me.
So now I have no good way to inform him that **#2844 is very bad**. So I take this odd way to ask for your attention.

Kisslorand claims this at the introduction of the PR:
> This PR was tested rigorously for a one week period for all the involved MCUs of this FW (even more, it was also tested on GD32F305VCT6 MCU too).

**This is a complete non-sense** The PR is completely non-functional and buggy, and doesn't fit in the current firmware strategy.
But still Kisslorand keeps force-pushing the PR starting from the day he published it.

In my view, PRs should be created based on an open discussion and they should be peer reviewed and tested, especially for PR that have hardware specific code. It would also be nice if the creator of the original code would have a chance to create a PR.

From @Kisslorand I want to ask to stop making competing PRs based on other peoples' work without any discussion or cooperation. Please cancel #2844, if you want to help you can join #2840.

谢谢，
龙海业

rondlh, 2023-09-28:
For reference, here an archive of #2844 (source code only), please don't download, this is non-functional buggy code!
[Archive #2844 2023-09-27.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/12746340/Archive.2844.2023-09-27.zip)

@kisslorand You claim that this software "was tested rigorously for a one week period for all the involved MCUs of this FW".
Which TFTs did you actually test this on?

rondlh, 2023-09-30:
By now it's clear that @kisslorand is spreading misinformation. In #2850 and #2851 he is lying about #2824, trying to get the PR reversed, which would be very painful and unneeded. #2824 is a great update providing ADVANCED_OK capabilities. @kisslorand's #2844 is buggy/faulty and completely non-functional, his code has not been tested, but still he is force-pushing the PR. 
@kisslorand's behavior is very harmful and despicable. I don't know what drives him to act in this way. Perhaps his pride was hurt when I informed him about the flaws and incorrect claims in #2833.

kisslorand, 2023-10-07:
Sparky!

kisslorand, 2023-10-08:
Most certainly you have the right to your own opinion.

I will not lower myself to your level and call you names, anyone around here mature enough can see what's there to see.

As a side note, thank you very much for the free publicity, bringing the attention towards me will help more and more people to try the alternative FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW) and the result of the work I've done there would be the truth, not what random trolls are mumbling on the internet. The only thing people care about is to have finished prints done as perfectly as possible. [The proof is in the pudding.](https://www.dictionary.com/e/slang/the-proof-is-in-the-pudding/)

rondlh, 2023-10-08:
@kisslorand Nobody cares about your buggy and slow repository, you must be out of your mind to even consider it. Your code is of very poor quality, because you do not communicate, coordinate, test or do peer reviews. Of course the outcome cannot be taken seriously. Your false performance claims have been debunked over and over again. **Don't forget to re-enable the ban!**

kisslorand, 2023-10-08:
\- Sparky! Who's the bad man?
 \- Wuff! Wuff!

> Don't forget to re-enable the ban!

Hmmm. Actually I am enjoying the show. Can you show us again how you're chasing your tail?
<br>

As a side note, thank you very much for the free publicity, bringing the attention towards me will help more and more people to try the alternative FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW) and the result of the work I've done there would be the truth, not what random trolls are mumbling on the internet. The only thing people care about is to have finished prints done as perfectly as possible. [The proof is in the pudding.](https://www.dictionary.com/e/slang/the-proof-is-in-the-pudding/)

rondlh, 2023-10-09:
### **Everyone: Be smart, don't install closed source firmware from other sites/repositories, it could be ransomware**

kisslorand, 2023-10-09:
Too late! What's done is done! It's ransom time baby! The evil plan was set in motion, let's burn some bridges!

Good boy, Sparky! Good boy! Now please do some roll-overs, let's see how entertaining you are...

As a side note, thank you very much for the free publicity, bringing the attention towards me will help more and more people to try the alternative FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW) and the result of the work I've done there would be the truth, not what random trolls are mumbling on the internet. The only thing people care about is to have finished prints done as perfectly as possible. [The proof is in the pudding.](https://www.dictionary.com/e/slang/the-proof-is-in-the-pudding/)

rondlh, 2023-12-25:
Thanks @bigtreetech, the correct code (#2840) by @digant73 was merged, the buggy update (#2844) by kisslorand has become obsolete.

closed 2023-12-25 by rondlh