bigtreetech/BIGTREETECH-TouchScreenFirmware #2849: [BUG] Non-functional corner leveling using a probe

radek8, 2023-09-29:
After four months I updated Marlin, and found I had the same problem as shown here:
https://github.com/MarlinFirmware/Marlin/issues/26171

The SERIAL_FLOAT_PRECISION option no longer works in Marlin.
The TFT display is no longer able to analyze the values from the leveling probe due to the format of the value sent by the Marlin.
When leveling and straightening corners with a probe, the measured values are not displayed.
Would someone look into this and fix the problem?
Thank you

### Hardware Variant
TFT 35 E3 V3.0
### TFT Firmware Version & Main Board Firmware details
current TFT Firmware 
current version Marlin 2.1 bugfix

kisslorand, 2023-09-29:
I already made several PRs in Marlin repository but they haven't merged any PRs since 26th of August 2023.

radek8, 2023-09-29:
Thank you @kisslorand 
I'll test your fix tomorrow.

radek8, 2023-09-30:
PR https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2825
did not solve this problem.

kisslorand, 2023-10-01:
BTW, this bug is not because of the "SERIAL_FLOAT_PRECISION" not being applied but because the syntax of the response of `G30` has changed and as a result of it the TFT cannot recognize it.
It's one of the fixes in PR #2825.

radek8, 2023-10-02:
Thank you for the explanation.
After editing, your PR is functional.

kisslorand, 2023-10-04:
BTW, my PRs regarding the "SERIAL_FLOAT_PRECISION" and G30 response precision has been merged into Marlin.

rondlh, 2023-10-09:
@radek8 Be careful with Kisslorand, he is not a reliable player and has done several attempts to sabotage this software repository. I'm not sure what his target is, perhaps ransomware, so be sure to **not install any firmware from other sources**. Be warned!

PR #2873 (2024-01-15): cumulative patches + cleanup
**IMPROVEMENTS:**
- cumulative patches/improvements: it includes some fixes/improvements provided up to PR #2844. They have been refined/fixed here and there to be fully working.  All credits to kisslorand
- cumulative cleanup: cleanup on different files just to remove from _.h_ files useless references to functions invoked only internally on related _.c_ files. Order of functions have been changed in _.c_ files just to make cleanup on related _.h_ files. _inline_ function qualifier used when needed to reduce code size
- other minor improvements/cleanup

**BUGFIXES [... cut]
closed 2024-01-15 by bigtreetech