bigtreetech/BIGTREETECH-TouchScreenFirmware #2855: [FR] Home button in the corner leveling menu

radek8, 2023-10-05:
I'm missing the Home button in the corner leveling menu.
If I move the screws, the center of the bed also moves, and I would like to execute the G28 command again.
If the M48 button were moved elsewhere, there would be a free position.
The M48 button is not important when leveling corners. 
The M48 button is already in the BL-Touch menu.

What do you think about replacing the M48 button with a Home button?

kisslorand, 2023-10-05:
What do you think what would be a better option? To home all axes or only Z?

radek8, 2023-10-05:
Both options are possible, but you can only choose one :-)
For most users, only the Z axis would probably be enough.
My opinion is that the G28 Z would be enough.
Corner leveling doesn't take long to time out the stepper motors.

radek8, 2023-10-05:
Could a condition be used?
If the stepper motors hold, will G28 be executed Z if the motors do not hold, will G28 be executed?
The display probably has no way of detecting whether the stepper motors are active.

kisslorand, 2023-10-05:
The ideal situation would by entering into this menu to disable Y steppers idle timeout and when exiting from this menu to restore original idle timeout value. Unfortunately there is no way to enquire the idle timeout value set up in Marlin.
Your reasoning about using `G28 Z` seems valid to me. 
I was also thinking about making that button to home only Z when pressed short and home XYZ when pressed long. The problem is that users most probably would be unaware of the long press function of that button.
Another solution could be to replace "Distance" with "Next" and that button to rotate the function of the "Home" button between "Distance", "Home All" and "Home Z".

What do you think?

radek8, 2023-10-05:
Switching parameters with the Next button is useless.
I would set the G28 z.
If someone wants Home for all axes, they can clean up the parameter and recompile.
If he can't do it, I'll be happy to help him with it if he asks :-)

kisslorand, 2023-10-05:
How does this look?

[image]

radek8, 2023-10-05:
Nice

sarvenn, 2023-10-05:
This is only required for printers that uses a probe for Z homing, right?

kisslorand, 2023-10-05:
The whole menu is useful only for printers that use a probe for Z homing.

digant73, 2023-10-05:
it should be better a submenu as done in other menus (e.g. `Manual Leveling` menu)

kisslorand, 2023-10-05:
> it should be better a submenu as done in other menus (e.g. `Manual Leveling` menu)

Hmmmm, I was just about to propose the removal of the submenu in the manual leveling menu because in my opinion "DisarmXY" has no practical use there.

Original:   
[image]

Proposed:
[image]

kisslorand, 2023-10-06:
PR #2825 is updated now with the discussed changes.

digant73, 2023-10-06:
basically the opposite of #2633 and what users ask for

kisslorand, 2023-10-06:
Sorry, I do not understand. What is exactly what users are  asking for that is implemented in https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2633 and taken away in #2825?

digant73, 2023-10-07:
> > basically the opposite of #2633 and what users ask for
> 
> Sorry, I do not understand. What is exactly what users are asking for that is implemented in #2633 and taken away in #2825?

People asks for integrating new/missing settings not to replace a feature with his preferred one. In #2825 you removed two functions (one in LevelCorner and one in Leveling). The unlock function in Leveling is useful for a fast/initial setup of the bed (moving the nozzle all around the bed) before fine tuning with knobs and eventually with LevelCorner (for printers with a probe). I also use it
Don't forget to re-enable the ban!

rondlh, 2023-10-07:
> Don't forget to re-enable the ban!

LOL! Don't worry, fools will be fools :D

kisslorand, 2023-10-07:
> People asks for integrating new/missing settings not to replace a feature with his preferred one. In https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/2825 you removed two [...]

Understood, thank for the clarifications. 
You think people need to know the standard deviation of their probe before doing the corner leveling and they need to enter into the manual leveling menu to be able to do the "moving the nozzle all around the bed" by hand, which is a very stupid thing to do, it can fry the motherboard.
BTW, I heard you have a new puppy, do not forget to pet him. He seems to be so happy barking at people you dislike, all he needs in return just approval and some attention from you.
<br>

> Don't forget to re-enable the ban!

No need, there's an automated troll detection activated.
<br>

> LOL! Don't worry, fools will be fools :D

\- Sparky! Where's the bad man?!
\-  Wufff! Wufff!"
<br>

******************************************************

Note to people in their right mind: __PLEASE DO NOT__ activate the stepper motors by hand! It can be fatal for the motherboard. I know it the hard way.

digant73, 2023-10-08:
It's up to you to provide another PR difficult to be merged. The request from radek8 was lawful and involving a specific menu very easy to be satisfied. FR received a suggestion from what you consider an enemy and then you put your effort on changing something your enemy referred to instead of what has been requested by this FR adding/justifying your changes with your usual non sense words or bug reports. I usually provide/ed support to all users requesting help, clarification and support for improving the project. You were also one of them before you started to manifest your attitude cutting by yourself the bridge with all the rest of the team even banning people here and there. I'm not sorry you are ridiculing yourself, to say at least, I'm sorry you put more effort on providing spites in the project most of the time. Here, simply another example

kisslorand, 2023-10-08:
Most certainly you have the right to your own opinion.

I will not lower myself to your level and call you names, anyone around here mature enough can see what's there to see.

As a side note, thank you very much for the free publicity, bringing the attention towards me will help more and more people to  try the alternative FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW) and the result of the work I've done there would be the truth, not what random trolls are mumbling on the internet. The only thing people care about is to have finished prints done as perfectly as possible. [The proof is in the pudding.](https://www.dictionary.com/e/slang/the-proof-is-in-the-pudding/)

rondlh, 2023-10-08:
@kisslorand Nobody cares about your buggy and slow repository, you must be out of your mind to even consider it. Your code is of very poor quality, because you do not communicate, coordinate, test or do peer reviews. Of course the outcome cannot be taken seriously. Your false performance claims have been debunked over and over again. **Don't forget to re-enable the ban!**

PR #2873 (2024-01-15): cumulative patches + cleanup
**IMPROVEMENTS:**
- cumulative patches/improvements: it includes some fixes/improvements provided up to PR #2844. They have been refined/fixed here and there to be fully working.  All credits to kisslorand
- cumulative cleanup: cleanup on different files just to remove from _.h_ files useless references to functions invoked only internally on related _.c_ files. Order of functions have been changed in _.c_ files just to make cleanup on related _.h_ files. _inline_ function qualifier used when needed to reduce code size
- other minor improvements/cleanup

**BUGFIXES [... cut]
closed 2024-01-15 by bigtreetech