bigtreetech/BIGTREETECH-TouchScreenFirmware #2858: [BUG] (short description)

lightmaster, 2023-10-20:
### Description

I haven't used my printer in about 8 months now. Trying to build this now and I'm getting the following error for a TFT43_V3. Seems its a issue with the if blocks in parseACK.c, specifically for when `DELTA_PROBE_TYPE != 0`. Not too familiar with the coding, but that's the spot where the first error is poping up at.

```
TFT/src/User/API/parseACK.c: In function 'parseACK':
TFT/src/User/API/parseACK.c:939:9: error: expected '}' before 'else'
         else
         ^~~~
TFT/src/User/API/parseACK.c:369:10: warning: variable 'avoid_terminal' set but not used [-Wunused-but-set-variable]
     bool avoid_terminal = false;
          ^~~~~~~~~~~~~~
TFT/src/User/API/parseACK.c:949:5: error: 'else' without a previous 'if'
     else if (ack_starts_with("M200"))
     ^~~~
Compiling .pio/build/BIGTREE_TFT43_V3_0/src/src/User/Fatfs/ffsystem.o
TFT/src/User/API/parseACK.c:1398:10: error: 'avoid_terminal' undeclared (first use in this function); did you mean 'menuTerminal'?
     if (!avoid_terminal && MENU_IS(menuTerminal))
          ^~~~~~~~~~~~~~
          menuTerminal
TFT/src/User/API/parseACK.c:1398:10: note: each undeclared identifier is reported only once for each function it appears in
TFT/src/User/API/parseACK.c: At top level:
TFT/src/User/API/parseACK.c:1422:1: error: expected identifier or '(' before '}' token
 }
 ^
*** [.pio/build/BIGTREE_TFT43_V3_0/src/src/User/API/parseACK.o] Error 1
```

### Steps to reproduce

1. Try to build when using a Delta printer
2. Fails to build with the error messages above

**Expected behavior**

**Actual behavior**

### Hardware Variant

TFT43_V3

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.

https://pastebin.com/uYdfFTWi

* Provide pictures or links to videos that clearly demonstrate the issue.

digant73, 2023-10-23:
to temporary fix the issue simply replace lines starting from line 903 with the following:

```
        {
          popupDialog(DIALOG_TYPE_SUCCESS, LABEL_DELTA_CONFIGURATION, LABEL_EEPROM_SAVE_INFO,
                      LABEL_CONFIRM, LABEL_CANCEL, saveEepromSettings, NULL, NULL);
        }
        else
        {
          popupReminder(DIALOG_TYPE_SUCCESS, LABEL_DELTA_CONFIGURATION, LABEL_PROCESS_COMPLETED);
        }

```

closed 2023-10-31 by bigtreetech

lightmaster, 2023-11-01:
It builds fine now. Thanks.