bigtreetech/BIGTREETECH-TouchScreenFirmware #2872: TFT24 V1.1 does not enable 2 extruders

darfij, 2023-12-11:
Configuration: TFT24 V1.1, SKF V1.4 Turbo, Marlin 2.0.x
Second extruder and second hot end successfully configured in Marlin and can be seen in the Marlin display
However the graphical display is still having only one extruder and one nozzle.

-upgraded TFT24 V1.1 to newest firmware (V1.1.27)
-config.ini changed (hotend_count:2, ext_count:2)

Marlin reports:

SENDING:M503
echo:  G21    ; Units in mm (mm)
echo:  M149 C ; Units in Celsius
echo:Filament settings: Disabled
echo:  M200 D3.00
echo:  M200 T1 D3.00
echo:  M200 D0
echo:Steps per unit:
echo: M92 X160.00 Y160.00 Z2400.00 E360.00
echo:Maximum feedrates (units/s):
echo:  M203 X500000.00 Y500000.00 Z200.00 E25.00
echo:Maximum Acceleration (units/s2):
echo:  M201 X1500.00 Y1500.00 Z100.00 E200.00
echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
echo:  M204 P1000.00 R200.00 T3000.00
echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
echo:  M205 B20000.00 S0.00 T0.00 J0.01
echo:Home offset:
echo:  M206 X0.00 Y0.00 Z0.00
echo:Hotend offsets:
echo:  M218 T1 X61.50 Y0.00 Z0.000
echo:Unified Bed Leveling:
echo:  M420 S1 Z0.00
Unified Bed Leveling System v1.01 active
Active Mesh Slot: -1
EEPROM can hold 7 meshes.
echo:Material heatup parameters:
echo:  M145 S0 H180 B70 F0
echo:  M145 S1 H240 B110 F0
echo:PID settings:
echo:  M301 P30.96 I4.33 D55.39
echo:  M301 P30.96 I4.33 D55.39
echo:Z-Probe Offset (mm):
echo:  M851 X6.00 Y37.00 Z0.00
echo:Tool-changing:
echo: Z2.00

>>>M115
SENDING:M115
FIRMWARE_NAME:Marlin 2.0.x (Bigtree) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:KOCKA3 EXTRUDER_COUNT:2 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Cap:SERIAL_XON_XOFF:0
Cap:BINARY_FILE_TRANSFER:0
Cap:EEPROM:1
Cap:VOLUMETRIC:1
Cap:AUTOREPORT_TEMP:1
[Configuration.h.txt](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/13633373/Configuration.h.txt)

Cap:PROGRESS:0
Cap:PRINT_JOB:1
Cap:AUTOLEVEL:1
Cap:Z_PROBE:1
Cap:LEVELING_DATA:1
Cap:BUILD_PERCENT:0
Cap:SOFTWARE_POWER:0
Cap:TOGGLE_LIGHTS:0
Cap:CASE_LIGHT_BRIGHTNESS:0
Cap:EMERGENCY_PARSER:0
Cap:PROMPT_SUPPORT:0
Cap:AUTOREPORT_SD_STATUS:0
Cap:THERMAL_PROTECTION:1
Cap:MOTION_MODES:0
Cap:CHAMBER_TEMPERATURE:0

radek8, 2023-12-13:
All variants of the TFT display use the same FW.
I use TFT 35 E3 V3.0, Marlin 2.1.x from (4/10/2023)
I have two hot-ends and the display shows me two nozzles.
The problem will be in your configuration, or you are using an old version of Marin

thisiskeithb, 2023-12-20:
> FIRMWARE_NAME:Marlin 2.0.x (Bigtree) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin 

You should definitely update to upstream Marlin since we're up to `2.1.2.1` which includes a lot of improvements, including Input Shaping & Linear Advance fixes for TMC2208/TMC2225 drivers.

Note: Marlin `2.1.1` is listed as the minimum requirement under the [Marlin Dependencies section of the README](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware#marlin-dependencies) as well.

darfij, 2023-12-22:
Thanks Keith, for your reply
Upgraded to Marlin 2.1.21 still same result.
Can change icons and fonts via display SD card, but no mention of second extruder/hotend in the graphical display.
Rebooting or reloading display firmware brings no change. 
Switching to Marlin display shows both with full functionality.

digant73, 2024-01-15:
> Thanks Keith, for your reply Upgraded to Marlin 2.1.21 still same result. Can change icons and fonts via display SD card, but no mention of second extruder/hotend in the graphical display. Rebooting [...]

Are you referring to the info displayed on top of the display in the `status screen` menu (the initial menu you see at TFT startup)? If so, do you have also a chamber enabled (so 2 hotends + chamber + bed)? if so, your issue should be fixed by the current TFT fw (released today)

closed 2024-03-26 by github-actions[bot]