bigtreetech/BIGTREETECH-TouchScreenFirmware #2874: [FR] Cold extrusion collaboration with M302

Tannoo, 2023-12-20:
**Is your feature request related to a problem? Please describe.**
Trying to test the stepper direction upon a wire replacement and new terminations, the TFT is preventing cold extrusion. M302 P1 disables cold extrusion prevention in Marlin, but the TFT will not allow this. Setting M302 S0 to allow extrusion at any temp, the TFT will not allow less than 20 deg.

**Describe the solution you'd like**
Remove the min temp setting entirely, or co-exist with M302 settings, give an option for disabling the cold extrusion monitoring (maybe even temporarily?)

**Describe alternatives you've considered**
Marlin mode works just fine and allows M302 to work as intended. The Touch screen mode does not. 

**Additional context**
Would like to have consistency. Would love to not have to go to marlin mode to do anything. (Meaning to remove those EXP cables entirely.)

Tannoo, 2023-12-20:
I tried to lower the min temp to -50 to get this setting out of the way of M302. Not having any luck.

In `config.h`:
```
#define MIN_TOOL_TEMP           -50  // extruder temp less than this will not be parsed.
```
I got the TFT now telling me to heat the hotend to 65486!! LOL I guess it is not a floating number and the -50 has rolled under and back over.

Tannoo, 2023-12-20:
My T0 is not connected yet. Marlin lets me test the extruder motor by disabling the cold extrusion via `M302 P1`.

The TFT doesn't honor it.

Tannoo, 2023-12-20:
Here is some backstory for some context.

I have a new Anycubic Kobra Max printer.
I got fustrated with the hacked Marlin firmware that is only compilable via Keil uVision programing software.
The fustration is wanting to customize the firmware options and some were removed or just didn't work. It is also based on 2.0.1. Marlin is well beyond that now.

So, I have decided to change the mainboard to an SKR 3 EZ. This required re-wiring and re-terminating just about everything. I got the Y motor tested before running the wires for the other motors. Testing what I can as I go. Most has been done via the TFT touch mode.

I am currently still getting the toolhead terminated and will be testing those options soon. Like the heater and thermistor. Then the BL Touch that I am adding.

That is another thing. Where is the BL Touch test menu?

rondlh, 2024-02-28:
Check the Cold Extrusion section in your TFT config.ini. The TFT is not aware of the cold extrusion setting in Marlin, I don't think it is reported. The default minimum extrusion temperature set there is 180 degrees, which is active independent of what you setup in Marlin. You can lower the temperature there as requried.

```
#### Cold Extrusion Minimum Temperature
# Minimum temperature needed to extrude/retract.
# Any extrusion/retraction below this temperature will be prevented.
#   Unit: [temperature in °C]
#   Value range: [min: 20, max: 1000]
min_temp:180
```

petaflot, 2024-02-28:
@rondlh  yes it is reported, just issue `M302` and it will tell you if cold extrudes are enabled and what is the temperature threshold.

Tannoo, 2024-02-28:
And.... I have moved on to Klipper.

closed 2025-02-19 by github-actions[bot]