bigtreetech/BIGTREETECH-TouchScreenFirmware #2875: [BUG] BLTouch REPEAT crashes machine

Tannoo, 2023-12-21:
When trying to do the repeatability test from the Touch display, it homes X and Y then tries to home Z but ignores the Zmin limit (not the probe), hence, crashes the nozzle into the bed.

This does not happen in Marlin mode. Marlin mode does not home the machine. It just executes the M48 where it sits.

### Steps to reproduce

1. Power on the machine or kill the steppers.
2. Select REPEAT icon.
3. Watch it crash.

**Expected behavior**
Home and return to a safe location before testing the probe (ex: X100 Y100) and not crash.

**Actual behavior**
Homes X and Y and crashes while trying to home Z.
G28 works fine in all other instances.
M48 will not execute while X and Y are at home. Errors with probe out of bounds.
M48 will not execute if machine is not homed first.

I can manually execute G28. All fine.
Then execute G1 X100 Y100 and it moves the probe to a probable location.
Then execute M48 and the repeatability test executes as it should.

### Hardware Variant

Board: BIGTREETECH_GD_TFT#%_E#_V3.0

Firmware: V27.x Dec 20 2023 in 120Mhz
SPIFlash: Used 96.83% (7.75MB/8MB)

System: Marlin bugfix-2.1.x (Dec 21 2023 12:52:56)

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/13746128/Configuration.zip)

Tannoo, 2023-12-21:
Enabling Z_SAFE_HOMING in Marlin should get around this.

rondlh, 2023-12-25:
I assume that both the BLTouch and the Z-min endstop are wired to the motherboard?
If so, then this is Marlin issue, not a TFT issue.
A potential solution would be to also use the BLTouch for homing.

Tannoo, 2023-12-25:
G28 homes just fine from the TFT terminal or the Home icon.

It is not a homing issue aside from the M48 trying to home.

Being that M48 ezecutes just fine from the Marlin mode screen says that Marlin is just fine.

Tannoo, 2023-12-25:
M48 does not home.

If M48 is issued without being homed.
 From the TFT term or from a serial connection, it just says, "Home XYZ first."

Tannoo, 2023-12-25:
If I home XYZ from the TFT, or TFT terminal, or from PC, or from Octoprint, all homes just fine.

I do have a separate Zmin ans BLTouch is only for probing...  not homing.

Tannoo, 2023-12-25:
Once it is homed (Commanded from any source), it runs M48 probing test from TFT terminal, PC, and Octoprint just fine.

Tannoo, 2023-12-25:
Those sources do not re-home for an M48 command.

Tannoo, 2023-12-25:
BUT

Tannoo, 2023-12-25:
The TFT BLTouch Repeat icon DOES re-home all axis and fails homing Z by crashing the nozzle into the bed.

EVERY SINGLE TIME!

Now... HOW IN THE $&#*@^$&@&@ IS THIS A MARLIN ISSUE?

Tannoo, 2023-12-25:
Btw.... Deviation has been calculated at .000806

kisslorand, 2023-12-25:
The "Repeat" command on the TFT issues a `G28` and than an `M48` command, nothing more, nothing less.
I do not see it as a bug, it just doesn't fit your printer. I do not see it being an issue of the TFT if your printer crashes the nozzle into the bed when either G28 or M48 is called.

What happens if you issue a `G28` and a `M48' one after the other in your Octoprint/terminal/etc?

Tannoo, 2023-12-25:
I did disable the G28 command to make this consistenat with all other sources.

kisslorand, 2023-12-25:
It's great that you found the solution that fits your needs.

Tannoo, 2023-12-25:
Now, with the machine un-homed, the BLTouch Reset Icon produces the `Home XYZ first`.

So, go back to the Home all icon and it homes X and Y, then crashes Z.

Tannoo, 2023-12-25:
Okay. I am going to take this up with the marlin team.

The same crash will happen if M48 is issued first and then G28. Then it is expecting to home with the probe and it is not deployed, not used for homing.

Thanks for letting me vent this out!

Tannoo, 2023-12-25:
> It's great that you found the solution that fits your needs.
> 
> BTW, is your probe offset properly set up in Marlin?

Yes it is and it does perfor the repeatability test just fine.

I am unsure why the repeat icon will crash but run the G28 and M48 back to back from the terminal works just fine.

Tannoo, 2023-12-25:
Probe setup... well, it homes just fine all the time when not trying to run M48.

Tannoo, 2023-12-25:
Issued from all sources, G28 fails by failing to home Z using the Zmin switch but ONLY after getting the M48 failed message, "Home XYZ first".

Why does the BLTouch Icon have the same issue but issuing G28 then M48?

I can issue G28 and M48 with no issues from other sources.

MrKuskov, 2024-02-20:
В прошивке Марлин сделай замену
#define RESTORE_LEVELING_AFTER_G28
//#define ENABLE_LEVELING_AFTER_G28
на
//#define RESTORE_LEVELING_AFTER_G28
#define ENABLE_LEVELING_AFTER_G28

Tannoo, 2024-02-20:
I have moved on. No longer using Marlin on this machine.

PR #2901 (2024-02-28): Bugfixes on top of last merged PR
**BUGFIXES:**
- Fixed #2895: Bug introduced by #2889 and affecting TFT variants with a LED on knob (e.g. TFT35 E3 V3.0 etc.) in case `knob_led_idle` setting on `config.ini` file was enabled. The bug caused a freeze exiting from PID/MPC menus or during the `Event LED` process, if enabled, at the beginning of a print. The bug was due to the missing `volatile` qualifier on `ms` attribute of OS_COUNTER data struct.
  **NOTE:** The bug severity can be considered major for the affected TFT variants
- Fixed #2898: Bug causing a freeze entering on `Preheat` menu. The  [... cut]
closed 2024-02-28 by bigtreetech