bigtreetech/BIGTREETECH-TouchScreenFirmware #2877: @BEDLEVELVISUALIZER and End of Print Issues

Spider1179, 2023-12-28:
I am having 2 issues with neither of them being major and I'm sure there are just a couple settings I am missing. The first is an unknown command message I get with every print. I have searched around and cannot figure out how to solve for this uncommand. 
[image]

The second issue is relating the completing prints. At the end of each print the timer on the display keeps going and I have to hit Stop manually. How do complete the print as normal?

kisslorand, 2023-12-28:
You must have one single empty line at the end of the gcode file. Some slicers put more than one empty line at the end of the gcode file (ex. Orca Slicer). It's known issue/bug, I am on it, it will be fixed.
For a very long time it was standard for the slicers to put one empty line at the end of the gcode file, some of them do not respect it anymore...

Spider1179, 2023-12-28:
Gotcha. I was able to find the bed level issue in my slicer setting and remove.

kisslorand, 2023-12-31:
I couldn't reproduce your end of print issue. Would you be so kind to upload any gcode file you have an issue with?

Spider1179, 2023-12-31:
Here is an example code. I found that if I manually edit the code to delete 2 extra blank lines it completes like expected. I've been playing in Orca trying to figure out if there is a way to keep the extra lines from showing but no luck so far. 
[wall-honeycomb-bambu-211x248-fixed_PETG_2h25m.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/13801745/wall-honeycomb-bambu-211x248-fixed_PETG_2h25m.zip)

kisslorand, 2023-12-31:
I still cannot reproduce it, even with the file you provided. What FW are you using on your TFT?

Spider1179, 2024-01-01:
Here is the version info

[image]

kisslorand, 2024-01-01:
Based on the picture above you have a TFT with GD32F305 MCU. This repository doesn't offer support for that TFT with the mentioned MCU. Based on the date of the TFT's FW, it's most probably you are using the alternative FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW). If it's so than I suggest to take the latest version from there as it has been updated with some bugfixes, one of them being the issue with the gcode files from OrcaSlicer.

Spider1179, 2024-01-05:
I finally got around to trying out updated FW from your repository and it worked like a charm. Thanks for your help

kisslorand, 2024-01-05:
Thanks for the feedback, I am glad I could be useful.

digant73, 2024-01-10:
@Spider1179 It seems the issue you reported is related to a firmware not provided in this repo. I tested also your file and I confirm that the official fw in this repo is not affected by the end of print issue. As already reported by other users, the `unknown command` error is also not a bug on TFT.
Please, close this ticket just to avoid to make confusion mixing fw provided by third party releasers and eventually open the ticket in the proper repo. Thanks

closed 2024-01-10 by Spider1179