bigtreetech/BIGTREETECH-TouchScreenFirmware #2880: [BUG]  language problem

frth47, 2024-01-01:
language files seem not to match with current firmware version

it is impossible to change language

kisslorand, 2024-01-01:
> language files seem not to match with current firmware version

How did you come to this conclusion?

frth47, 2024-01-01:
when I put the language_fr file on the sd card to update the tft, I have a series of errors:
invalid keyword (about 20 invalid)
i make the same with language_en and have the same issue...

kisslorand, 2024-01-01:
Did you download/use the latest language files?

frth47, 2024-01-02:
i download in the same package. from the current main page

kisslorand, 2024-01-02:
I just tested it:
- downloaded FW file from https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
- downloaded "config.ini" from https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
- downloaded language pack from https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/Language%20Packs

With these files the update just went smooth as butter.

I have no idea what you're doing wrong but the issue you are reporting cannot be reproduced with the current files.

frth47, 2024-01-02:
thank you for your time.
I used the same files you suggested.
the only item I didn't specify is the tft file: BIGTREE_GD_TFT70_V3.0.27.x.bin
I will try your FW.
it's not a very serious problem: I can keep the initial language

rondlh, 2024-02-26:
**WARNING:** Kisslorand's closed source firmware is not recommended, it's buggy, outdated and might damage your printer and/or display. His PRs here are not merged because they are buggy, slow and not peer reviewed, just have a look here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pulls.
Kisslorand's firmware might cause data corruption of the TFT configuration in the TFT.

kisslorand, 2024-02-27:
WARNING: There's a troll between us with ZERO contribution to this repository spreading false claims without any proof other than his rabid bark.

His hatred started when I refused to share with him the source code of how I made the FW for these TFTs more reliable than the one in this repository. The proof is all around in this repository how he praised my work until I declined his request to release the source code for the "Hesitation Guard" feature that I came up with and implemented. One must wonder why he wanted so badly the "buggy and slow" source code...
He teamed up with Mr. Benito and came up with the "brilliant" idea to implement the deprecated ADVANCED_OK feature that absolutely not a single 3D printer manufacturer uses, no remote controller solution (ESP3D, Octoprint, Pronterface, etc) uses. No remote printing capable slicer uses that neither. Even thou he claims he can analyze someone else's code with a simple glance, he had to team up with somebody because he was plain simple not up to the task. There are a few unfinished and/or non-functional code attempts from him and one single pull request that hasn't been approved by the maintainers of this repository. He lives in the shadow of Mr. Benito, it's him who happily picks up the garbage he left and makes it functional.

That's the puppy of Mr. Benito spreading his "wisdom" from the safety of his master's shadow.

Go Sparky, go! Woof, woof!
Make Il Duce happy! Woof, woof

PR #2901 (2024-02-28): Bugfixes on top of last merged PR
**BUGFIXES:**
- Fixed #2895: Bug introduced by #2889 and affecting TFT variants with a LED on knob (e.g. TFT35 E3 V3.0 etc.) in case `knob_led_idle` setting on `config.ini` file was enabled. The bug caused a freeze exiting from PID/MPC menus or during the `Event LED` process, if enabled, at the beginning of a print. The bug was due to the missing `volatile` qualifier on `ms` attribute of OS_COUNTER data struct.
  **NOTE:** The bug severity can be considered major for the affected TFT variants
- Fixed #2898: Bug causing a freeze entering on `Preheat` menu. The  [... cut]
closed 2024-02-28 by bigtreetech