bigtreetech/BIGTREETECH-TouchScreenFirmware #2881: [BUG] (Display menu names)

radek8, 2024-01-05:
I found a problem in displaying the menu names.
I will choose a secondary language (tested in CZ and DE)
Enter the menu. The name of the menu should appear in the upper left corner. Instead of the name, there is a 10% sign.
if i select the language english the menu name is displayed correctly.

DE

[image]

[image]

EN

[image]

[image]

kisslorand, 2024-01-05:
This is a very old bug, I knew about it. Actually it was already fixed in the FW in [my repository](https://github.com/kisslorand/BTT-TFT-FW).
I am sick and tired of making bugfixes not to be merged and than copied by others in their PRs.

rondlh, 2024-01-09:
@radek8 Thanks for the bug report, I will look into it.

@kisslorand Your bugfixed do not get merged because you (force)push buggy, slow and faulty code and don't do peer review resulting in your repository becoming a buggy unsupported mess, making your repository closed source cannot hide that.

digant73, 2024-01-09:
@rondlh the bug has been identified and it will be fixed on the existing PR plus other bug fixes found on the same piece of code. The major bug was in `drawTemperatureStatus` function that was calling a useless code `GUI_StrPixelWidth(LABEL_10_PERCENT); `

rondlh, 2024-01-10:
> @rondlh the bug has been identified and it will be fixed on the existing PR plus other bug fixes found on the same piece of code. The major bug was in `drawTemperatureStatus` function that was calling a useless code `GUI_StrPixelWidth(LABEL_10_PERCENT);

@digant73 Right, I also found it and was just about to ask you to delete the line, thanks!

@radek8 For a quick fix just delete the line "GUI_StrPixelWidth(LABEL_10_PERCENT);" at the end of PersistentInfo.c, the problem also doesn't occur if you disable persistent info.

kisslorand, 2024-01-13:
Sparky and Il Duce... inseparable...

PR #2873 (2024-01-15): cumulative patches + cleanup
**IMPROVEMENTS:**
- cumulative patches/improvements: it includes some fixes/improvements provided up to PR #2844. They have been refined/fixed here and there to be fully working.  All credits to kisslorand
- cumulative cleanup: cleanup on different files just to remove from _.h_ files useless references to functions invoked only internally on related _.c_ files. Order of functions have been changed in _.c_ files just to make cleanup on related _.h_ files. _inline_ function qualifier used when needed to reduce code size
- other minor improvements/cleanup

**BUGFIXES [... cut]
commit 156e04b6 (2024-01-15): cumulative patches + cleanup (#2873)

* cumulative patches + cleanup
* fixed #2881 and wrong buffer allocation
* global fix on Level Corner menu
closed 2024-01-15 by bigtreetech