bigtreetech/BIGTREETECH-TouchScreenFirmware #2884: [BUG] BTT TFT35 E3 v3.0.1

ant0nyk1ng, 2024-01-16:
Trying to update the firmware for the BigTreeTech TFT35 E3 v3.0.1 with the GD32F205 chip using the recompiled BIGTREE_GD_TFT35_V3.0_E3.27.x.bin has left me with a black screen and after the initial firmware load.

I have tried 3 different SD card all working and re formatted.

It starts and you can see the firmware writing... it gets to 100% and then the screen goes black. Nothing else happens.
When I remove the SD card and put it back into my computer the .bin file has not changed to .cur nor has the Config.ini and TFT32 Folder.
I have used it with the config.ini both modified and then the original unmodified. Every time it stop with black screen after firmware writing shown on the screen.

thurask, 2024-01-16:
Same issue here.

ant0nyk1ng, 2024-01-18:
Ok after finding version history... I was able to get it working with I think the previous version. 

Here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/6d49c6e16f03dc561ecf3b02c6a45a22976b40a7

Now if you run this you will find the Listening Mode warning. You just turn it off in the menu it seems.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2202

digant73, 2024-01-18:
did you try to compile the code by yourself? Also try to reset the TFT (punt the `reset.txt` file on the SD's root folder)

ant0nyk1ng, 2024-01-18:
@digant73 Yes it loads and does connect to the printer seems to be fine. 

It does also have the TFT Listening Mode warning that has to be set from menu.

digant73, 2024-01-18:
@ant0nyk1ng ok thanks for the check. The bug is present on all GD versions and is going to be fixed on a new PR

baikal, 2024-01-21:
Tried to update the TFT on a Biqu B1 (an old one with STM32 chip on TFT), using BIGTREE_TFT35_V3.0_B1.27.x.bin and config.ini from 2023-DEC-15. Update worked fine, but the Display was in 'TFT Listening Mode!' from the immediately after the update - no clicking so far. Must still be some bug that triggers as 'disconnect' after an update?

kisslorand, 2024-01-22:
Guys, there's a mature FW [here](https://github.com/kisslorand/BTT-TFT-FW).

digant73, 2024-01-22:
> Tried to update the TFT on a Biqu B1 (an old one with STM32 chip on TFT), using BIGTREE_TFT35_V3.0_B1.27.x.bin and config.ini from 2023-DEC-15. Update worked fine, but the Display was in 'TFT [...]

this is ok. `Listening Mode` flag has not been made configurable in `config.ini` but only from TFT GUI. Simply enable/disable `Listening Mode` from TFT (it will be also saved on flash) when needed

baikal, 2024-01-22:
@digant73 Thank you. That is why I wrote this comment, as I was not sure if that glitch would only be fixed for GD-Variantes by #2885 or all of them (mine was a STM32). Yes, I did enable the connection in the menu and that 'sticks'.

digant73, 2024-01-22:
> @digant73 Thank you. That is why I wrote this comment, as I was not sure if that glitch would only be fixed for GD-Variantes by #2885 or all of them (mine was a STM32). Yes, I did enable the connection in the menu and that 'sticks'.

it was a minor bug for all TFT variants (STM and GD). It is now fixed on #2885

closed 2024-01-23 by bigtreetech