bigtreetech/BIGTREETECH-TouchScreenFirmware #2888: Weird errors after firmware update

dyocis, 2024-02-08:
I'm running a BTT SKR 3EZ and a TFT70.  I got all the hardware installed a few weeks ago, and everything went pretty good after I figured out how to use Marlin again.  It's been a few years since I messed with any of this.  

I got a little cocky and decided I wanted to update the TFT after realizing it was running 2 year old firmware.  I pulled the files from this repository and configured the .ini file according to the instructions.  After I uploaded it, I'm getting weird errors and unknown commands, along with no longer being able to print.  My Z won't calibrate at all.  Every time I try to run the probe offset wizard it fails, and I have to reboot the machine.  

I've tried using the reset.txt on the TFT and the problem persists.  I've tried turning off all of the settings I changed in my printer firmware, along with incrementally changing setting on the TFT. Nothing seems to work.  I'm sure it's something I've done, or a setting I have incorrect, but I can't figure out what it is.  I even went as far as verifying all of the cables between the motherboard and tft.  If anyone can help, I'd greatly appreciate it.

Here's my specs:

Ender 5 Plus
SKR 3 EZ  - Marlin 2.1.2.1 (master branch)
TFT70 V3 (GD chip) V.x.x.27 and the TFT70_GD bin file (master branch)

Here's some screenshots:

[image]
[image]
[image]
[image]

I can provide my config files if needed, but I'll need someone to tell me how to upload them, or if I just need to copy and paste. I don't use GitHub a lot so this is all new to me.

dyocis, 2024-02-08:
Here's my config files:

[Firmware.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14212576/Firmware.zip)

kisslorand, 2024-02-08:
I suggest you to try the firmware from [my repository](https://github.com/kisslorand/BTT-TFT-FW).

digant73, 2024-02-09:
@dyocis you had a lot of mandatory and recommended settings not enabled in Marlin.
Attached the changes applied on both Marlin and TFT configuration files. Recompile Marlin using the attached Marlin's files. Load the changed `config.ini` file on the TFT (I only disabled `advanced ok`). If everything will properly work then you could enable `advanced ok` again.

[Firmware.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14219875/Firmware.zip)

dyocis, 2024-02-10:
@digant73 My apologies.  I uploaded the wrong marlin config files somehow.  Here are my current files.  I'd like to understand what I've done wrong, so if you could take a look and tell me, I'd really appreciate it!  

[Firmware_2.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14227923/Firmware_2.zip)

@kisslorand I loaded your firmware, and it all seems to be working correctly.  Thanks!  If digant73 can't help me figure out what I did wrong, I will definitely be using your code.

digant73, 2024-02-10:
@dyocis Marlin's configuration files are ok. Did you try do disable advanced ok (`advanced_ok `set to 0 in `config.ini` or from `Feature` menu) on TFT as I previously suggested?

dyocis, 2024-02-10:
@digant73  Thanks for taking a look at them.  Yes, I did that during my initial troubleshooting efforts. It didn't seem to change anything.  But I'll go in and try it again, and let you know what I find.

dyocis, 2024-02-11:
@digant73 - I used your config.ini file from above, and the same issue occured.  I'm still getting unknown commands and I can't use the probe offset wizard.  If you have any other suggestions, please let me know.

@kisslorand - I noticed something while loading your firmware.  The install screen is rotated 180 deg. It's not really a big deal, it goes back to normal after it loads, but I thought I'd let you know.  Here's a pic.
[image]

kisslorand, 2024-02-12:
@dyocis 
Since you are just updating the firmware, it (the new FW) cannot be responsible for anything going on. It's the old/previous firmware doing the display of the update progress. Just update again over the new FW, check if it's still upside down.

rondlh, 2024-02-17:
@dyocis Please be careful with kisslorand's firmware, it's closed source, slow, buggy and might damage your printer and/or display. He's not a reliable player. It's best never to install a firmware from a unreliable source. @digant73 is the correct person to talk to.

kisslorand, 2024-02-20:
Get your popcorn, Sparky is putting up a show again! :)

rondlh, 2024-02-23:
@kisslorand Please cleanup your outdated PRs? You are polluting this repository

kisslorand, 2024-02-23:
Nah, I'm not in the mood today. I'd rather sit back and enjoy the show. It's so amusing watching how for every new issue the first advice is "turn off ADVANCED_OK". Your hatred started when I warned about "ADVANCED_OK", now it's your cocktail so drink it!
It's also a great source of entertainment watching how you're making fool of yourself by considering everyone a stupid puppy as yourself thinking that someone with ZERO contribution has real credibility bashing me and my work without any proof. My proof is my work, yours is only your rabid bark.
   Don't get me wrong I am not mad at you, I know it's not your fault. God must have been in a mood for some grotesque jokes and He made you. :)
On another note, if you are so concerned about pollution you should shut your mouth, that's where the biggest pollution comes from.

Come on Sparky! Woof, woof!
Make Il Duce happy! Woof, woof!

dyocis, 2024-02-27:
@kisslorand Sorry for the slow response, I've been out of town for work.  Everything seems to be working well with your firmware, I appreciate it.  

@rondlh I'm not sure what your issue is with kisslorand, or what all of this back and forth is about.  As far as your first comment to me, I haven't seen any response from digant since my last @ to him.  If he has any other info on my issue, or how to resolve it, I'm happy to work with him.  Until then, Kisslorand's fix seems to be working.  I appreciate your input, but in this instance it wasn't helpful.

closed 2024-02-27 by dyocis

digant73, 2024-02-28:
@dyocis a new TFT fw has been released today fixing some bugs. You can try it. Also remember to set `serial_port` in `config.ini` so  both the TFT's and mainboard's baudrates are the same (the issue you reported seems to be more a mismatch of baudrate)