bigtreetech/BIGTREETECH-TouchScreenFirmware #2892: x y offset is ignored when I probe for Z[BUG] (short description)

demetan71, 2024-02-18:
Hi, my first post

I have an Anycubic Chiron with SKR 1.4 Turbo and Tft35 E3 V3

The problem is, I put in marlin firmware home offset for x and y axis, and Since I use #define HOMING_Z_WITH_PROBE I had to put x and y offset on lcd.  The value I put in firmware is completely ignored.....

Before, when I use optical endswitch for z, it was probe offset that was ignored......

Any idea??

kisslorand, 2024-02-19:
Try the FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW).

digant73, 2024-02-20:
@demetan71 did you disable `probing_z_offset` in config.ini? if not, try to disable it. Not sure what is your setup (you need home offset in your printer or you tried home offset as a sort of workaround to fix the issue you are reporting). However, home offset didn't work very well in Marlin. If you don't need home offset (you need only probe offset), never define home offset (it probably causes side effects here and there and it is not under TFT control).

demetan71, 2024-02-20:
Thank for your answer, but I don't have this option in my config file....

But now it seem's to work but in my lcd tft35 even if in my firmware my home offset is set, in lcd home offset are at 0, but seem's to work....

But my main problem now, is since I home z with probe I can't use G34 auto align, do the move and everything, but doesn't compensate at all...

digant73, 2024-02-20:
if you don't have the `probing_z_offset` setting on your TFT's `config.ini` file then you have a very old TFT fw. Update to latest one and possibly the issue is fixed

demetan71, 2024-02-20:
not shure I understand!!!  

I don't have probing_z_offset in my config.ini, but in configuration.h I had set, on my tft35 I have the option too.....

Anyway like I said, I put home offset in marlin configuration.h seem's to work even if nothing apear in home offset menu on lcd tft35...

I'm using SKR 1.4 Turbo tft35 E3 V3 wiith Marlin 2.1.x

I Will delete this post....  My main concern now is to align Z that is not working since I probe for z

digant73, 2024-02-21:
@demetan71 I mean that `probing_z_offset` setting is present on TFT's `config.ini` file since 2 years at least so if it is not present on your `config.ini` file then your TFT firmware is very outdated and you must first update the TFT firmware to the latest one before opening bug reports

demetan71, 2024-02-21:
[digant73](https://github.com/digant73) Euh, not shure I understand, my firmware is too old???  Where I can find newer version, because look at the picture....  Everything I find is more than 2 years old!!!  I really like new version!!!
[image]
[image]

demetan71, 2024-02-21:
[digant73](https://github.com/digant73)  Maybe I wasn't clear.  In my marlin configuration, I have home offset set.  When I use enstop for homing, everything was fine, and when I look in the TFT35 menu I see the same offset that I put in Marlin writing on the LCD.  Since I homing Z with probe in my lcd home offset menu x y x are at zero.......

I spent hours ans hours try to understang how everything work....  For example, the second print I start it retract filament so far that it came out of the extruder.  Sometime it said revent too long extrusion, but not everytime....   

It's ok, I can do fine print, but I don't understand how everything work between TFT35 and SKR motherboard.

I can run the machine but got to watch everytime I start a print......

But, I wanna say thank you for your help....   Did I say that I'm a french canadian, so excuse my poor english...

digant73, 2024-02-22:
Read the sections on this links http://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
to understand how to update the TFT fw.
Basically you have to provide on an SD card:
- `BIGTREE_TFT35_V3.0_E3.27.x.bin`
- `config.ini`
- `TFT35` folder with all icons and fonts
You have to edit first the `config.ini` file providing proper setting for your printer

You can download the needed files reported above here https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update

Flash this latest TFT fw and verify if it works

demetan71, 2024-02-22:
@digant73 digant73 just did the update....

Now I have 3 invalid keyword.....

And yes I define advance_ok on configuration adv! 
[image]
[image]
[image]

digant73, 2024-02-22:
All of those settings are present in the `config.ini` file here 
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14375613/config.zip)
They must not be provided in Malrin's configuration files (e.g Configuration.h or Configuration_adv.h) as you did.
You have to properly set the values of the different settings present in `config.ini` file according to your Printer. Make sure that all the Marlin settings reported at the beginning of `config.ini` file are enabled in Marlin fw (e.g. `EEPROM_SETTINGS` (in `Configuration.h`) in order the TFT can properly communicate/work with Marlin.

demetan71, 2024-02-22:
[digant73](https://github.com/digant73)

Done, everything, stay the same!!!  3 invalid keywords.....

digant73, 2024-02-22:
Not sure you are familiar to properly use e git repo. Attached the remaining resources you need to update your TFT fw. Unzip both zip files and copy the content on a SD card (formatted to FAT32) with the `config.ini` file I already provided to you.

[BIGTREE_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14376417/BIGTREE_TFT35_V3.0_E3.27.x.zip)
[TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14376420/TFT35.zip)

demetan71, 2024-02-22:
[digant73](https://github.com/digant73)

Sorry, samething.... 3 invalid keyword

digant73, 2024-02-22:
but was the TFT fw (the file `BIGTREE_TFT35_V3.0_E3.27.x.bin`) installed before? or do you see only the config update process? The The sequence is:
1. TFT fw `BIGTREE_TFT35_V3.0_E3.27.x.bin` is loaded and installed
2. `TFT35` content (icons and fonts) are loaded and installed
3. `config.ini` file is loaded

demetan71, 2024-02-22:
Yes, I just comment # advanced_ok  tx_slot and checksum.....

So no error.....   But before I did everything you said, auto align Z do is thing but not compensate, and now probing fail.... So It's worst!!!

But correct me if I'm wrong, if the firmware is already the lst one, it will not update? Because when I put only BIGTREE_TFT35_V3.0_E3.27.x.bin that does nothing, so it mean I have the lastest firmware?

demetan71, 2024-02-22:
> Try the FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW).

Why???  Did you modified something??

digant73, 2024-02-22:
if you have a TFT35 V3 E3 the BIGTREE_TFT35_V3.0_E3.27.x.bin is always loaded and installed and you will not have the errors you reported during the config process.
So I suspect that you are reporting a wrong TFT35 model. You probably have a GD version so the proper TFT fw to use is this 
[BIGTREE_GD_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14377778/BIGTREE_GD_TFT35_V3.0_E3.27.x.zip)

demetan71, 2024-02-22:
How can I know? 
[image]

demetan71, 2024-02-22:
[digant73](https://github.com/digant73)

How can I get mcu report???

kisslorand, 2024-02-22:
> [kisslorand](https://github.com/kisslorand)
> 
> > Try the FW from [my repository](https://github.com/kisslorand/BTT-TFT-FW).
> 
> Why??? Did you modified something??

Other than speed increase, reliability enhancements, Marlin compatibility, bugfixes... not much. :)
The main feature is the "Hesitation guard" implemented in my version of the FW. It is a dynamic guard during printing to reliably ensure a constant flow of gcodes to avoid any hesitation during printing that results in blobs or other artifacts/defects.
A more detailed description is provided in [my repository](https://github.com/kisslorand/BTT-TFT-FW).

demetan71, 2024-02-24:
[digant73](https://github.com/digant73)

I have a lot of trouble, worst and worst, I didn't count the number of time I uplaod firmware, but I tried BIGTREE_GD_TFT35_V3.0_E3.27.x.bin and just realized that It stock with this firmware and can't put another one!!!! Wierd!!  And I have a question, it's not possible update firmware without updating font and bmp??

digant73, 2024-02-26:
@demetan71 I strongly suggest you to not update bootloader (not needed at all). You should be able to reinstall the TFT fw as many times you want.

demetan71, 2024-02-26:
[digant73](https://github.com/digant73)

Like I said, since I put GD firmwawre version I can't put any other firmware and GD is not the good one....

kisslorand, 2024-02-27:
WARNING: There's a troll between us with ZERO contribution to this repository spreading false claims without any proof other than his rabid bark.

His hatred started when I refused to share with him the source code of how I made the FW for these TFTs more reliable than the one in this repository. The proof is all around in this repository how he praised my work until I declined his request to release the source code for the "Hesitation Guard" feature that I came up with and implemented. One must wonder why he wanted so badly the "buggy and slow" source code...
He teamed up with Mr. Benito and came up with the "brilliant" idea to implement the deprecated ADVANCED_OK feature that absolutely not a single 3D printer manufacturer uses, no remote controller solution (ESP3D, Octoprint, Pronterface, etc) uses. No remote printing capable slicer uses that neither. Even thou he claims he can analyze someone else's code with a simple glance, he had to team up with somebody because he was plain simple not up to the task. There are a few unfinished and/or non-functional code attempts from him and one single pull request that hasn't been approved by the maintainers of this repository. He lives in the shadow of Mr. Benito, it's him who happily picks up the garbage he left and makes it functional.

That's the puppy of Mr. Benito spreading his "wisdom" from the safety of his master's shadow.

Go Sparky, go! Woof, woof!
Make Il Duce happy! Woof, woof

demetan71, 2024-02-27:
[digant73](https://github.com/digant73)

So,  Finally, I still have errors, I still can print from SDcard or USB on TFT, but I enable SDSUPPORT in configuration.h and disable SDCARD connectivity in configuration.ADV and now I see my Micosd card from skr board on my computer so I can dump, file and print from my computer or TFT when I choose the MicroSD card....  So at least I can print.....

Look at my video.....  https://www.youtube.com/watch?v=1GQBCWBQuvs
https://www.youtube.com/watch?v=5wxMC30oBr4

So For trying to fix my offset problem, I got another one instead....

So take a look if you're curious....

Thank you again...

digant73, 2024-02-27:
@demetan71 ok good. The TFT fw is installed. `TFT35` folder (for icons&font) and `config.ini` (for TFT settings) are needed when the TFT fw (the first one installed) will ask for that (a message on the screen will report if you have to provide them). So You usually need only to provide the fw binary (eventually after it is installed, the just installed fw will ask you to provide on the sd card also TFT35 folder and/or config.ini file).
Yes I found also a bug in the current TFT fw that will freeze the fw on the TFT with a LED on the knob (as your TFT). I will let you know when the new fw is available for download. In the meanwhile you should be able to avoid the freeze simply disabling `event_led` and `knob_led_idle` on `config.ini` file (you have to set that settings to 0 in config.ini and of course load the new config putting the file on the sd card and rebooting the TFT to load the new configuration).

demetan71, 2024-02-27:
[digant73](https://github.com/digant73)

Don't understand!!!  The TFT fw is installed???   It's anoter file???   Don't understand

digant73, 2024-02-27:
the official BigTreeTech (manufacturer of your TFT) link for all TFT's resources (TFT fw binary, TFT35 folder and config.ini) is only on this link https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update. As you can see in that link you can see all the TFT fw binaries (the binary for your TFT is `BIGTREE_GD_TFT35_V3.0_E3.27.x.bin`) the `config.ini` (it is independent from the TFT fw binary) and all TFTxx icons&fonts folders (you are using the `TFT35` folder contained on the `Round Miracle Theme` package folder but you could use any TFT35 folder of another package folder).
According to your video the TFT fw is installed (the first percentage from 0 to 100% is related to TFT fw update) but as I said in the previous post, the TFT fw has a bug that will be fixed on a next fw release.

demetan71, 2024-02-27:
Definitively connection problem, It start the process of printing, but almost impossible to do live change always said busy......

Babystep, storing stuff, etc...

demetan71, 2024-02-27:
> the official BigTreeTech (manufacturer of your TFT) link for all TFT's resources (TFT fw binary, TFT35 folder and config.ini) is only on this link:
> 
> https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update
> 
> As you can see in that link you can see all the TFT fw binaries (the binary for your TFT is `BIGTREE_GD_TFT35_V3.0_E3.27.x.bin`) the `config.ini` (it is independent from the TFT fw binary) and all TFTxx icons&fonts folders (you are using the `TFT35` folder contained on the `Round Miracle Theme` package folder but you could use any TFT35 folder of another package folder). According to your video the TFT fw is installed (the first percentage from 0 to 100% is related to TFT fw update and also the binary file name being loaded is displayed on top of the display) but as I said in the previous post, the TFT fw has a bug that will be fixed on a next fw release.

If you said that is installed, I trust you, but I'm sorry, still don't undertand FW I did'nt see it anywere....  Really don't understand!!!
Sorry!

digant73, 2024-02-27:
TFT fw (binary) update on your video start at 16 and is completed at 22. Then it starts the TFT35 (fonts&icons) update and finally the config.ini update

demetan71, 2024-02-27:
I confirm, my printer is not usable, can not do live change...........  Anyway, I will still dig the problem....

Thank you

digant73, 2024-02-27:
try to attached TFT fw. Unzip the .zip file and, put the unzipped .bin file on the sd card and proceed with TFT fw update. It should fix your freezes

[BIGTREE_GD_TFT35_V3.0_E3.27.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14422759/BIGTREE_GD_TFT35_V3.0_E3.27.x.zip)

demetan71, 2024-02-27:
Just did it... 
[image]

Just to help you, I pause a print and I've got this, look at the picture, after screen freeze...

digant73, 2024-02-27:
Is the screen frozen or you can press on `Ok` button? Eventually disable the command checksum (set `command_checksum`  to 0 on `config.ini`)

demetan71, 2024-02-27:
i think I have a problem with my SKR, I had to almost triple E-step on extruder.....

Maybe I short something, somewhere I did redo all my wire, maybe touch something, I don't know.....  but it print...

digant73, 2024-02-27:
I think so. Please close this ticket and possibly even #2896 (it has nothing to do with TFT fw. G34 is entirely managed by the mainboard not by TFT). Once you properly tuned/fixed your printer setup you can open bug/info report here for TFT related topics

closed 2024-02-27 by demetan71