bigtreetech/BIGTREETECH-TouchScreenFirmware #2895: TFT35 freezes before Purging

MrKuskov, 2024-02-20:
### Description

### Steps to reproduce
I am sending a file for printing
[flowrate_0_PLA_7m36s.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14345696/flowrate_0_PLA_7m36s.zip)
all G-code commands pass normally until the M117 Purging command line 291 in the file. Then the screen freezes. If you send it to print from Marlin mode, then the file is printed. But it is not possible to go back to "Touch mode". The mode does not switch when holding the knob. The same happens when printing any files.

**Expected behavior**

**Actual behavior**

### Hardware Variant
 GD_TFT35_V3.0_B1_V27.x
### TFT Firmware Version & Main Board Firmware details

The TFT firmware version from February 20th. From February 17, it worked fine. Marlin 2.1.2.2
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14345832/Configuration.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.

MrKuskov, 2024-02-21:
> If you are on the latest TFT firmware, do you have the new setting `command_checksum` enabled on TFT's `config.ini` file? eventually try to disable it (although it should not be related to the issue) and let me know the result. Thanks
Yes. The function is available, I tried to turn it on and off in the screen settings. my screen settings via the [link](https://github.com/MrKuskov/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini)

> **EDIT:** I tested your gcode file on two printers without any issue. Before `M117 Purging` gcode the nozzle temp needs to reach its target. After it was reached, the print started and terminated without any issue

Yes, in Marlin mode, I also type without problems. Printing freezes after line 
`G1 X6.1 Y20 Z0.3 F5000.0; Move to start position`
 line 290 in the code. When the hotend starts moving to the specified line, the screen stops responding. The "M117 Purging" line is no longer being executed. The command is executed without problems via Pronterface.

MrKuskov, 2024-02-21:
I found out that printing is interrupted in "Touch mode" only if the card is installed in the USB or SD of the screen. Printing is normal when installed in the SD board.

digant73, 2024-02-21:
@MrKuskov try with another SD card, if not already done. I have no issue at all on my printers (5 prints without any problem)

MrKuskov, 2024-02-21:
I flashed the version from January 23rd. https://github.com/MrKuskov/BIGTREETECH-TouchScreenFirmware/blob/2ae78153bf685e901e504a04a88d9b6021450057/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0_B1.27.x.bin Everything works correctly in it.

digant73, 2024-02-21:
@MrKuskov please attach here the `config.ini` file you are using

MrKuskov, 2024-02-22:
No, it doesn't work. The screen froze when the hotend moved to the starting point, but the hotend executed three more commands: completed two cleaning lines and moved to the center of the table.

digant73, 2024-02-22:
@ok thanks, I will continue to investigate. Could you please attach your `config.ini` file?  so I will try to use it

digant73, 2024-02-22:
> I have made a fork from your repository and am making changes to it for both the marlin firmware and the screen firmware https://github.com/MrKuskov/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini

ok. I will use your `config.ini` file to try to replicate the issue

digant73, 2024-02-23:
@MrKuskov Even with your `config.ini` file I have no issue. the only thing I saw is that command M280 at line 282 was skipped by Marlin (I received the `Unknow Command` message) due it is not configured/supported on my printer.

MrKuskov, 2024-02-23:
Sorry I made a mistake. The firmware version is from January 23rd. [This one](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/2ae78153bf685e901e504a04a88d9b6021450057/Copy%20to%20SD%20Card%20root%20directory%20to%20update/BIGTREE_GD_TFT35_V3.0_B1.27.x.bin). I use the ru language pack. But I tried switching to the en version. The screen freezes the same way.

MrKuskov, 2024-02-24:
@digant73 Yes!!! The `Event Led` function is to blame for the screen freezes. When it is turned off, everything works correctly. If the function is enabled during printing, the screen freezes immediately.

digant73, 2024-02-24:
ok thanks. Do you have a LED strip configured in your printer or a simple LED on the hotend?

MrKuskov, 2024-02-24:
Multi-color RGB lights from [Biqu B1](https://biqu.equipment/products/biqu-b1-3d-printer-skr-v1-4-32bit-motherboard-tft35-b1-v3-0-dual-operation-system-fdm-impressora-3d-printer-beginner?variant=31952915005538)

digant73, 2024-02-24:
could you let me know which Marlin settings reported at the beginning of `config.ini` file (on section `Supported Marlin Firmware Versions`) are enabled and disabled on your Marlin?

MrKuskov, 2024-02-24:
I have enabled all the settings specified at the beginning of the config.ini file in my Marlin
[Configuration.h](https://github.com/MrKuskov/Configurations/blob/release-2.1.2.2/config/examples/BIQU/B1%20-%20SKR%201.4/Configuration.h)
[Configuration_adv.h](https://github.com/MrKuskov/Configurations/blob/release-2.1.2.2/config/examples/BIQU/B1%20-%20SKR%201.4/Configuration_adv.h)
in my repositories

MrKuskov, 2024-02-25:
> If the function `Event LED` is enabled during printing, the screen freezes immediately. 
But if I go to the LED menu, I can turn on any backlight colors and the screen works correctly.

digant73, 2024-02-26:
Ok. I will check on `LED Event` code

MrKuskov, 2024-02-27:
@digant73 everything is working correctly. very good. thanks

digant73, 2024-02-27:
ok. but remember that I commented out just a piece of code to have a confirmation of the issue. I will provide a PR with the fix. Leave this bug report open, it will automatically closed once the PR is merged. Thanks for support

closed 2024-02-28 by bigtreetech

MrKuskov, 2024-02-28:
> **EDIT:** Please if possible let me know if the attached fw is ok (that should fix the issue).

Everything works.

digant73, 2024-02-28:
thanks. The fix was also merged in the official btt fw just today.

PR #2901 (2024-02-28): Bugfixes on top of last merged PR
**BUGFIXES:**
- Fixed #2895: Bug introduced by #2889 and affecting TFT variants with a LED on knob (e.g. TFT35 E3 V3.0 etc.) in case `knob_led_idle` setting on `config.ini` file was enabled. The bug caused a freeze exiting from PID/MPC menus or during the `Event LED` process, if enabled, at the beginning of a print. The bug was due to the missing `volatile` qualifier on `ms` attribute of OS_COUNTER data struct.
  **NOTE:** The bug severity can be considered major for the affected TFT variants
- Fixed #2898: Bug causing a freeze entering on `Preheat` menu. The  [... cut]