bigtreetech/BIGTREETECH-TouchScreenFirmware #2896: z auto align G34 not compensate[BUG] (short description)

demetan71, 2024-02-20:
I run a SKR 1.4 Turbo with TFT35 E3 V3
Since I probe for Z G34 sequence do all what he had to do, but doesn't compensate.  Just do movemnet, probe and report...

When I use endstop for Z everything work's fine....

Any idea??

rondlh, 2024-02-24:
It's very difficult to judge the situation from the provided info, but it sounds like a problem with the flux capacitor.

demetan71, 2024-02-24:
Thank's everybody, but, I don't understand....  It's worst, I can't print from sd card, bed heating, after said colling bed, screen freeze...  I'm discourage....  Maybe I'm too noob.....

digant73, 2024-02-27:
TFT fw provides only a button on Menu->Movement->Bed Level->ABL/UBL->Z Align to send a G34 gcode to the printer. To enable this button on the TFT fw you need to set to 1 `z_steppers_alignment` setting on `config.ini` file. To enable the Z auto align on Marlin fw, on Marlin's `Configuration_adv.h` you need to enable `Z_STEPPER_AUTO_ALIGN` setting.

demetan71, 2024-02-27:
https://github.com/digant73

I realized that I maybe post it on the wrong forum.  Yes I know for the button.  But it was working when I used to home with endstop.  Now I probe z for homing, since that, do the move but doesn't correct anything.

And yes everything is enable in marlin firmware

closed 2024-02-28 by bigtreetech

demetan71, 2024-03-01:
https://github.com/digant73

Humm...  this issue was closed, but not resolved..... where can I put it???  Do I miss something??

digant73, 2024-03-01:
it is a possible bug on Marlin so it should be reported here https://github.com/MarlinFirmware/Marlin/issues

PR #2901 (2024-02-28): Bugfixes on top of last merged PR
**BUGFIXES:**
- Fixed #2895: Bug introduced by #2889 and affecting TFT variants with a LED on knob (e.g. TFT35 E3 V3.0 etc.) in case `knob_led_idle` setting on `config.ini` file was enabled. The bug caused a freeze exiting from PID/MPC menus or during the `Event LED` process, if enabled, at the beginning of a print. The bug was due to the missing `volatile` qualifier on `ms` attribute of OS_COUNTER data struct.
  **NOTE:** The bug severity can be considered major for the affected TFT variants
- Fixed #2898: Bug causing a freeze entering on `Preheat` menu. The  [... cut]