bigtreetech/BIGTREETECH-TouchScreenFirmware #2899: TFT35 freezes when exiting the PID calibration menu

MrKuskov, 2024-02-25:
### Description

TFT35 freezes when exiting the PID calibration menu

### Steps to reproduce

1. Go to the PID calibration menu. 
2. Perform calibration.
3. press the "back" button

TFT35 freezes

### Hardware Variant
BIGTREE_GD_TFT35_B1_V3_0

Firmware from February 23rd. I think the problem is that the LED backlight turns on, despite the fact that the "Event LED" menu is disabled. Since I have a problem with the backlight.
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2895

digant73, 2024-02-26:
is the PID process completed successfully (green LED switched on)?

MrKuskov, 2024-02-26:
@digant73 Yes. At first, the LED glows red, then green and suggests writing the values to memory. I click "record" the data is being recorded successfully. After pressing the back button, the screen freezes.

MrKuskov, 2024-02-27:
after flashing [this ](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2895#issuecomment-1965017258)version, everything works correctly. Thanks

closed 2024-02-27 by MrKuskov