bigtreetech/BIGTREETECH-TouchScreenFirmware #2900: [BUG] (My TFT display stopped communicating with the printer)

MartinNovan, 2024-02-25:
### Description
TFT randomly stops communicating or even freeze during printing or just being powered on. I have several errors that are already addressed here in the issues tab, for example when I print from a USB/SD card the display throws random errors such as "unknown command (1 corrupted line from gcode)". When i was printing from SD card the printer just skips layer or just sopped mid print and after a and after a while the Z axis moved down and crushed the model. Or the latest bug that happened to me was that i was printing for sd card on board and display randomly cancelled the print, but printer still was printing. The first error that appeared to me was that if I wanted to perform an offset calibration the tft display froze, this problem disappeared after the first TFT firmware update and did not appear again.

### Steps already taken

1. tried to update TFT firmware (also tried FW from kisslorand)
2. tried 3 marlin FW on board (marlin 2.0. ; marlin 2.1. and also tried bugfix 2.1)
3. tried different board with marlin 2.0 to eliminate that the problem comes from the current board 
4. lowerred the baudrate from 250000 to 115200 that also didnt solve the problem
5. reconnect tft cable
6. turned off advanced_ok 
any of these steps doesnt help

I also tried printing from computer and it did work wonderfully, but i just cant leave my laptop connected to the printer for a longer prints.
**Expected behavior**

TFT display should work normally
**Actual behavior**

It doesnt
### Hardware Variant
¨
I am using BIGTREETECH_TFT35_V3.0.
### TFT Firmware Version & Main Board Firmware details

TFT firmware: version: BIGTREE_TFT35_V3.0_E3.27.bin
SKR mini E3 V3 firmware: marlin 2.1.x "bugfix" 

### Additional Information
I have also tried running stock marlin FW with stock tft FW and problem is still here so it souldnt be caused with configs. 

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/assets/116807577/1cf50806-6f7b-4d17-b9cd-eb064495c161
This is the print where TFT display cancelled the print but it kept going. You can see the errors "USY: PROCESSING", that even the errors are corrupted.

[image]
This is the model, where the printer Z axis went down. You can see the melted hole from the nozzle.

rondlh, 2024-02-26:
WARNING: Kisslorand's closed source firmware is not recommended, it's buggy, outdated and might damage your printer and/or display. His PRs here are not merged because they are buggy, slow and not peer reviewed, just have a look here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pulls.

MartinNovan, 2024-02-26:
I have tried like 3 different sd cards and formated the to FAT32 and any of them doesnt solve the issue, same for USB. The print said it was cancelled like i cancelled it, but i didnt. The print the after like 7h finished without problem but i wasnt able to control the print settings. Screen was responsive and i was able to open menu and go into the terminal but after i send a gcode it freezed completely. These errors on my screen is totaly random, some print starts at the beginning of the print and for some until maybe later. The tempeatures on the tower was ok and hotend changed tempeature like it was supposed to do.

Today i tried to print from SD card again and again there was some bug. 
[image]
The extruder started overextruding like hell (yeah i know 80% infill is much but i need it heavy) but it started to overflowing. also after a while it started to print in the air (again), there are some plastic strings in the back.

after this i started new print but i decided to print in marlin mode and it was ok until i came back and see this on the screen

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/assets/116807577/b8f156ad-6d11-4a30-a3b4-e27f55c86b87

Now the print is still going but the screen is totaly freezed so its something like before. Could it be bad nand flash memory?

kisslorand, 2024-02-27:
WARNING: There's a troll between us with ZERO contribution to this repository spreading false claims without any proof other than his rabid bark.

His hatred started when I refused to share with him the source code of how I made the FW for these TFTs more reliable than the one in this repository. The proof is all around in this repository how he praised my work until I declined his request to release the source code for the "Hesitation Guard" feature that I came up with and implemented. One must wonder why he wanted so badly the "buggy and slow" source code...
He teamed up with Mr. Benito and came up with the "brilliant" idea to implement the deprecated ADVANCED_OK feature that absolutely not a single 3D printer manufacturer uses, no remote controller solution (ESP3D, Octoprint, Pronterface, etc) uses. No remote printing capable slicer uses that neither. Even thou he claims he can analyze someone else's code with a simple glance, he had to team up with somebody because he was plain simple not up to the task. There are a few unfinished and/or non-functional code attempts from him and one single pull request that hasn't been approved by the maintainers of this repository. He lives in the shadow of Mr. Benito, it's him who happily picks up the garbage he left and makes it functional.

That's the puppy of Mr. Benito spreading his "wisdom" from the safety of his master's shadow.

Go Sparky, go! Woof, woof!
Make Il Duce happy! Woof, woof

digant73, 2024-02-27:
@MartinNovan The issue you are reporting under Marlin mode (and based also on the picture you attached) seems to be entirely  due to EMI. You should shield and/or move the TFT's cables far from other components (e.g. stepper motors)

MartinNovan, 2024-02-27:
Update: Yesterday after i got these error i tried to reflash firmware multiple times on both devices and it seems to work! :) Now printing from sd card on board, no issues so far. After this print will be completed i will try to print from TFT. Thank you all for help. I'll let you know how it looks when printing from TFT. So far it seems to just reflashing firmware multiple times helps!

MartinNovan, 2024-02-28:
Tried printing from sd card on TFT but when i start print the TFT freezes after heating up to the right tempeature. Same happens for the USB. Printing from board is working so i will stick for some time with that.

closed 2024-02-28 by MartinNovan

digant73, 2024-02-28:
> Tried printing from sd card on TFT but when i start print the TFT freezes after heating up to the right tempeature. Same happens for the USB. Printing from board is working so i will stick for some time with that.

Update the fw with the latest one (released just today). That bug was fixed in that fw release