bigtreetech/BIGTREETECH-TouchScreenFirmware #2905: [BUG] The bed heating temperature is reset.

MrKuskov, 2024-03-04:
### Description

Each even sending of the model for printing turns off the heating of the bed at the beginning of printing.

### Steps to reproduce

1. I am printing the first model. 
2. I put the second one on the seal. 
3. The bed starts to heat up and after a few seconds the heating stops.
4. The heating of the hot end does not start, as it waits for the bed to heat up.
5. I set the bed heating manually.
6. When the set temperature is reached, the hotend is heated and then printed.
7. We are finishing printing.
8. The following model is printed normally.
9. If you reset the screen, then printing is fine.

**Expected behavior**

**Actual behavior**

### Hardware Variant
TFT35 V3.0 B1 
### TFT Firmware Version & Main Board Firmware details
TFT February 28th Marlin 2.1.2.2

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
my [Configuration.h](https://github.com/MrKuskov/Configurations/blob/release-2.1.2.2/config/examples/BIQU/B1%20-%20SKR%201.4/Configuration.h) and [config.ini](https://github.com/MrKuskov/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update/config.ini)

digant73, 2024-03-04:
can you provide a gcode file affected by the issue?

MrKuskov, 2024-03-04:
it does not depend on the model. Most likely from the initial G-Code
[ring5_PLA_33m3s.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14480635/ring5_PLA_33m3s.zip)
`; BIQU B1 Start G-code
;LAYER_COUNT:[total_layer_count]
M117 Нагрев стола! ; Getting the bed up to temp!
M140 S[first_layer_bed_temperature]; Set Heat Bed temperature
M190 S[first_layer_bed_temperature]; Wait for Heat Bed temperature
M117 Нагрев Хотэнда! ; Getting the extruder up to temp!
M104 S[first_layer_temperature]; Set Extruder temperature
G92 E0; Reset Extruder
G4 P2000 ; delay for BLTouch
M280 P0 S160 ; BLTouch alarm release
G4 P100 ; delay for BLTouch
M117 Parking; Homing axes
G28; Home all axes
M420 S1 
; G29 ; auto bed leveling
M109 S[first_layer_temperature]; Wait for Extruder temperature
G1 Z2.0 F3000; Move Z Axis up little to prevent scratching of Heat Bed
G1 X6.1 Y20 Z0.3 F5000.0; Move to start position
M117 Purging
G1 X6.1 Y200.0 Z0.3 F1500.0 E15; Draw the first line
G1 X6.4 Y200.0 Z0.3 F5000.0; Move to side a little
G1 X6.4 Y20 Z0.3 F1500.0 E30; Draw the second line
G92 E0; Reset Extruder
M117 Print; Lets make
G1 Z2.0 F3000; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0; Move over to prevent blob squish`

digant73, 2024-03-04:
@MrKuskov in the meanwhile try to disable `emulated_m109_m190` in `config.ini` file

MrKuskov, 2024-03-05:
> What do you mean at step 1) and 2)? do you mean that you print successfully the first gcode file (step 1) and then you load a second gcode file (step 2) and there you have the issue?

And this too, but even if you run the same file for re-printing, the temperature is reset. If you interrupt printing and start printing the next model or start printing again, then the problem manifests itself randomly. The temperature may drop, or it may work out correctly.

> @MrKuskov in the meanwhile try to disable `emulated_m109_m190` in `config.ini` file.

It seems there is such a switch in the settings. Is it enough to disable it, or is it necessary to fix config.ini?

rondlh, 2024-03-05:
When the bed heating stops, does the display report that the target bed temp was changed, or does it stay at 60 degrees?
How about when you send a M105 in the terminal? What is the target bed temp?

To me this sounds like a lose or broken bed heating cable...

digant73, 2024-03-05:
@MrKuskov just made some testing yesterday with your 2 gcode files and your `config.ini` file and I didn't have any issue (printed one gcode file followed by the other, printed the same file more times, aborted a print followed by printing it again etc...)

MrKuskov, 2024-03-05:
@digant73 if you turn off emulated_m109_m190, the bed does not heat up at all.

digant73, 2024-03-05:
@MrKuskov it should properly work with both `emulated_m109_m190` enabled and disabled. This issue seems to be on Marlin side.
See also rondlh's suggestion

MrKuskov, 2024-03-05:
To be precise, here is the algorithm of work when everything works correctly: 1. I put the file to print. 2. The bed starts to heat up. 3 the hot end is lifted up. 4 the heating temperature of the bed is reset to 0. 5. The hotend begins to move to the coordinates X0 and Y0. 6. The bed starts to heat up again.

digant73, 2024-03-05:
as far as I see from the gcode file and also confirmed on TFT:
1., the bed temp is set to 60° with M140 (non blocking gcode)
2. the bed temp is also set to 60° with M190 (blocking gcode) until the target temp is reached
3. the target bed temp is reached and the hotend temp is set to 190° with M104 (non blocking gcode)
4. homing is performed while hotend temp is also increasing
5. the hotend temp is set to 190° with M109 (blocking gcode)
6. the target hotend temp is reached and the printing is started

As expected there is no bed temp reset neither on gcode file nor during the printing on my printers

MrKuskov, 2024-03-15:
It looks like the Advanced OK setting has solved the temperature problem. Since after [these corrections](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2910#issuecomment-1988557561), there was no more temperature reset.

closed 2024-03-15 by MrKuskov