bigtreetech/BIGTREETECH-TouchScreenFirmware #2906: [BUG] TFT70 not sending commands to the printer.

IoakeimSogiakas, 2024-03-05:
Well this is a weird issue. I have a skr v3 with custom firmware on it that if needed I will share (both on bugfix and on stable release). The issue is as follows, the screen works fine in marlin mode and does connect with the board, or at least I think it does, (tft in listening mode). However although it can display all the numbers temp bed size etc, it cannot or rather will not send to the printer any commands. I can send the firmware for both the board and the screen if needed.

digant73, 2024-03-05:
`TFT listening mode` feature (pressing on the button on the `Connection` menu) is needed only by few mainboards (e.g. MKS GENL 1) when printing from mainboard's USB port. That was due to the bus shared by mainboard and TFT. When in TFT listening mode, the TFT simply listen on serial line but doesn't send any gcode. For your mainboard you need to always have `TFT listening mode` disabled

IoakeimSogiakas, 2024-03-05:
oh so it is not connecting correctly then... RIght? the board is a skr 3 and it is connected via uart also here are the files

IoakeimSogiakas, 2024-03-05:
[FM Fles.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14494344/FM.Fles.zip)

closed 2024-03-06 by IoakeimSogiakas