bigtreetech/BIGTREETECH-TouchScreenFirmware #2907: [BUG] unknown G17 command

MrKuskov, 2024-03-05:
### Description

in the attached file, when printing through the screen, the message "Unknown G17 command" is written and after the first layer, the hotend moves along the isi Y until it stops. when printing via Marlin Mode, it prints correctly.
[Uploading ring_PLA_22m1s.zip…]()

digant73, 2024-03-05:
the attachment cannot be downloaded. However, the error message means that the G17 (CNC gcode) command present on the gcode file has to be enabled on Marlin side. If not enabled, Marlin will skip the command (it will not process it) and sends back to TFT that error message. It is not an issue at all on TFT fw

MrKuskov, 2024-03-05:
Thank you very much. This parameter is disabled, I'll try to turn it on.

MrKuskov, 2024-03-05:
turned on - the error is gone. thanks

closed 2024-03-05 by MrKuskov