bigtreetech/BIGTREETECH-TouchScreenFirmware #2910: [BUG]  Line number is not Last Liner Number+1

MrKuskov, 2024-03-09:
### Description

When printing any G-Code with the `Command checksum` function enabled, I constantly see errors. For example, in the calibration retraction test, I see the following errors at the beginning of printing. 
line: cur=6437, exp =6431. 
Line number is not Last Liner Number+1, last line: 6442. 
line: cur=6433, exp =6431.
 Line number is not Last Liner Number+1, last line: 6434.
 line: cur=6440, exp =6435. 
This is all when printing this file. The values change for different files, but the essence does not. At the same time, messages like "unknown command *84" or "-1.65 Y23" appear in the upper line, that is, it is clear that these are card reading errors (*84 - X84). Skipping one command has almost no effect on print quality, but when printing a thin-walled part, it is almost always a defect. I have tried using different cards in different connectors and USB FLash, sometimes errors disappear, but most often 1 or 2 per printing session appear. How to set up error-free printing or G-code duplication?
[retraction_tower_PETG_16m39s.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14545206/retraction_tower_PETG_16m39s.zip)

TFT35 B1 V3

digant73, 2024-03-09:
it seems you enabled advanced_ok but it is not properly supported/configured on Marlin side. Disable it on TFT side only (it is enough). Those errors you reported are related to a corruption on Marlin side (data are properly read from SD card)

MrKuskov, 2024-03-09:
yes it is enabled. And how to set it up correctly? It seems that I have followed all the instructions for setting up

rondlh, 2024-03-09:
> what buffer size should I set?

32 * 96 = 3KB, so if your hardware supports it set RX_BUFFER_SIZE to 2048, this should be fine.
Your BUFSIZE of 32 is quite big, you could also consider lowering it to 8 or 16. 4 is the default, and I personally use 8.
Without advanced ok you basically only send 1 command at a time and wait for Marlin to acknowledge it before sending the next command.

MrKuskov, 2024-03-10:
@rondlh I have a Biqu B1 [SKR 1.4](https://biqu.equipment/collections/control-board/products/bigtreetech-skr-v1-4-skr-v1-4-turbo-control-board-tmc2209-tmc2208-eeprom-v1-0-exp-mot-v1-0-3d-printer-parts-for-ender-3) board, can you tell me which buffer can be used for this board?

MrKuskov, 2024-03-11:
I used the following parameters: 
config.ini
` tx slots:4` 
Configuration_adv.h:
```
#define MAX_CMD_SIZE 96
#define BUFSIZE 8
#define TX_BUFFER_SIZE 32
#define RX_BUFFER_SIZE 1024
#if RX_BUFFER_SIZE >= 1024
  #define SERIAL_XON_XOFF
#endif
```
everything seems to be working correctly

closed 2024-03-11 by MrKuskov

digant73, 2024-03-15:
@MrKuskov remember that `tx_slots` in `config.ini` is used by TFT only if ADVANCED_OK is not enabled in `Configuration_adv.h` (in Marlin fw). If you enable ADVANCED_OK then  you enable the real advanced ok feature provided by Marlin and `tx_slots` in `config.ini` has no effect at all because the tx slots are reported by Marlin.