bigtreetech/BIGTREETECH-TouchScreenFirmware #2912: Machine name

jzagyi13, 2024-03-13:
Upgrade my BIQU B1 printer with a SKR 3 EX board so I upgraded the Marlin firmware to latest bug fix and I also updated the BTT TFT35 touchscreen fw to latest. "BIGTREE_TFT35_V3.0_B1.27.x" . After all updates my screen on TFT is showing the machine name as "Joe's BIQU " wich I defined in CUSTOM_MACHINE_NAME in the configuration.h file but the screen is adding: Kinematics: Cartesian
I have rebuilt marlin fw multiple times and installed and also the same with touchscreen fw.
I am also randomly getting and error of "Line Number is not Last Line Number+1, Last Line:0" after every Marlin fw install.
Attached pics show what screen was b4 new Marling FW and TFT upgrade and what it shows after upgrades:

[image]

[image]

thisiskeithb, 2024-03-13:
As of a couple weeks ago, Marlin now includes the machine`KINIMATICS` type in `M115`: https://github.com/MarlinFirmware/Marlin/commit/3fd175af8ea56db48ea41d7f3d55e967f0df8cd8

jzagyi13, 2024-03-13:
Is this something I should be able to disable? When I comment that line out Line 67:  #define MACHINE_KINEMATICS "" \.
I get error when building : [{
	"resource": "/e:/Marlin/Marlin-bugfix-2.1.x/Marlin/src/gcode/host/M115.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "expected ')' before 'MACHINE_KINEMATICS'",
	"startLineNumber": 84,
	"startColumn": 20,
	"endLineNumber": 84,
	"endColumn": 20
}]

thisiskeithb, 2024-03-13:
Patching [`/TFT/src/User/API/Mainboard_AckHandler.c`](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/API/Mainboard_AckHandler.c) seems to work for both newer & older `M115` output:

```diff
diff --git a/TFT/src/User/API/Mainboard_AckHandler.c b/TFT/src/User/API/Mainboard_AckHandler.c
index 6e83b9b..bd29674 100644
--- a/TFT/src/User/API/Mainboard_AckHandler.c
+++ b/TFT/src/User/API/Mainboard_AckHandler.c
@@ -1273,7 +1273,9 @@ void parseAck(void)
         string = &ack_cache[ack_index];
         string_start = ack_index;
 
-        if (ack_seen("EXTRUDER_COUNT:"))
+        if (ack_seen("KINEMATICS:"))
+          string_end = ack_index - sizeof("KINEMATICS:");
+        else if (ack_seen("EXTRUDER_COUNT:"))
         {
           if (MIXING_EXTRUDER == 0)
             infoSettings.ext_count = ack_value();
```

I'll submit a PR shortly.

thisiskeithb, 2024-03-13:
> I am also randomly getting and error of "Line Number is not Last Line Number+1, Last Line:0" after every Marlin fw install.

See https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/2908

I don't always get this, but disabling the "Command checksum" feature under `Menu -> Settings -> Feature` stops the errors on the TFT.

jzagyi13, 2024-03-13:
I am sorry but I am totally new to all this. What do I need to do with your suggested fix? I did a rebuild with a fresh bug fix(Most current) and also ran the TFT35 fw update using the "BIGTREETECH-TouchScreenFirmware-pr-fix_machine_name" . After all that I am still getting the same naming issue? Did I miss something? Again I am new to this github site and updating all this firmware.
Thanks so much for help.

thisiskeithb, 2024-03-13:
Manually replace `/TFT/src/User/API/Mainboard_AckHandler.c` with `Mainboard_AckHandler.c` from this zip: 
[Mainboard_AckHandler.c.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14593449/Mainboard_AckHandler.c.zip)

Recompile the TFT firmware with this replaced file & update the TFT firmware to test the new code.

jzagyi13, 2024-03-13:
I obviously am doing something wrong. I took your zip file, Mainboard_AckHandler.c and copied it to the \TFT\src\User\API and overrode the existing Mainboard_AckHandler.c. Opened the the touchscreen firmware folder in Visual Studio Core and "cleaned" and the "Built". Then I did the update of the firmware on my BIQU B1 printer and everything was successful but the name is still adding "Kinematics: Cartesian".

thisiskeithb, 2024-03-14:
Ignore the odd colors, but I cannot reproduce the issue on any of the Marlin builds / TFTs I have here with the referenced `Mainboard_AckHandler.c` patch:

[image]

jzagyi13, 2024-03-14:
So I have noticed that with the Marlin-bugfix-2.1.x I get the issue of the "Kinematics: Cartesian" but with the Marlin-2.1.2.2 I do NOT get the issue.

thisiskeithb, 2024-03-14:
> So I have noticed that with the Marlin-bugfix-2.1.x I get the issue of the "Kinematics: Cartesian" but with the Marlin-2.1.2.2 I do NOT get the issue.

The kinematic info in `M115` output was added to `bugfix-2.1.x` a couple weeks ago. `2.1.2.2` is a minor update to `2.1.2` which was released in December of 2022.

PR #2913 (2024-03-25): Fix MACHINE_TYPE with KINEMATICS in M115
### Description

Fixes #2912

As of https://github.com/MarlinFirmware/Marlin/commit/3fd175af8ea56db48ea41d7f3d55e967f0df8cd8 / https://github.com/MarlinFirmware/Marlin/pull/26806, Marlin now outputs the `KINEMATICS` type in the `M115` report. This causes `KINEMATICS:<type>` to also be displayed after `MACHINE_TYPE` (printer name) on the main screen in the Status section and on the Info screen.

I tested the old & new `M115` output with this patch and it seems to work correctly.

### Benefits

`KINEMATICS:<type>` will no longer be appended to the `MACHIN [... cut]
closed 2024-03-25 by bigtreetech