bigtreetech/BIGTREETECH-TouchScreenFirmware #2914: [Q] ESP3D and (TFT35 or SKR1.)

MrKuskov, 2024-03-13:
I want to set up ESP3D based on ESP32-01S to send print status notifications (time to finish printing and end of printing) to telegram. tell me where it is better to install the module ESP32-01S. in the SKR1.4 board or in the TFT35 B1 V3 screen.

radek8, 2024-03-13:
I also bought an ESP32-01S.
Works on SKR V1.4 as well as TFT.
ESP mapped it as a virtual serial port and controlled the printer via the serial port. Unfortunately, there was still random print cancellation when printing from the SD card. My antivirus sometimes performed a port and network scan, and random sequences that came to the printer canceled the prints.
Now when printing Wi-Fi, I disconnect the module and use it only when calibrating and setting up the printer. I have a printer and a PC in different rooms, and USB cannot be used.
  This is my experience with this module. :-)

radek8, 2024-03-14:
The WIfi module communicates via the serial port.
TFT forwards the communication that comes to the WIFI serial port to Merlin.
In my opinion, it doesn't matter if you connect the module to SKR or TFT.
The result will be the same.
There is nothing easier than trying both variants and using the one that suits your needs.

closed 2026-03-20 by github-actions[bot]