bigtreetech/BIGTREETECH-TouchScreenFirmware #2915: [ISSUE] z motor not stop

devilangelsoul, 2024-03-14:
Hi, I have a problem, I updated to the latest firmware, but it's giving me a problem, if I send the printer to home the second motor of the Z axis (the one on the right) doesn't stop, the end stop isn't recognised.

### Hardware Variant
TFT35 V3.0 on anycubic I3 Mega s

Before the update everything worked without problems

rondlh, 2024-03-15:
The motor logic is controlled by the host(mainboard probably running Marling), it's not controlled by the TFT.
The TFT just sends a G28, then the motherboard handles the execution like stepper control and endstops.
So check your host FW setup and wiring...

devilangelsoul, 2024-03-15:
> The motor logic is controlled by the host(mainboard probably running Marling), it's not controlled by the TFT. The TFT just sends a G28, then the motherboard handles the execution like stepper control and endstops. So check your host FW setup and wiring...

but before updating the monitor everything worked fine.

rondlh, 2024-03-17:
> but before updating the monitor everything worked fine.

Then check your wiring, you can use M119 to check if your endstops are working

closed 2026-03-20 by github-actions[bot]