bigtreetech/BIGTREETECH-TouchScreenFirmware #2916: [BUG] TFT35 does not preview only 95x95 sketches

MrKuskov, 2024-03-15:
### Description

The TFT35 preview sketches with a resolution of 95x95 G-code. When you specify the BTT TT thumbnail format and size 95x95 in OrcaSlicer, they are not displayed. The Cura plugin generates 5 thumbnail resolutions at once. I have experimentally established that the minimum number of image resolutions required to display on the TFT35 B1 V3 screen is 3 pieces: 70x70, 95x80 and 95x95. I have attached all the options in the archive. Only one of them displays a thumbnail on the screen.
[gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14612866/gcode.zip)

rlneumiller, 2024-04-12:
Which one of the five gcode files displays the thumbnail properly for you?

MrKuskov, 2024-04-15:
@rlneumiller Sorry. None of them. I forgot to attach it. there must be 3 resolutions in the file 70x70, 95x80 and 95x95 for the thumbnail to be displayed.
[OrcaCube_PETG_1h52m.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14973927/OrcaCube_PETG_1h52m.zip)

rlneumiller, 2024-04-15:
This looks like an issue with the gcode output of your slicer (assuming that `THUMBNAIL_PARSER = 1` was set in `Configuration.h` used to build your firmware)

The thumbnail signatures in the gcode files you've provided that contain a single thumbnail are not formatted properly.
For a single thumbnail in RGB565 format with a size of 0x5Fx5F (95x95 dec), your slicer needs to write the following signature at the top of the thumbnail data in the gcode file:
`; bigtree thumbnail begin 005f005f`

Here is a corrected version of your 95.gcode that should work.
[corrected-95.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/14979628/corrected-95.zip)

MrKuskov, 2024-04-16:
@rlneumiller there is no THUMBNAIL_PARSER in my version of Configuration.h
https://github.com/MrKuskov/Configurations/blob/patch-1/config/examples/BIQU/B1/Configuration.h

rlneumiller, 2024-04-16:
The Marlin default setting for THUMBNAIL_PARSER is THUMBNAIL_PARSER = 0 (which requires 3 thumbnails in the gcode file)
If you add THUMBNAIL_PARSER = 1 to the end of your Configuration.h and rebuild\flash the firmware for your printer, the gcode in corrected-95.zip should then work as expected.

MrKuskov, 2024-04-17:
@rlneumiller I made changes to Marlin #define THUMBNAIL_PARSER 1 And it doesn't work. with the file corrected-95.zip . And I don't understand how changes in Marlin can affect what is displayed on the TFT35 display, because the thumbnail is read before the file is sent to print.

rlneumiller, 2024-04-17:
Apologies. I referenced Marlin in error and lost sight of the fact that this display is essentially a remote control for Marlin.

Your desired behavior would require adding functionality to TFT to allow for control of the thumbnail format via a setting within TFT's config.ini.

You could still workaround by modification of Configuration.h as described earlier but within the TFT source tree (see below), which would then require rebuilding BIGTREE_TFT35_V3.0_B1.27.x.bin (or whichever bin matches your hardware), and ensuring that your slicer is outputting the proper thumbnail header.

See: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/TFT/src/User/Configuration.h

Ultimately this issue is more of a feature request than a bug.

MrKuskov, 2024-04-17:
@rlneumiller Thank you for pointing out the course of action. I have reassembled the firmware for the TFT screen. With the THUMBNAIL_PARSER =1 parameter, thumbnails are displayed with a delay, but they did not appear for the corrected-95 file. Therefore, I use the THUMBNAIL_PARSER =2 parameter, in the slicer settings I chose the PNG format and the sketch size 95x95, gcode turns out to be smaller than BTT 70x70, 95x80, 95x95. I will have to choose to display quickly or a smaller file)))